Ysurac/openmptcprouter #3205: OMR disconnects from the backup VPS if the master comes online later.

vempire-ghost, 2024-03-12:
## Expected Behavior
OMR connects without issues when configured with a backup VPS.

## Current Behavior
"OMR oscillates between master and backup if both are connected for a while until it connects to the master."

## Context (Environment)
In snapshots from 06/03 and earlier, when OMR connected to the VPS, it would connect to whichever was available first and stay connected to it until it was no longer available, whether it was the backup or the master. 
However, now it seems to want to connect to the master. If it connects to the backup first, as soon as it manages to connect to the master, it disconnects from the backup, dropping all connections. In the processes, sometimes it does this several times in a row.
Here are some logs.
Mar 12 04:09:32 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down ( xx.215.248.xxx)
Mar 12 04:09:34 OpenMPTCProuter user.notice OMR-Tracker-Server: Use backup server  (xx.233.166.xxx)
Mar 12 04:09:39 OpenMPTCProuter daemon.info omr-tracker-xray: Reload xray rules
Mar 12 04:09:39 OpenMPTCProuter user.notice xray: Rules UP
Mar 12 04:09:39 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Mar 12 04:09:39 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Mar 12 04:09:40 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Mar 12 04:09:48 OpenMPTCProuter daemon.err glorytun[7880]:  xx.215.248.xxx.65001: connected
Mar 12 04:09:55 OpenMPTCProuter user.notice OMR-VPS: Get config from server vps_us...
Mar 12 04:09:55 OpenMPTCProuter user.notice OMR-VPS: V2ray restart...
Mar 12 04:09:55 OpenMPTCProuter user.notice OMR-VPS: Xray restart...
Mar 12 04:09:55 OpenMPTCProuter daemon.err glorytun[7880]:  xx.215.248.xxx.65001: key exchange failed
Mar 12 04:09:55 OpenMPTCProuter daemon.info glorytun[7880]: STOPPED tun0
Mar 12 04:09:55 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Mar 12 04:09:55 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Mar 12 04:09:55 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Mar 12 04:09:55 OpenMPTCProuter user.notice xray: add rules
Mar 12 04:09:55 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Mar 12 04:09:55 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Mar 12 04:09:56 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Mar 12 04:09:56 OpenMPTCProuter daemon.info xray[23686]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.22.0 linux/amd64)
Mar 12 04:09:56 OpenMPTCProuter daemon.info xray[23686]: A unified platform for anti-censorship.
Mar 12 04:09:56 OpenMPTCProuter daemon.info xray[23686]: 2024/03/12 04:09:56 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Mar 12 04:09:56 OpenMPTCProuter user.notice OMR-VPS: Glorytun restart...
Mar 12 04:09:56 OpenMPTCProuter daemon.err glorytun[7880]:  xx.215.248.xxx.65001: connected
Mar 12 04:09:56 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn
Mar 12 04:09:56 OpenMPTCProuter user.notice OMR-VPS: Glorytun UDP restart...
Mar 12 04:09:56 OpenMPTCProuter user.notice OMR-VPS: OpenVPN restart...
Mar 12 04:09:56 OpenMPTCProuter user.notice OMR-VPS: MLVPN restart...
Mar 12 04:09:56 OpenMPTCProuter user.notice OMR-VPS: DSVPN restart...
Mar 12 04:10:01 OpenMPTCProuter daemon.info procd: Instance glorytun::instance1 pid 7880 not stopped on SIGTERM, sending SIGKILL instead
Mar 12 04:10:01 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Mar 12 04:10:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Mar 12 04:10:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Mar 12 04:10:01 OpenMPTCProuter daemon.info glorytun[28366]: INITIALIZED tun0
Mar 12 04:10:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Mar 12 04:10:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Mar 12 04:10:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Mar 12 04:10:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Mar 12 04:10:01 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Mar 12 04:10:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Mar 12 04:10:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Mar 12 04:10:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Mar 12 04:10:02 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Mar 12 04:10:02 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Mar 12 04:10:02 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Mar 12 04:10:02 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar 12 04:10:02 OpenMPTCProuter daemon.err glorytun[28366]: xx.233.166.xxx.65001: connected
Mar 12 04:10:03 OpenMPTCProuter daemon.info glorytun[28366]: STARTED tun0
Mar 12 04:10:03 OpenMPTCProuter user.notice MPTCP: Set eth1 to master
Mar 12 04:10:03 OpenMPTCProuter user.notice MPTCP: Set eth2 to on
Mar 12 04:10:03 OpenMPTCProuter user.notice MPTCP: Set eth3 to backup
Mar 12 04:10:03 OpenMPTCProuter user.notice MPTCP: Set eth4 to on
Mar 12 04:10:03 OpenMPTCProuter user.notice MPTCP: Set eth5 to on
Mar 12 04:10:03 OpenMPTCProuter user.notice MPTCP: Disabling MPTCP on interface ifb4tun0 not found in enabled multipath list
Mar 12 04:10:08 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Mar 12 04:10:09 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Mar 12 04:10:09 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Mar 12 04:10:09 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Mar 12 04:10:09 OpenMPTCProuter user.notice xray: add rules
[... 97 lines cut ...]
Mar 12 04:10:51 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Mar 12 04:10:51 OpenMPTCProuter daemon.info glorytun[23159]: INITIALIZED tun0
Mar 12 04:10:51 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Mar 12 04:10:51 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Mar 12 04:10:51 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Mar 12 04:10:51 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar 12 04:10:52 OpenMPTCProuter user.notice MPTCP: Set eth1 to master
Mar 12 04:10:52 OpenMPTCProuter user.notice MPTCP: Set eth2 to on
Mar 12 04:10:52 OpenMPTCProuter user.notice MPTCP: Set eth3 to backup
Mar 12 04:10:52 OpenMPTCProuter user.notice MPTCP: Set eth4 to on
Mar 12 04:10:52 OpenMPTCProuter daemon.err glorytun[23159]: xx.233.166.xxx.65001: connected
Mar 12 04:10:52 OpenMPTCProuter user.notice MPTCP: Set eth5 to on
Mar 12 04:10:52 OpenMPTCProuter daemon.info glorytun[23159]: STARTED tun0
Mar 12 04:10:53 OpenMPTCProuter user.notice MPTCP: Disabling MPTCP on interface ifb4tun0 not found in enabled multipath list
Mar 12 04:10:53 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0

Would it be possible to add a configuration for the user to decide whether they want OMR to force the Master even if it is connected to the backup, or if OMR, once connected to a VPS, does not disconnect from it to connect to the master when it comes online?

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60rc1-5.4 r0+16862-170d9e4 snap from 11/03
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1029-test
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: x86_64
  - Country: Brazil

aladdinshammr, 2024-04-15:
now if I want the backup connection to work I need a second VPS ??
therefore when I lose the connection to the first I will still connected to the second one ?

vempire-ghost, 2024-04-15:
> now if I want the backup connection to work I need a second VPS ?? therefore when I lose the connection to the first I will still connected to the second one ?

No, this topic is about the backup server of the VPS, which serves in case the main server of the VPS goes down or becomes offline. It's not related to backup connection; that works normally with just one VPS and doesn't require a backup VPS.

closed 2024-07-20 by github-actions[bot]

levelasquezd, 2026-10-09:
I'm currently evaluating OpenMPTCProuter for a production high-availability architecture using two independent VPS providers and two WAN connections (fiber and Starlink).

This issue is particularly important for our use case. Automatically switching back to the primary VPS and terminating existing connections when the backup VPS is healthy defeats part of the purpose of high availability.

Could the maintainers please clarify:

1. Has this behavior been fixed in the current stable release (v0.64.x)?
2. If so, could you reference the relevant commit or configuration option?
3. If not, could this issue be reopened and considered for a future release?

Ideally, failback to the primary VPS should be configurable, with the option to remain on the currently healthy backup until a controlled switchover can occur.

We are evaluating this architecture before deployment, so I cannot claim to have reproduced the issue on the latest release. However, predictable failover and failback behavior is a critical requirement for our implementation.

Thank you for maintaining this project.