Ysurac/openmptcprouter #3206: Interface metric assingnment is wrong

rdmitry0911, 2024-03-12:
## Expected Behavior

When I add more than 1 interface without multipath support I can use all of them in omr-bypass
## Current Behavior

I can have only one interface without multipath support in the system
When I add via luci/openMPTCProuter/Settings Wizard several interfaces without multipath support, the script generates metric 2 (and interface id 2) for all added interfaces which makes it impossible to use in omr-bypass
## Possible Solution

## Steps to Reproduce the Problem

  1. Add first interface without multipath support
  2. Add second interface without multipath support
  3. Add third interface without multipath support and check interface id in /etc/config/omr-bypass in my case eth3, eth4 and eth5 are interfaces without multipath support and omr-bypass looks like this:
```
.
.
.
config interface 'eth0'
	option id '9999'

config interface 'eth1'
	option id '2'

config interface 'eth2'
	option id '3'

config interface 'tun0'
	option id '1500'

config global 'global'
	option vpn_ipv4_md5 '68b329da9893e34099c7d8ad5cb9c940'
	option vpn_ipv6_md5 '68b329da9893e34099c7d8ad5cb9c940'

config interface 'eth3'
	option id '1'

config interface 'eth4'
	option id '2'

config interface 'eth5'
	option id '2'
```
## Context (Environment)

As a result traffic is routing via wrong rules

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60beta2-6.1 r0+24843-acf40c022e 
  - OpenMPTCProuter VPS version: 0.1029-test 6.1.0-18-amd64
  - OpenMPTCProuter VPS provider:Hetzner
  - OpenMPTCProuter platform: x86_64
  - Country: GR

Ysurac, 2024-03-12:
Can you try with latest snapshot ?

rdmitry0911, 2024-03-12:
Yes, this problem exists with the latest commits on Mar 11, 2024

Ysurac, 2024-03-12:
I'm not able to reproduce the issue.
Metric are set in /etc/config/openmptcprouter by /etc/init.d/mptcp. omr-bypass only use them.

closed 2024-03-13 by rdmitry0911

Ysurac, 2024-03-13:
Thanks for the feedback.
I will add a command to fix previous backup.