Ysurac/openmptcprouter #3207: VPN tunnel down, Cant get public ip address from shadowsocks, cant contact server admin (no token yet available)

TaperPowel, 2024-03-12:
## Expected Behavior

the expected behavior is every thing works correctly 
## Current Behavior

i can still access the internet but its pretty slow at points in the day
from the start I've been getting these errors, i assume this is why there is slow internet
VPN tunnel down
Cant get public Ip address from shadowsocks
cant contact server admin (no token yet available)
[image]
https://pastebin.com/nN55AJx1
## Specifications
Powered by openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter version:  openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version:  Ubuntu 20.04.6 OpenMPTCProuter VPS 0.1028
  - OpenMPTCProuter VPS provider: google 
  - OpenMPTCProuter platform: RPI3

Ysurac, 2024-03-12:
It can't connect to VPS API, did you open 65000-65535 port range on GCP ?

aladdinshammr, 2024-04-08:
@Ysurac 

I face the same issue.
when I configure the router it works just fine.
but when I start to consume all the aggregated  speed I see " No Server ping response after 1 second" on the master connection. then I lose the connection to the VPS.

If there is a way to exclude the master connection from the aggregation and let it's solely job to connect to the server.
or do I need to configure something in OMR-tracker?

in a whisper voice => the highest speed here is just 8 Mbps

router version : v0.60 rc2 
VPS : 5.4.207-mptcp
provider: google cloud service

vempire-ghost, 2024-04-09:
> @Ysurac
> 
> I face the same issue. when I configure the router it works just fine. but when I start to consume all the aggregated speed I see " No Server ping response after 1 second" on the master connection. then I lose the connection to the VPS.
> 
**> If there is a way to exclude the master connection from the aggregation and let it's solely job to connect to the server. or do I need to configure something in OMR-tracker?**
> 
> this is what I get in the log
> 
> Apr 8 23:28:06 OpenMPTCProuter daemon.err omr-tracker[25131]: sh: 66.6667: bad number Apr 8 23:35:23 OpenMPTCProuter daemon.err omr-tracker[25130]: sh: 66.6667: bad number
> 
> in a whisper voice => the highest speed here is just 8 Mbps
> 
> router version : v0.60 rc2 VPS : 5.4.207-mptcp provider: google cloud service

I made this request in this topic [https://github.com/Ysurac/openmptcprouter/issues/3131](url) but didn't get a response, which would be the option to configure the Master WAN as backup, so the MPTCP would only use it to initiate connections and communicate with the VPS, and the other WANs would be used for heavy traffic. Is it possible to configure this way in the context of MPTCP, but in OMR, it's configured to force the master WAN as default.

There's a way to work around this by disabling this configuration `set network.globals.mptcp_force_multipath='0'` and adding this script to the Scheduled Tasks: `*/1 * * * * multipath eth1 backup`, assuming your master WAN is wan1. If it's another one, change it accordingly.

However, keep in mind that with this configuration, your master WAN will no longer aggregate the connection speed of all WANs and will be used solely to contact the VPS and initiate connections.

aladdinshammr, 2024-04-09:
@vempire-ghost 

hi, Thanks for help .
I ran this in the terminal "set network.globals.mptcp_force_multipath='0'"
and put this in the Scheduled Tasks like this 
[image]

is everything correct ??

vempire-ghost, 2024-04-09:
> @vempire-ghost
> 
> hi, Thanks for help . I ran this in the terminal "set network.globals.mptcp_force_multipath='0'" and put this in the Scheduled Tasks like this ![task](https://private-user-images.githubusercontent.com/91631145/320723096-b4ab8750-ac25-458e-95b4-4ef2f729674d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI2Mzc5NDEsIm5iZiI6MTcxMjYzNzY0MSwicGF0aCI6Ii85MTYzMTE0NS8zMjA3MjMwOTYtYjRhYjg3NTAtYWMyNS00NThlLTk1YjQtNGVmMmY3Mjk2NzRkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA5VDA0NDA0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTExYTNjZDhmNDJiMzg1ZGJhYTgxYjM0ZGNhZmNjZDYzYTEwOWVlMDZiNzlkN2NjMWE0NmVlMzMwYzA4ZDExMTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XyHWbujcDGLOsh1EML_kAFYeJEca1mDbpRdNATZ1_GQ)
> 
> is everything correct ??

No, the command should use eth1 for wan1, eth2 for wan2... So if your wan1 is the master, keep it exactly as I put in the example `*/1 * * * * multipath eth1 backup`.

To check if it's working, you can use the command `multipath` in the terminal and check if eth1 is set as backup instead of default. You can also see in the Network Overview of OMR; it will be marked in red that the WAN is set as backup because OMR does not accept this status as correct.

aladdinshammr, 2024-04-09:
@vempire-ghost 
it marked in red for a while after that it became master again.

erspan0 is deactivated
eth0 is deactivated
gre0 is deactivated
gretap0 is deactivated
ifb4tun0 is deactivated
ifb4wan1 is deactivated
ifb4wan2 is deactivated
ifb4wan3 is deactivated
ifb4wan4 is deactivated
ifb4wan5 is deactivated
ip6gre0 is deactivated
ip6tnl0 is deactivated
lo is deactivated
sit0 is deactivated
teql0 is deactivated
tun0 is deactivated
wan1 is in default mode
wan2 is in default mode
wan3 is in default mode
wan4 is in default mode
wan5 is in default mode
wlan0 is deactivated

vempire-ghost, 2024-04-09:
> @vempire-ghost it marked in red for a while after that it became master again.
>
> erspan0 is deactivated eth0 is deactivated gre0 is deactivated gretap0 is deactivated ifb4tun0 is deactivated ifb4wan1 [...]

Check in the advanced settings if the line `network.globals.mptcp_force_multipath='0'` is indeed set to the value 0. However, OMR can sometimes change the value back to Default, but the script would revert it back to backup every minute.

There is another way to force the master as backup, but it involves modifying a file in OMR, it's a simple change but it can cause issues, so have a backup handy if you want to proceed.
Edit the file /etc/init.d/mptcp of OMR using nano, go to line 179, and change lines 179 to 182, replacing the values "On" with "Backup", so it becomes like this:
```
			mode="backup"
			config_set "$config" multipath "backup"
			uci -q set network.${config}.multipath="backup"
			uci -q set openmptcprouter.${config}.multipath="backup"
```
Go to line 413 and delete the lines 413, 414, and 415 as follows:
```
		if ([ "$mode" = "master" ] || [ "$mode" = "on" ]) && [ -z "$(multipath $iface | grep default)" ]; then
			logger -t "MPTCP" "Set $iface to $mode from $(multipath $iface)"
			multipath "$iface" "on"
```
And then add these new lines in place:
```
		if [ "$mode" = "master" ] && [ -z "$(multipath $iface | grep backup)" ]; then
			logger -t "MPTCP" "Set $iface to $mode from $(multipath $iface)"
			multipath "$iface" "backup"
		elif [ "$mode" = "on" ] && [ -z "$(multipath $iface | grep default)" ]; then
			logger -t "MPTCP" "Set $iface to $mode from $(multipath $iface)"
			multipath "$iface" "on"
```
With these changes, the OMR will set the master WAN as backup without forcing it to default, although it will still appear in red in the Network Overview.

aladdinshammr, 2024-04-09:
@vempire-ghost
I appreciate your effort, thanks. 
I'll try it and give you a feedback

Here you have to wait for people to go to sleep then you can use the internet

aladdinshammr, 2024-04-10:
@vempire-ghost 

unfortunately, it didn't work.

frankly I don't have much experience in networks, but after some searching I found out that this issue is not because of the heavy load on master connection.

my problem is exactly this https://github.com/multipath-tcp/mptcp/issues/108 
especially when I connect mikrotik router to OMR router.

Do you know how did they fix the issue???

vempire-ghost, 2024-04-10:
> @vempire-ghost
> 
> unfortunately, it didn't work.
> 
> frankly I don't have much experience in networks, but after some searching I found out that this issue is not because of the heavy load on master connection.
> 
> my problem is exactly this [multipath-tcp/mptcp#108](https://github.com/multipath-tcp/mptcp/issues/108) especially when I connect mikrotik router to OMR router.
> 
> Do you know how did they fix the issue???
> 
> if I disabled this thing "net.ipv4.tcp_mtu_probing" will OMR work perfectly?

It's unfortunate that it didn't work. I'm afraid I don't have much networking knowledge and can't assist you with the issue you've reported.

aladdinshammr, 2024-04-11:
@vempire-ghost 

it's okay. You helped me a lot so far.

thanks a million.

Ysurac, 2024-04-11:
@aladdinshammr can you open a new issue for this problem ?

aladdinshammr, 2024-04-11:
@Ysurac 

sure I will do.

thanks to you as well

MoweME, 2024-04-13:
I did a fresh installation of my VPS and router and still had the same issue.

When activating "Show advanced settings" in the Settings Wizard, I noticed that my router wasn't able to fetch all the tokens needed for the VPN connections.

After further investigation, I've noticed that many services were inactive on my VPS!
[image]

After starting the omr service, the router was able to fetch the tokens from the VPS again:
`systemctl start omr.service`

This fixed most of my issues, bringing up a new one:
[image]

I was capable of fixing that by switching the Default Proxy from "Shadowsocks-Rust 2022" to "Shadowsocks"!
[image]

It seems like the current version has some broken settings out-of-the-box, which required me to spend a Friday evening on this issue...

Ysurac, 2024-04-13:
@MoweME please open a new issue.

closed 2024-08-30 by github-actions[bot]