Ysurac/openmptcprouter #3209: install openmptcprouter on a mini PC hard drive ?

patimages, 2024-03-14:
Hello, 
Stupid question : how can I install openmptcprouter on a mini PC hard drive ? If I install Debian on it, I cannot reinstall on the booted system so how to proceed ? 
Many thanks for your help !!!

Ysurac, 2024-03-14:
Via an USB key with any  Linux based live OS and then via something like ```wget -O- 'https://download.openmptcprouter.com/release/v0.60rc1-6.1/x86_64/targets/x86/64/openmptcprouter-v0.60rc1-6.1-r0+24843-acf40c022e-x86-64-generic-squashfs-combined.img.gz' | gunzip | dd of=/dev/sda```

Ysurac, 2024-03-14:
Yes, as when the OS is running on USB stick it doesn't access the hard drive, so you can write without issues.
OMR is a full image os, you can write is from any live running OS it will erase everything on destination drive.

patimages, 2024-03-14:
And all worked perfectly : I am on v0.60rc1 and the VPS on 0.1029-test (is it the right version for the VPS ?).
Merci encore Ycarus !

Ysurac, 2024-03-15:
For the VPS you can use rc version or snapshot version, if the kernel is correctly using a kernel 6.1 it's ok ```uname -a``` to verify (else it's indicated in status page).

closed 2024-03-15 by Ysurac