Ysurac/openmptcprouter #3210: OMR is trafficing a large amount of data just to connect to the VPS.

vempire-ghost, 2024-03-15:
## Expected Behavior
OMR does not consume unnecessary bandwidth when connecting to the VPS.

## Current Behavior
"OMR, when connecting to the VPS, traffics a large amount of data through the proxy or VPN for no apparent reason."

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60rc1-5.4 r0+16862-170d9e4
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1029-test
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: x86_64

Since the snapshot of March 8th, whenever OMR connects to the VPS, it has been trafficing a strange amount of data. Normally, upon connecting, without any usage, there would be less than 1MB of data trafficked, but now it reaches over 1GB within seconds, then stops and normalizes.
This has never happened before. I don't have the 'Calculate speed' option enabled on any interface. Do you know if OMR downloads anything from the VPS when connecting? Since I use data traffic quota connections, I have to optimize my traffic as much as possible, and losing over 1GB of quota just by connecting OMR to the VPS is a big problem.

Mar 15 01:20:40 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later
Mar 15 01:20:40 OpenMPTCProuter user.notice xray: Rules DOWN
Mar 15 01:20:46 OpenMPTCProuter daemon.err glorytun[10407]: xx.215.224.xx.65001: key exchange failed
Mar 15 01:20:46 OpenMPTCProuter daemon.info glorytun[10407]: STOPPED tun0
Mar 15 01:20:47 OpenMPTCProuter daemon.err glorytun[10407]: xx.215.224.xx.65001: connected
Mar 15 01:20:47 OpenMPTCProuter daemon.info glorytun[10407]: STARTED tun0
Mar 15 01:20:48 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Mar 15 01:20:49 OpenMPTCProuter user.notice post-tracking-020-status: New public ip detected for omrvpn (tun0): xx.215.224.xx (previous: xx.133.198.xx)
Mar 15 01:20:49 OpenMPTCProuter user.notice post-tracking-020-status: Reload MPTCP for tun0
Mar 15 01:20:52 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 1.1.1.1)
Mar 15 01:20:59 OpenMPTCProuter daemon.info omr-tracker-xray: Reload xray rules
Mar 15 01:21:00 OpenMPTCProuter user.notice xray: Rules UP
Mar 15 01:21:00 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Mar 15 01:21:00 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Mar 15 01:21:00 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Mar 15 01:22:00 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps up (xx.215.224.xx), set it back
Mar 15 01:22:00 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Mar 15 01:22:00 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Mar 15 01:22:00 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Mar 15 01:22:00 OpenMPTCProuter user.notice xray: add rules
Mar 15 01:22:00 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Mar 15 01:22:00 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Mar 15 01:22:01 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Mar 15 01:22:01 OpenMPTCProuter daemon.info xray[14648]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.22.0 linux/amd64)
Mar 15 01:22:01 OpenMPTCProuter daemon.info xray[14648]: A unified platform for anti-censorship.
Mar 15 01:22:01 OpenMPTCProuter daemon.info xray[14648]: 2024/03/15 01:22:01 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Mar 15 01:22:01 OpenMPTCProuter daemon.info glorytun[10407]: STOPPED tun0
Mar 15 01:22:01 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Mar 15 01:22:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Mar 15 01:22:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Mar 15 01:22:01 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn
Mar 15 01:22:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Mar 15 01:22:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Mar 15 01:22:01 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Mar 15 01:22:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Mar 15 01:22:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Mar 15 01:22:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Mar 15 01:22:01 OpenMPTCProuter daemon.info glorytun[15299]: INITIALIZED tun0
Mar 15 01:22:01 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Mar 15 01:22:01 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Mar 15 01:22:02 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Mar 15 01:22:02 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar 15 01:22:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Change in routes, set ip route replace default scope global nexthop via 192.168.0.1 dev eth1 weight 100 nexthop via 192.168.0.1 dev eth2 weight 1 nexthop via 192.168.0.1 dev eth4 weight 1 nexthop via 19
Mar 15 01:22:02 OpenMPTCProuter daemon.err glorytun[15299]: xx.215.224.xx.65001: connected
Mar 15 01:22:02 OpenMPTCProuter user.notice MPTCP: Set eth1 to master
Mar 15 01:22:02 OpenMPTCProuter daemon.info glorytun[15299]: STARTED tun0
Mar 15 01:22:02 OpenMPTCProuter user.notice OMR-VPS: Get config from server vps...
Mar 15 01:22:02 OpenMPTCProuter user.notice MPTCP: Set eth2 to on
Mar 15 01:22:03 OpenMPTCProuter user.notice MPTCP: Set eth3 to backup
Mar 15 01:22:03 OpenMPTCProuter user.notice OMR-VPS: V2ray restart...
Mar 15 01:22:03 OpenMPTCProuter user.notice MPTCP: Set eth4 to on
Mar 15 01:22:03 OpenMPTCProuter user.notice MPTCP: Set eth5 to on
Mar 15 01:22:03 OpenMPTCProuter user.notice OMR-VPS: Xray restart...
Mar 15 01:22:03 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
[... 121 lines cut ...]
Mar 15 01:22:30 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Mar 15 01:22:30 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Mar 15 01:22:30 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Mar 15 01:22:30 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Mar 15 01:22:30 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Mar 15 01:22:30 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Mar 15 01:22:30 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Mar 15 01:22:30 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Mar 15 01:22:30 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Mar 15 01:22:30 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Mar 15 01:22:30 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar 15 01:22:31 OpenMPTCProuter daemon.err glorytun[18576]: xx.215.224.xx.65001: connected
Mar 15 01:22:31 OpenMPTCProuter daemon.info glorytun[18576]: STARTED tun0
Mar 15 01:22:31 OpenMPTCProuter user.notice MPTCP: Set eth1 to master
Mar 15 01:22:31 OpenMPTCProuter user.notice MPTCP: Set eth2 to on
Mar 15 01:22:31 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Change in routes, set ip route replace default scope global nexthop via 192.168.0.1 dev eth1 weight 100 nexthop via 192.168.0.1 dev eth2 weight 1 nexthop via 192.168.0.1 dev eth4 weight 1 nexthop via 19
Mar 15 01:22:31 OpenMPTCProuter user.notice MPTCP: Set eth3 to backup
Mar 15 01:22:31 OpenMPTCProuter user.notice MPTCP: Set eth4 to on
Mar 15 01:22:32 OpenMPTCProuter user.notice MPTCP: Set eth5 to on
Mar 15 01:22:32 OpenMPTCProuter user.notice MPTCP: Disabling MPTCP on interface ifb4tun0 not found in enabled multipath list
Mar 15 01:22:34 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Mar 15 01:23:00 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
Mar 15 01:23:02 OpenMPTCProuter user.notice firewall.omr-bypass: reload omr-bypass rules
Mar 15 01:23:02 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Mar 15 01:23:02 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running

vempire-ghost, 2024-03-15:
I found that these two lines are causing some kind of looping when connecting to the VPS with both backup and master active at the same time, and during this looping, OMR keeps connecting and disconnecting to each of the VPS and generates all this data traffic. I removed these two lines, and this problem no longer occurred, nor was there any more data flow.

In the file omr-tracker-server
171 set openmptcprouter.${name}.get_config='1'
322 set openmptcprouter.${name}.get_config='1'

Ysurac, 2024-03-15:
These are lines are used when there is master/backup server switch, but they shouldn't be need with latest changes on synchronization.

vempire-ghost, 2024-03-15:
> These are lines are used when there is master/backup server switch, but they shouldn't be need with latest changes on synchronization.

What are the latest changes in the synchronizations?

closed 2024-06-19 by github-actions[bot]