Ysurac/openmptcprouter #3212: StarLink compatibility ?

JeedHome44, 2024-03-17:
Hello,

Can we add Starlink IP 192.168.1.1  in OMR to add it with ethernet adaptator and bypass mode ?

Thank you

vinc32, 2024-03-17:
Yes, but no need to add the ip. just let it as DHCP.

JeedHome44, 2024-03-17:
> Yes, but no need to add the ip. just let it as DHCP.

I manage DHCP with OMR.

I don't want StarLink to do DHCP. that's why I want to put it in by-pass mode

Il faut donc bien mettre l'IP du routeur StarLink sur OMR pour le mettre en MacVLan !?

JeedHome44, 2024-03-18:
@Ysurac 

Hello,

I have a bypass mode in my StarLink routeur.

Can I only change the IP of OMR to 192.168.101.1 for example and leave all devices in 192.168.100.X ?

There will only be OMR in DHCP on the LAN since StarLink will be in bypass !

And like that, 192.168.100.1 will be for the StarLink web interface ! is it ok like that ?

JeedHome44, 2024-03-18:
[image]

vinc32, 2024-03-18:
You just need to let the WAN Interface for Starlink in DHCP, the interface will get a Public or DNAT IP (depending on contract) and you're fine.
OMR still acting as DHCP for Clients.

Ysurac, 2024-03-18:
If you have as a router something with different ethernet card yes, you dedicate one for starlink, it's easy. The problem is more when you use a Raspberry or anything with only one port.

JeedHome44, 2024-03-18:
> You just need to let the WAN Interface for Starlink in DHCP, the interface will get a Public or DNAT IP (depending on contract) and you're fine.
> 
> OMR still acting as DHCP for Clients.

we cannot leave 2 DHCPs on a single LAN !

I ordered a StarLink ethernet adapter. it will therefore be on the same LAN as the Raspberry OMR.

I just want to know if I can keep all my devices in 192.168.100.X and just change the Raspberry OMR to 192.168.101.1 for example so as not to conflict with the Starlink web interface which is necessarily in 192.168.100.1

JeedHome44, 2024-03-18:
> If you have as a router something with different ethernet card yes, you dedicate one for starlink, it's easy. The problem is more when you use a Raspberry or anything with only one port.
> 
> 

I only have a Raspberry Pi 4B with OMR which manages all the DHCP in the house

dfadev, 2024-03-20:
[image]

[image]
[image]

[image]

Static IP with Starlink works in non-bypass "normal" mode.  OMR can service everything through one NIC and devices can be on the same unmanaged switch as long as you use different IP networks and have only one DHCP server.  In my setup the OMR DHCP server serves addresses to clients connecting to the wifi AP.

JeedHome44, 2024-03-20:
I have difficulty understanding English.

I just want to know if I can keep the IP 192.168.100.1 for OMR if I put StarLink on bypass!

It's going to take me a long time to change all the fixed IPs on my DHCP and especially on Jeedom and Home Assistant with all the devices!

So it would be much simpler to put the StarLink on bypass and put it in MacVLan on OMR?

dfadev, 2024-03-20:
I think the problem will be your devices will broadcast a DHCP request and both OMR and Starlink will try to answer.  You will need a managed switch with VLAN tagging to control which DHCP server replies.

JeedHome44, 2024-03-20:
in by-pass mode, StarLink no longer does DHCP!

dfadev, 2024-03-20:
How else would you get an address?

JeedHome44, 2024-03-20:
I don't understand ! my 4G router has its DHCP disabled and is on the same LAN as my Free box which also has its DHCP disabled and I manage to have both by pointing the MacVLan to their respective IP!

It’s OMR that does the DHCP for my entire LAN network!

dfadev, 2024-03-20:
https://www.reddit.com/r/Starlink/comments/158qlzp/using_your_own_router_with_starlink_bypass_mode/

you will need a managed switch.

JeedHome44, 2024-03-20:
I ordered the Starlink ethernet adapter to have an ethernet port 😉

JeedHome44, 2024-03-23:
I think I understood ! I need to get a router to put StarLink in by-pass mode on it, deactivate DHCP and make a routing rule so I don't have to change the OMR IP but still have access to the interface StarLink web. Is it possible like that?

closed 2024-03-29 by JeedHome44