Ysurac/openmptcprouter #3214: Lots of ss-server connection reset errors on VPS

adamb, 2024-03-18:
## Current Behavior

I noticed some lag on my home network and started looking at the VPS server and found that 
I'm seeing a lot of this on my VPS server.  

> Mar 18 12:53:18 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 12:53:26 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 12:53:53 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 12:55:00 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 12:55:00 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 12:55:28 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 12:55:28 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 12:58:17 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 12:59:33 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 13:02:11 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 13:02:43 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 13:07:12 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 13:09:23 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 13:11:09 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 13:11:12 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 13:11:35 myvps.xyz ss-server[1967644]: remote recv: Connection reset 
> Mar 18 13:11:38 myvps.xyz ss-server[1967644]: remote recv: Connection reset 

Note myvps.xyz is not my real vps.  I changed the name to protect the innocent... 

Is this normal?  If not, any idea how to track down what's happening?

The server doesn't have a lot of load on it.  These errors seem to be happening a lot, every minute or so for days.  

I'm using a TMobile connection for my master (wan1) and Starlink for wan2... 

[image]

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  -  OpenMPTCProuter VPS provider: Linode
  - 
  - OpenMPTCProuter platform: RPI 4

Ysurac, 2024-03-18:
This error mean the server side close the TCP connection with a RESET. This can be due to a timeout or some network issue on VPS side.
I would begin by a reboot of VPS and router side.

alchool, 2024-04-11:
is proxy active? Can you show  "ss -s" resoult?

adamb, 2024-04-11:
root@OpenMPTCProuter:~# ss -s
Total: 532
TCP:   1056 (estab 649, closed 538, orphaned 0, timewait 387)

Transport Total     IP        IPv6
RAW	  4         2         2
UDP	  16        11        5
TCP	  518       434       84
INET	  538       447       91
FRAG	  0         0         0

closed 2024-07-16 by github-actions[bot]