Ysurac/openmptcprouter #3216: Addressing MPTCP Tunnel Challenges: Raspberry Pi to Google Cloud VM Connectivity

Raquel1L, 2024-03-20:
## Expected Behavior
I am using OpenMPTCPRouter to create an MPTCP tunnel connecting the gateway (Raspberry Pi) to a Debian 11 Google Cloud VM (with the requested ports open). Connected to the RPI is a computer that sends all packets to it. In this case, I am using two TCP subflows, aggregating WiFi and 5G. But for now just testing with just one path.

## Current Behavior
In an experiment, I ran curl -I https://www.example.com in the VM terminal while doing tcpdump, and the capture result was as follows: 

[image]

As you can see in the figure, my problem is that when the Google Cloud VM receives MPTCP packets, it sends a TCP Reset packet (Additionally, aren't all packets sent by RPI supposed to be MPTCP?).

Finally, I leave you with the following figure, which shows the general state of the system.

[image]

## Specifications
  - OpenMPTCProuter version: OpenMPTCProuter v0.60 beta 2
  - OpenMPTCProuter VPS version: 5.10.0-28-cloud-amd64
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: RPI4

Ysurac, 2024-03-20:
Your VPS have a 5.10.0 kernel. It's not supported. You need to install the VPS part. With the 0.59.1 version you are using, this will install a Linux kernel 5.4.

Raquel1L, 2024-03-20:
Thanks for your quick reply! 

To install the VPS I ran these two commands in the GCP VM terminal:
apt-get update && apt-get upgrade
wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | sh

Your suggestion is that I change the debian-x86_64.sh file so that I can install the Linux patched kernel 5.4?

Ysurac, 2024-03-20:
The correctly executed ? (you get all keys at the end)
You rebooted the VPS after execution ?

Raquel1L, 2024-03-20:
Yes, after running the second command, I got these keys:

[image]

Do you think the reboot will change the version problem?

Ysurac, 2024-03-20:
Reboot is need to use new kernel. If it was not already done, it's the problem.

Raquel1L, 2024-03-20:
The problem with rebooting the machine is that after doing so I lose the ssh connection to it.
(ssh: connect to host x.x.x.x port 65222: Operation timed out)

Ysurac, 2024-03-20:
Check that port 65222 is correctly open in GCP firewall.
If it's the case, try to use GCP VM console access.

Raquel1L, 2024-03-20:
Thank you very much for your help!!!

[image]

I'm going to repeat the experiment now and see if everything works!

Raquel1L, 2024-03-20:
Regarding the server problem, everything has been resolved, thank you. However, the multipath is still not working. The general state of the system is as follows:

[image]

Do you have any idea what could be causing this?

Ysurac, 2024-03-20:
A box can block MPTCP or the ISP can block it.
It's often a good idea to try aggregation (so with more than one connection) even if there is the warning.
You can also update to lastest snapshot with 6.1 kernel and check if it's working or not.
If nothing is working, you will have to use MPTCP over VPN checkbox, that use MPTCP over Wireguard.

closed 2024-06-24 by github-actions[bot]