Ysurac/openmptcprouter #3217: Kernel panic - not syncing

Kalu01, 2024-03-20:
## Expected Behavior
The system should boot successfully without any errors using the OpenMPTCProuter version openmptcprouter-v0.60rc1-6.1-r0+24843-acf40c022e-x86-64-generic-ext4-combined-efi.img.gz.

## Current Behavior
During the boot process, we encountered a Kernel panic - not syncing: Fatal exception error. However, when using an older kernel version openmptcprouter-v0.60rc1-5.4-r0+16862-170d9e447d-x86-64-generic-ext4-combined-efi.img.gz, the system was able to boot successfully.

We are also experiencing issues with NVMe M.2 drives. The boot process fails when attempting to boot from an NVMe drive.

## Specifications
OpenMPTCProuter version: openmptcprouter-v0.60rc1-6.1-r0+24843-acf40c022e-x86-64-generic-ext4-combined-efi.img.gz and openmptcprouter-v0.60rc1-5.4-r0+16862-170d9e447d-x86-64-generic-ext4-combined-efi.img.gz
OpenMPTCProuter platform: x86_64
Hardware used:

Device: HSIPC Pentium Gold 7505 Firewall Micro Appliance, Mini PC, Nano PC, Router PC with 6 RJ45 2500M (16G 256G), AES-NI, HDMI, USB3.0, Console Port for Network Security Appliance
Graphics Card Type: HD
Brand Name: HSIPC
Processor Main Frequency: Dual Core Four Threads 2.0GHz (Turbo 3.5GHz)
Use: Commercial/Industrial
Memory: 2*DDR4 3200M Max. 64GB
Processor Brand: Intel
Processor Model: Intel formerly Tiger Lake Pentium Gold
Model Number: HS-7505
CPU: Intel Pentium Gold 7505
Chipset: Intel Chipset
Storage: 1NVME 2280 1SATA, 1*Mini PCIE (Support Wifi)
RAM: 16G
Hard Drive: 256G
Video: HDMI+DP
Network: 6*Intel i225/i226 2500M
USB: 4USB, 1Type C

We intend to use the system with 3 StarLinks.

Ysurac, 2024-03-20:
This device only support UEFI ?
I would need full log error.

Kalu01, 2024-03-21:
The device support Legacy and UEFI

Logfile:
I hope the log file is complete. We could only create it using image to text, since the device doesn't have a serial port.

Booting `OpenMPTCProuter'

[4.670458] Console: switching to colour frame buffer device 100x37
[4.677384] fb0: EFI UGA frame buffer device
[4.681724] hid: raw HID events driver (C) Jiri Kosina
[4.686900] hu_umbus: registering driver hid hyperu

[4.691819] usbcore: registered new interface driver usbhid 
[4.697420] usbhid: USB HID core driver
[4.701341] hu_umbus: registering driver hu_balloon

[4.706705] NET: Registered PF_INET6 protocol family

[4.711876] Segment Routing with IPv6
[4.715575] In-situ OAM (IOAM) with IPv6

[4.719549] NET: Registered PF_PACKET protocol family
[4.724637] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[4.7381101 8021q: 802.1Q VLAN Support v1.8
[4.742818] NET: Registered PF_VSOCK protocol family
[4.7485251 microcode: sig=0x806c1, pf=0x80, revision=0x8a
[4.754565] microcode: Microcode Update Driver: v2.2.
[4.7545701 IPI shorthand broadcast: enabled
[4.764854] AVXZ version of gcm_enc/dec engaged.
[4.769961] AES CTR mode by8 optimization enabled

[4.775327] sched_clock: Marking stable (3606718032, 1168595772)-> (4838047024, -62733220) 
[4.784669] Waiting for root device PARTUUID=Zaf91ce4-288b-da 12-e3b3-8a8a91b22c02... 
[4.868606] usb 3-4: new low-speed USB device number 2 using xhci_hcd
[5.077870] input: Microsoft Wired Keyboard 600 as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4: 1.0/0003:045E:07F8.0001/input/input3
[5.158633] hid-generic 0003:045E:07F8.0001: input, hidraw0: USB HID v1.11 Keyboard [Microsoft Wir ed Keyboard 600] on usb-0000:00:14.0-4/inputo
[5.174854] input: Microsoft Wired Keyboard 600 Consumer Control as /devices/pci0000:00/0000:00:1 4.0/usb3/3-4/3-4:1.1/0003:045E:07F8.0002/input/input4
[5.228655] usb 4-2: new SuperSpeed USB device number 2 using xhci_hcd
[5.268635] input: Microsoft Wired Keyboard 600 System Control as /devices/pci0000:00/0000:00:14. 0/usb3/3-4/3-4:1.1/0003:045E:07F8.0002/input/input5
[5.283030] hid-generic 0003:045E:07F8.0002: input, hidraw1: USB HID v1.11 Device [Microsoft Wired Keyboard 600] on usb-0000:00:14.0-4/input1 5.298247] usb-storage 4-2:1.0: USB Mass Storage device detected
[5.304944] scsi host2: usb-storage 4-2:1.0

[5.2830301 hid-generic 0003:045E:07F8.0002: input, hidraw1: USB HID v1.11 Device [Microsoft Wired Keyboard 6001 on usb-0000:00:14.0-4/input1
[5.298247] usb-storage 4-2:1.0: USB Mass Storage device detected

[5.304944] scsi host2: usb-storage 4-2:1.0 6.3194631 scsi 2:0:0:0: Direct-Access 

[6.328404] sd 2:0:0:0: [sdal 120176640 512-byte logical blocks: (61.5 GB/57.3 GiB) 
[6.337594] sd 2:0:0:0: [sda] Write Protect is off
SanDisk Ultra USB 3.0

[6.3433311 sd 2:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or
[6.3670801 sda: sda1 sda2 sda3
[6.370995] sd 2:0:0:0: [sda] Attached SCSI removable disk

[6.386426] EXT4-fs (sda2): (sda2): INFO: recovery required on readonly filesystem 
[6.394023] EXT4-fs (sda2): write access will be enabled during recovery 
[6.403353] EXT4-fs (sda2): recovery complete

[6.433087] Freeing unused kernel image (initmem) memory: 1228K 
[6.508604] Write protecting the kernel read-only data: 18432k
[6.410851] EXT4-fs (sda2): mounted filesystem with ordered data mode. Quota mode: disabled. 
[6.419867] VFS: Mounted root (ext4 filesystem) readonly on device 8:2. 
[6.429314] deutmpfs: mounted
[6.515851] Freeing unused kernel image (text/rodata gap) memory: 2044K 
[6.523205] Freeing unused kernel image (rodata/data gap) memory: 844K 
[6.530296] Run /sbin/init as init process
[6.776513] init: Console is alive
[6.786839] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[6.803573] pps_core: LinuxPPS API ver. 1 registered
[... 71 lines cut ...]
tunne14 ip_tunnel hfcpci hfcmulti veth tun ovpn_dco_v2 udp_tunnel ip6_udp_tunnel mISDN dsp 11oip MISDN_core xfrm_user

[12.81206] xfrm_ipcomp af key xfrm_algo autofs4 br2684 atm nls_utf8 zram zsmalloc pcs_xpcs nats emi macsec ena sha512_ssse3 
sha512_generic sha256_ssse3 sha256_generic sha1_ssse3 sha1_generic seqiu jitterentropy_rng_drbg md5 kpp hmac echainiu des_generic 
libdes deflate zlib_deflate cmac cbc authe ncesn authenc crypto_acompress nf_nat_sip nf_nat nf_conntrack_sip nls_iso8859_1 nls_cp437 
sdhci_pltf m xhci_plat_hcd fsl_mph_dr_of ehci_platform ehci_fsl igb gpio_button_hotplug ufat fat button_hotplug tg3 b44 ssb ptp realtek 
broadcom bcm_phy_lib pps_core phylink mii libphy
[12.957107] ---[ end trace 0000000000000000 1---

[12.962343] RIP: 0010:exc_control_protection+0xbd/0xc0
12.9680901 Code: fb ff 45 31 c9 49 89 d8 b9 09 00 00 00 48 8b 93 80 00 00 00 be fc 00 00 00 48 c ? c? a2 ad 01 82 e8 98 6f 70 ff e9 6e 
ff ff ff <0f> Ob 90 66 Of 1f 00 55 48 89 e5 41 54 53 48 89 fb f6 87 88 00 00

[12.988686] RSP: 0018:ffffc90000443bb0 EFLAGS: 00010002
[12.9945551 RAX: 0000000000000035 RBX: ffffc90000443bd8 RCX: c0000000ffffefff 
[13.002344] RDX: 00000000000002fb RSI: 00000000ffffefff RDI: 0000000000000001 
[13.010138] RBP: ffffc90000443bc8 R08: 0000000000000000 R09: ffffc90000443a48 
[13.017906] R10: 0000000000000003 R11: ffffffff822b4ee8 R12: 0000000000000003 
[13.025656] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[13.033414] FS: 00007fe6e0f71b48 (0000) GS: ffff88849fb00000 (0000) kn1GS:0000000000000000 
[13.042102] CS: 0010 DS: 0000 ES: 0000 CRO: 0000000080050033
[13.0484551 CR2: ffffffffa023efd6 CR3: 00000001046e4005 CR4: 0000000000b70ee0
[13.056200] Kernel panic - not syncing: Fatal exception
[13.062074] Kernel Offset: disabled
[13.066187] Rebooting in 1 seconds..

Ysurac, 2024-03-22:
As it support legacy, did you try a not EFi image or squash image ?

vinc32, 2024-03-22:
Same issue with EFI Image on N100 system.

patimages, 2024-03-22:
I'm also using N100 (trigkey) : not working on RC1 5.4 EFI (driver issues I would guess) but working on 6.1 EFI (no panic, boot properly and works well). There is one issue though : the VPN goes down upon external access of the NAS (Plex) and I face lots of issues when connecting remotely. Was not the case on an older NUC on 0.59 5.4.

vinc32, 2024-03-22:
Thx for the tip! @patimages 

0.59.1-5.4 -> Ethernet i226-v not working
0.60rc1-5.4 -> Ethernet working & Booting
0.60rc1-6.1 -> Kernel Panic

patimages, 2024-03-22:
I tested again and actually confirm your results on N100 : 
0.59.1-5.4 -> Not working 
0.60rc1-5.4 -> Ethernet working & Booting -> but no external access (VPN going down upon attempt). 
0.60rc1-6.1 -> Kernel Panic

vinc32, 2024-03-23:
Your bypass is working? I've restored from backup and nothing is working - maybe I'll have to change something in bypass file?

Ysurac, 2024-03-23:
@vinc32 for OMR-ByPass, test snapshot first (there is some fixes) and if not working, open a new issue.

For the 0.60rc1-6.1, did you try both ext4 and squashfs images ?
I'm not able to reproduce the issue on an enabled UEFI VM.
Can you give me the result of ```dmesg``` on a booting 0.60rc1 ?

vinc32, 2024-03-23:
dmesg from **0.60rc1-5.4**

root@OpenMPTCProuter:~# dmesg
[    0.000000] Linux version 5.4.238 (runner@fv-az618-69) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+16862-170d9e447d)) #0 SMP Sat Jul 29 19:00:07 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=d8c84620-7726-b988-43a2-7c2782349e02 rootwait console=tty0 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x200: 'Protection Keys User registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[9]:  832, xstate_sizes[9]:    8
[    0.000000] x86/fpu: Enabled xstate features 0x207, context size is 840 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000450fbfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000450fc000-0x00000000450fcfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000450fd000-0x0000000045132fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000045133000-0x0000000045133fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000045134000-0x0000000071daefff] usable
[    0.000000] BIOS-e820: [mem 0x0000000071daf000-0x0000000074eaefff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000074eaf000-0x0000000074f91fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000074f92000-0x0000000075051fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000075052000-0x0000000075f65fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000075f66000-0x0000000075ffefff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000075fff000-0x0000000075ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000076000000-0x0000000079ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aa00000-0x000000007abfffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x00000000803fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fbfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.80 by American Megatrends
[    0.000000] efi:  ACPI=0x7502e000  ACPI 2.0=0x7502e014  TPMFinalLog=0x74ffd000  SMBIOS=0x75c39000  SMBIOS 3.0=0x75c38000  MEMATTR=0x6c30b118  ESRT=0x6e2c6918 
[    0.000000] SMBIOS 3.5.0 present.
[    0.000000] DMI: Default string Default string/Default string, BIOS 5.27 11/24/2023
[    0.000000] tsc: Detected 800.000 MHz processor
[    0.000132] tsc: Detected 806.400 MHz TSC
[    0.000132] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000134] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000139] last_pfn = 0x27fc00 max_arch_pfn = 0x400000000
[    0.000141] MTRR default type: write-back
[    0.000142] MTRR fixed ranges enabled:
[    0.000142]   00000-9FFFF write-back
[    0.000143]   A0000-BFFFF uncachable
[    0.000143]   C0000-FFFFF write-protect
[    0.000144] MTRR variable ranges enabled:
[    0.000145]   0 base 0080000000 mask 7F80000000 uncachable
[    0.000145]   1 base 007C000000 mask 7FFC000000 uncachable
[    0.000146]   2 base 2000000000 mask 6000000000 uncachable
[    0.000146]   3 base 1000000000 mask 7000000000 uncachable
[    0.000147]   4 base 0800000000 mask 7800000000 uncachable
[    0.000147]   5 base 0400000000 mask 7C00000000 uncachable
[    0.000148]   6 base 4000000000 mask 4000000000 uncachable
[    0.000148]   7 base 027FC00000 mask 7FFFC00000 uncachable
[    0.000149]   8 disabled
[    0.000149]   9 disabled
[    0.000238] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000415] last_pfn = 0x76000 max_arch_pfn = 0x400000000
[    0.008402] esrt: Reserving ESRT space from 0x000000006e2c6918 to 0x000000006e2c69f0.
[    0.008406] e820: update [mem 0x6e2c6000-0x6e2c6fff] usable ==> reserved
[    0.008418] Using GB pages for direct mapping
[... 839 lines cut ...]
[  797.867168] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[  799.197025] igc 0000:01:00.0 eth0: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[  799.197194] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  800.566962] igc 0000:03:00.0 eth2: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[  815.566400] igc 0000:03:00.0 eth2: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[  930.581106] pppoe-wan1: renamed from ppp0
[ 1904.285004] TCP: too many orphaned sockets
[ 1904.285013] MPTCP: out of memory: force closing socket
[ 1904.285075] TCP: too many orphaned sockets
[... cut ...]
[ 1912.821501] net_ratelimit: 3740 callbacks suppressed
[ 1912.821503] TCP: too many orphaned sockets
[ 1926.797817] traps: ipset[4852] trap invalid opcode ip:7f1020b66252 sp:7ffe70a9b810 error:0 in libipset.so.13.1.0[7f1020b5e000+a000]
[ 1927.799258] traps: ipset[4871] trap invalid opcode ip:7fcb30254252 sp:7fff3d374db0 error:0 in libipset.so.13.1.0[7fcb3024c000+a000]

dmesg from **0.60rc1-6.1**

only one picture - reboots instant after kernel panic so difficult. made a video if you need it.
[image]

QatarMo, 2024-03-23:
> @vinc32 for OMR-ByPass, test snapshot first (there is some fixes) and if not working, open a new issue.
> 
whats the command to build useing snapshot please

vinc32, 2024-03-23:
https://snapshots.openmptcprouter.com

Download it there

kevin3305, 2024-04-11:
hi intel N100 here with v0.60 6.1 stable same problem kernel panic then reboot, did you find a solution?
THANKS

patimages, 2024-04-11:
Works great on rc2 5.4.

Ysurac, 2024-04-11:
What is the device you use ?
In a few hours (it's slow to compile on Github Actions...), can you test OpenMPTCProuter 0.61 snapshot based on kernel 6.6 ? This will help to check if the issue is still here...
If it's the case, I will buy a N100 based device.

kevin3305, 2024-04-11:
here is the pc: https://fr.aliexpress.com/item/1005006198199636.html?pdp_npi=4%40dis%21EUR%21%E2%82%AC%20242%2C07%21%E2%82%AC%20169%2C45%21%21%21254.11%21177.88%21%40211b815c17128364491323957e452b%2112000036234094472%21sh01%21FR%210%21&spm=a2g0o.store_pc_home.productList_2007924639156.1005006198199636

where to find the snapshot in 6.6?
THANKS

Ysurac, 2024-04-11:
@kevin3305 as I said, it's still compiling. Then it will be available at https://snapshots.openmptcprouter.com/

kevin3305, 2024-04-11:
ok thank you as soon as available I will get back to you.

vinc32, 2024-04-11:
if it's finished by tomorrow I'll test it ;)

kevin3305, 2024-04-11:
hi I no longer tested 6.6 Snapshot same problem reboot loop kernel panic.

Ysurac, 2024-04-18:
I received my N100 device, I will test and try to fix ASAP

vinc32, 2024-04-24:
> I received my N100 device, I will test and try to fix ASAP

already any update on those devices?

patimages, 2024-04-25:
> Should be fixed now. Snapshot is compiling...

Should we use 5.4, 6.1 or 6.6 or all are supposed to work ? Thanks again !

Ysurac, 2024-04-25:
All are supposed to work.

kevin3305, 2024-04-25:
hello I tested 6.1, 6.6 update on April 24 still a kernel panic reboot.
Thank you for this work

kevin3305, 2024-04-25:
[image]
Here is the error received before the reboot

Ysurac, 2024-04-25:
I didn't removed all modules that make N100 crash, it's now done.
A new snapshot is compiling.

kevin3305, 2024-04-25:
Ok thanks for your feedback

vinc32, 2024-04-26:
Still not working -> tried squashfs and ext4 images (EFI)

openmptcprouter-v0.61-snapshot-6.6-r0+26018-84a48ce400-x86-64-generic-squashfs-combined-efi
openmptcprouter-v0.61-snapshot-6.6-r0+26018-84a48ce400-x86-64-generic-ext4-combined-efi

patimages, 2024-04-26:
Hi, 
To the people having issues : I also have an N100 and all is working great with 5.4 (openmptcprouter-v0.60rc1-5.4-r0+16862-170d9e447d-x86-64-generic-squashfs-combined-efi). I did not dare to try newer versions as the family would heavily complain if it were not to work. 
Out of curiosity, what is the rational of using 6.1 or 6.6 ? To be future-proofed or there is another reason ?

Ysurac, 2024-04-26:
@vinc32 I will test image, maybe I forgot to remove another module...

@patimages 
Packages of 5.4 are no more updated (can have security issues), kernel will stay the same with no changes, MPTCPv0 by default, ...

6.1 or 6.6 use MPTCPv1, packages are updated, new platforms will only use 6.x kernels, some new features will only be on 6.x version.

It's like using Debian 9 vs Debian 12 or Windows 98 vs Windows 11: both are working, but old versions are EOL.

patimages, 2024-04-26:
> @patimages Packages of 5.4 are no more updated (can have security issues), kernel will stay the same with no changes, MPTCPv0 by default, ...
>
> 6.1 or 6.6 use MPTCPv1, packages are updated, new [...]

Makes perfect sense indeed ! Guess, I ll take a chance with the snapshot to come and give a try to 6.1 and 6.6. Thanks !

kevin3305, 2024-04-27:
hi thank you for everything it works very well now (openmptcprouter-v0.61-snapshot-6.6-r0+26018-84a48ce400-x86-64-generic-squashfs-combined-efi)

vinc32, 2024-04-27:
@kevin3305 any special settings in bios?
I'm not able to boot it and hardware seems to be similar to yours.

patimages, 2024-04-27:
I took the plunge and installed openmptcprouter-v0.61-snapshot-6.6-r0+26018-84a48ce400-x86-64-generic-squashfs-combined-efi. All is fine but I lost all external accesses (IP_sever:port as defined in port forward). Any idea how to solve it ?

kevin3305, 2024-04-27:
@vinc32 default bios parameter, I did a clear cmos

patimages, 2024-04-27:
> I took the plunge and installed openmptcprouter-v0.61-snapshot-6.6-r0+26018-84a48ce400-x86-64-generic-squashfs-combined-efi. All is fine but I lost all external accesses (IP_sever:port as defined in port forward). Any idea how to solve it ?

back to 5.4 and matching VPS as I really need external access.

patimages, 2024-04-28:
Extra question : I see that most of my traffic goes through proxy rather than VPN : is it normal ? It is as safe in front of prying eyes ? 
[image]

vinc32, 2024-04-28:
Same here :) but should be like that. Only udp goes through vpn normally.

[image]

patimages, 2024-04-28:
good to know ! However, I was under the impression that VPN was safer than proxy, am I wrong ?

vinc32, 2024-04-28:
You're correct but also slows down a little bit. I'm not sure if all traffic can be routed through vpn with mptcp. 

@Ysurac you're the pro :)

patimages, 2024-04-28:
@Ysurac you're the pro, indeed :) !

Ysurac, 2024-04-28:
Proxy is encrypted, and Shadowsocks-Rust use latest release of Shadowsocks protocol that is safer.

patimages, 2024-04-28:
Sounds perfect ! However, I am using (historically) Shadowsocks. I just tried "Shadowsocks rust 2022" (AES 256 CGM) and get an error : "Can't get public IP from shadowsocks rust".

vinc32, 2024-04-28:
Same issue

Ysurac, 2024-04-28:
Port range 65000-65535 is used by OpenMPTCProuter.
Shadowsocks-Rust is multithreaded, so it offer a better result when the platform have more than one core available.

FabrizioBocci, 2024-05-14:
I have the same Kernel panic problem with N100 device. I tried to install UEFI version from https://www.openmptcprouter.com. I read above the using new kernel 6.6 this problem should be fixed but in the download section only 6.1 version is available. Where can i find 6.6 ?
I tried installing 5.4 and this one works.

Ysurac, 2024-05-14:
As in previous comments said: https://snapshots.openmptcprouter.com/

FabrizioBocci, 2024-05-14:
Thanks Ysurac, this snapshot works!!!

profclems, 2024-05-16:
@Ysurac I tried the 6.6 snapshot and it worked on the mini pc but couldn't detect my ethernet ports. I had to revert back to 5.4

Ysurac, 2024-05-16:
What is the chipset/driver used for your ethernet port ?

profclems, 2024-05-16:
▶ lspci -kvnn | sed -n '/Ethernet/,/^$/ p'
```
01:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)
	Subsystem: Realtek Semiconductor Co., Ltd. Device [10ec:0123]
	Flags: bus master, fast devsel, latency 0, IRQ 18
	I/O ports at 5000 [size=256]
	Memory at 80604000 (64-bit, non-prefetchable) [size=4K]
	Memory at 80600000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 01
	Capabilities: [b0] MSI-X: Enable+ Count=4 Masked-
	Kernel driver in use: r8169

03:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)
	Subsystem: Realtek Semiconductor Co., Ltd. Device [10ec:0123]
	Flags: bus master, fast devsel, latency 0, IRQ 17
	I/O ports at 3000 [size=256]
	Memory at 80404000 (64-bit, non-prefetchable) [size=4K]
	Memory at 80400000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 01
	Capabilities: [b0] MSI-X: Enable+ Count=4 Masked-
	Kernel driver in use: r8169
```

cc: @Ysurac

Ysurac, 2024-05-17:
@profclems Thanks  I'm working on this.

profclems, 2024-05-23:
The new snapshot works now [openmptcprouter-v0.61-snapshot-6.6-r0+26308-4341901f05-x86-64-generic-ext4-combined-efi.img.gz](https://snapshots.openmptcprouter.com/6.6/x86_64/targets/x86/64/openmptcprouter-v0.61-snapshot-6.6-r0+26308-4341901f05-x86-64-generic-ext4-combined-efi.img.gz)

Ysurac, 2024-05-23:
Thanks for the feedback, I added the driver but I didn't have the time to test...

profclems, 2024-05-24:
Thanks @Ysurac.

closed 2024-08-28 by github-actions[bot]