Ysurac/openmptcprouter #3219: MPTCP kernel fails to build v0.60RC1 Kernel 6.1

chriskinal, 2024-03-23:
## Expected Behavior
The VPC install script should create a MPTCP kernel.

## Current Behavior
MPTCP kernel fails to build.

## Possible Solution
Not sure.

## Steps to Reproduce the Problem

  1. Create a Debian Linux 11.6 or 12.5 Amazon Lightsail server.
  2. Run VPS install script.
  3. Reboot server and check running kernel.
  4. Check for kernel image: dpkg --list | grep linux-image
  5. No mptcp kernels found.

## Context (Environment)
Created a new Amazon light server and executed the VPS install script. Upon reboot no MPTCP kernel was shown or found on the system. I tried the install with a Debain 11.9 Bullseye and a Debian 12.5 Bookworm as the base Debian image.

## Specifications

  - OpenMPTCProuter version: v0.60RC1 Kernel 6.1
  - OpenMPTCProuter VPS version: Not sure. Whatever was downloaded by: wget -O - http://www.openmptcprouter.com/server-rc/debian.sh | UPSTREAM6="yes" sh
  - OpenMPTCProuter VPS provider: Amazon Lightsail
  - OpenMPTCProuter platform: x86-64
  - Country: US

**** Log of the install is too big for Github. Pastebin keeps failing with a 404 not found error when I submit the log. No other option but to attach a file to the issue.******

[omptcp_0.60RC1_lightsail.txt](https://github.com/Ysurac/openmptcprouter/files/14731564/omptcp_0.60RC1_lightsail.txt)

Ysurac, 2024-03-23:
With 6.1 script, default Debian kernel is used, MPTCP is enabled by default.

chriskinal, 2024-03-23:
OK. The end of the installer script has the line about rebooting and checking for an MPTCP kernel by doing 'uname -a' to ensure MPTCP is installed. Perhaps removing that line from the installer script for 6.1. One can look in 'sysctl' at net. mptcp.enabled . I found that out in a Google search.

chriskinal, 2024-03-23:
It is indeed enabled in the kernel. net.mptcp.enabled = 1.

closed 2024-03-23 by chriskinal

Ysurac, 2024-03-23:
Thanks for the feedback, I will remove the line.