Ysurac/openmptcprouter #3220: VPS Install Script Fails at Install "omr-vps-admin".

chriskinal, 2024-03-23:
## Expected Behavior
Install script runs to completion and install the OMPTCP VPS server software.
## Current Behavior
Install script fails during Python module install.

## Possible Solution
Not sure.

## Steps to Reproduce the Problem

  1. Create a Debian Linux 11.6 or 12.5 Amazon Lightsail server.
  2. Run VPS install script.

## Context (Environment)
Install script fails with an incorrect version number but I can't version of what?

Created new Lightsail instance with both Debian 11.6 and 12.5. The VPS script fails during Python module install.

## Specifications

  - OpenMPTCProuter version: v0.60RC1 Kernel 5.4
  - OpenMPTCProuter VPS version: Whatever is downloaded by: wget -O - http://www.openmptcprouter.com/server-rc/debian.sh | sh
  - OpenMPTCProuter VPS provider: Amazon
  - OpenMPTCProuter platform: x86_64
  - Country: US

admin@ip-172-26-13-54:~$ sudo su -
root@ip-172-26-13-54:~# wget -O - http://www.openmptcprouter.com/server-rc/debian.sh | sh
--2024-03-23 13:16:17--  http://www.openmptcprouter.com/server-rc/debian.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 37.19.207.34
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|37.19.207.34|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 88768 (87K) [application/octet-stream]
Saving to: ‘STDOUT’

-                                       0%[                                                                         ]       0  --.-KB/s               Check user...
Check Linux version...
Check architecture...
Check virtualized environment
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Update repo.openmptcprouter.com key
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
--2024-03-23 13:16:17--  https://repo.openmptcprouter.com/openmptcprouter.gpg.key
Resolving repo.openmptcprouter.com (repo.openmptcprouter.com)... 37.19.207.34
Connecting to repo.openmptcprouter.com (repo.openmptcprouter.com)|37.19.207.34|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1241 (1.2K) [application/octet-stream]
Saving to: '/etc/apt/trusted.gpg.d/openmptcprouter.gpg'

/etc/apt/trusted.gpg.d/openmptcproute 100%[========================================================================>]   1.21K  --.-KB/s    in 0s      

2024-03-23 13:16:18 (27.3 MB/s) - '/etc/apt/trusted.gpg.d/openmptcprouter.gpg' saved [1241/1241]

Remove lock and update packages list...
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian buster-backports InRelease
Get:3 file:/etc/apt/mirrors/debian.list Mirrorlist [38 B]
Get:7 file:/etc/apt/mirrors/debian-security.list Mirrorlist [47 B]
Hit:8 https://repo.openmptcprouter.com next InRelease    
Hit:9 https://repo.openmptcprouter.com buster InRelease  
Hit:4 https://cdn-aws.deb.debian.org/debian bookworm InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian bookworm-updates InRelease
Hit:6 https://cdn-aws.deb.debian.org/debian bookworm-backports InRelease
Hit:10 https://cdn-aws.deb.debian.org/debian-security bookworm-security InRelease
Reading package lists... Done
Install apt-transport-https, gnupg and openssh-server...
Reading package lists... Done
Building dependency tree... Done
[... 127 lines cut ...]
python3-psutil is already the newest version (5.9.4-1+b1).
python3-requests is already the newest version (2.28.1+dfsg-1).
pwgen is already the newest version (2.08-2).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
-- pip3 install needed python modules
If you see any error here, I really don't care: it's about a module not used for home users
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
netjsonconfig 1.0.2 requires jsonschema~=4.4.0, but you have jsonschema 4.21.1 which is incompatible.
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
netjsonconfig 1.0.2 requires jinja2~=3.0.0, but you have jinja2 3.1.3 which is incompatible.
netjsonconfig 1.0.2 requires jsonschema~=4.4.0, but you have jsonschema 4.21.1 which is incompatible.
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package omr-vps-admin is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Version '0.9+20240301' for 'omr-vps-admin' was not found

Ysurac, 2024-03-23:
You can use snapshot script.

chriskinal, 2024-03-23:
Where do I find the snapshot script?

Ysurac, 2024-03-23:
https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

chriskinal, 2024-03-23:
Thanks. Since this is beta, I suppose this will be fixed in release.

closed 2024-03-23 by chriskinal

Ysurac, 2024-03-23:
It's due to an update of the package that removed previous release.
I fixed rc script just now, snapshot script is the same as rc script.