Ysurac/openmptcprouter #3221: No more internet after wan3 ! Please HELP

JeedHome44, 2024-03-23:
Hello,

I received my Ethernet adaptator for StarLink. I activated by-pass in my StarLink routeur and connect it in Ethernet to my home LAN.

I add a wan 3 in OMR with MacVLan but OMR no see 192.168.1.1 to StarLink.

I delete wan 3 and now all wan see like that : 

[image]

[image]

how to make OMR work again?

JeedHome44, 2024-03-23:
I can no longer even access my Freebox and my 4G router locally even though I could do so without any problem before!

I do not understand anything anymore

closed 2024-03-23 by JeedHome44

Ysurac, 2024-03-23:
You fixed the problem ?

JeedHome44, 2024-03-23:
No I only used a backup.

I don't know why by restoring 2 WANs it no longer worked and with a BackUp yes