Ysurac/openmptcprouter #3222: OMR cannot fetch the keys from the new VPS.

vempire-ghost, 2024-03-23:
## Expected Behavior
"Setting up a new VPS works correctly."

## Current Behavior
The VPS script executes successfully, but OMR cannot download the keys.

## Steps to Reproduce the Problem
Started a new VM on Oracle Cloud with Debian 12, updated it, and successfully ran the script wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | sh.
Downloaded snapshot from 22/03 and configured it with this new VPS.

## Context (Environment)
In the OMR log, it keeps showing errors related to keys, and in the OMR wizard, all key fields are empty.
Mar 23 19:59:01 OpenMPTCProuter user.notice OMR-VPS: Get config from server vps...
Mar 23 19:59:01 OpenMPTCProuter user.notice OMR-VPS: V2ray restart...
Mar 23 19:59:01 OpenMPTCProuter user.notice OMR-VPS: Xray restart...
Mar 23 19:59:01 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Mar 23 19:59:01 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Mar 23 19:59:01 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Mar 23 19:59:01 OpenMPTCProuter user.notice xray: add rules
Mar 23 19:59:01 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Mar 23 19:59:01 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Mar 23 19:59:02 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 198.41.212.162, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 198.41
Mar 23 19:59:02 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Mar 23 19:59:02 OpenMPTCProuter daemon.info xray[660]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.22.0 linux/amd64)
Mar 23 19:59:02 OpenMPTCProuter daemon.info xray[660]: A unified platform for anti-censorship.
Mar 23 19:59:02 OpenMPTCProuter daemon.info xray[660]: Failed to start: main: failed to load config files: [/var/etc/xray/xray.main.json] > infra/conf: Failed to build TLS config. > infra/conf: failed to parse certificate > open /etc/luci-uploads/client.crt: no such file or dire
Mar 23 19:59:02 OpenMPTCProuter user.notice OMR-VPS: Glorytun restart...
Mar 23 19:59:02 OpenMPTCProuter daemon.err glorytun: Key empty
Mar 23 19:59:02 OpenMPTCProuter user.notice OMR-VPS: Glorytun UDP restart...
Mar 23 19:59:06 OpenMPTCProuter user.notice xray: Rules DOWN
Mar 23 19:59:07 OpenMPTCProuter daemon.info xray[2644]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.22.0 linux/amd64)
Mar 23 19:59:07 OpenMPTCProuter daemon.info xray[2644]: A unified platform for anti-censorship.
Mar 23 19:59:07 OpenMPTCProuter daemon.info xray[2644]: Failed to start: main: failed to load config files: [/var/etc/xray/xray.main.json] > infra/conf: Failed to build TLS config. > infra/conf: failed to parse certificate > open /etc/luci-uploads/client.crt: no such file or dir
Mar 23 19:59:12 OpenMPTCProuter daemon.info xray[3442]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.22.0 linux/amd64)
Mar 23 19:59:12 OpenMPTCProuter daemon.info xray[3442]: A unified platform for anti-censorship.
Mar 23 19:59:12 OpenMPTCProuter daemon.info xray[3442]: Failed to start: main: failed to load config files: [/var/etc/xray/xray.main.json] > infra/conf: Failed to build TLS config. > infra/conf: failed to parse certificate > open /etc/luci-uploads/client.crt: no such file or dir
Mar 23 19:59:17 OpenMPTCProuter daemon.info xray[4154]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.22.0 linux/amd64)
Mar 23 19:59:17 OpenMPTCProuter daemon.info xray[4154]: A unified platform for anti-censorship.
Mar 23 19:59:17 OpenMPTCProuter daemon.info xray[4154]: Failed to start: main: failed to load config files: [/var/etc/xray/xray.main.json] > infra/conf: Failed to build TLS config. > infra/conf: failed to parse certificate > open /etc/luci-uploads/client.crt: no such file or dir
Mar 23 19:59:22 OpenMPTCProuter daemon.info xray[5118]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.22.0 linux/amd64)
Mar 23 19:59:22 OpenMPTCProuter daemon.info xray[5118]: A unified platform for anti-censorship.
Mar 23 19:59:22 OpenMPTCProuter daemon.info xray[5118]: Failed to start: main: failed to load config files: [/var/etc/xray/xray.main.json] > infra/conf: Failed to build TLS config. > infra/conf: failed to parse certificate > open /etc/luci-uploads/client.crt: no such file or dir
Mar 23 19:59:27 OpenMPTCProuter daemon.info xray[5685]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.22.0 linux/amd64)
Mar 23 19:59:27 OpenMPTCProuter daemon.info xray[5685]: A unified platform for anti-censorship.
Mar 23 19:59:27 OpenMPTCProuter daemon.info xray[5685]: Failed to start: main: failed to load config files: [/var/etc/xray/xray.main.json] > infra/conf: Failed to build TLS config. > infra/conf: failed to parse certificate > open /etc/luci-uploads/client.crt: no such file or dir
Mar 23 19:59:27 OpenMPTCProuter daemon.info procd: Instance xray::xray.main s in a crash loop 6 crashes, 0 seconds since last crash

Manually filling in the key fields in the wizard does not solve the problem.
Configuring the snapshot with other VPS that were created previously connects and works normally.

One thing I noticed was that when starting the VPS installation script, a line appeared saying 'Update Mode,' even though it was a clean installation of Debian 12.

The necessary ports are open, and OMR can communicate with the OMR VPS admin.
[image]
I tried the option 'Retrieve settings from server,' but also without success.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60rc1-5.4 r0+16862-170d9e4
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1029-test
  - OpenMPTCProuter VPS provider: Oracle Cloud
  - OpenMPTCProuter platform: x86_64
  - Country:

Ysurac, 2024-03-23:
What is the result of ```curl -k https://127.0.0.1:65500/``` from the VPS or via a browser using https://<vpsip>:65500/ ?
What is the result of ```journalctl -u omr-admin``` from the VPS ?

vempire-ghost, 2024-03-23:
debian@instance-1:~$ curl -k https://127.0.0.1:65500/
"Welcome to OpenMPTCProuter Server part"debian@instance-1:~$

Mar 23 19:37:20 instance-1 omr-admin.py[8512]: FileNotFoundError: [Errno 2] No such file or directory: '/etc/shorewall/params.net'
Mar 23 19:37:22 instance-1 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Mar 23 19:37:22 instance-1 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Mar 23 19:37:22 instance-1 systemd[1]: omr-admin.service: Consumed 5.645s CPU time.
Mar 23 19:37:22 instance-1 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 4.
Mar 23 19:37:22 instance-1 systemd[1]: Stopped omr-admin.service - OMR-Admin.
Mar 23 19:37:22 instance-1 systemd[1]: omr-admin.service: Consumed 5.645s CPU time.
Mar 23 19:37:22 instance-1 systemd[1]: Started omr-admin.service - OMR-Admin.
Mar 23 19:37:30 instance-1 omr-admin.py[8632]: Traceback (most recent call last):
Mar 23 19:37:30 instance-1 omr-admin.py[8632]:   File "/usr/bin/omr-admin.py", line 73, in <module>
Mar 23 19:37:30 instance-1 omr-admin.py[8632]:     FILE = open('/etc/shorewall/params.net', "r")
Mar 23 19:37:30 instance-1 omr-admin.py[8632]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 23 19:37:30 instance-1 omr-admin.py[8632]: FileNotFoundError: [Errno 2] No such file or directory: '/etc/shorewall/params.net'
Mar 23 19:37:31 instance-1 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Mar 23 19:37:31 instance-1 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Mar 23 19:37:31 instance-1 systemd[1]: omr-admin.service: Consumed 5.452s CPU time.
Mar 23 19:37:31 instance-1 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 5.
Mar 23 19:37:31 instance-1 systemd[1]: Stopped omr-admin.service - OMR-Admin.
Mar 23 19:37:31 instance-1 systemd[1]: omr-admin.service: Consumed 5.452s CPU time.
Mar 23 19:37:31 instance-1 systemd[1]: Started omr-admin.service - OMR-Admin.
Mar 23 19:37:42 instance-1 omr-admin.py[8659]: Traceback (most recent call last):
Mar 23 19:37:42 instance-1 omr-admin.py[8659]:   File "/usr/bin/omr-admin.py", line 73, in <module>
Mar 23 19:37:42 instance-1 omr-admin.py[8659]:     FILE = open('/etc/shorewall/params.net', "r")
Mar 23 19:37:42 instance-1 omr-admin.py[8659]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 23 19:37:42 instance-1 omr-admin.py[8659]: FileNotFoundError: [Errno 2] No such file or directory: '/etc/shorewall/params.net'
Mar 23 19:37:43 instance-1 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Mar 23 19:37:43 instance-1 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Mar 23 19:37:43 instance-1 systemd[1]: omr-admin.service: Consumed 6.096s CPU time.
Mar 23 19:37:43 instance-1 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 6.
Mar 23 19:37:43 instance-1 systemd[1]: Stopped omr-admin.service - OMR-Admin.

Ysurac, 2024-03-24:
Seems that the script was really thinking it was an update.
To determine if it's an update, it check if the files /etc/motd or /etc/motd.head contains 'OpenMPTCProuter VPS' or if the file /root/openmptcprouter_config.txt exist.
To force update you can add ```UPDATE="no"``` as parameter of debian.sh script.

vempire-ghost, 2024-03-24:
Changed this line
# Check if OpenMPTCProuter VPS is already installed
echo "Check if OpenMPTCProuter VPS is already installed..."
update="no"

I ran the script on the problematic VPS, but it didn't solve the issue. Do I need to create a new VPS?

vempire-ghost, 2024-03-24:
Running on a fresh debian 12 install, the script ended with this error
cp: cannot create regular file '/etc/shorewall/interfaces': No such file or directory

Ysurac, 2024-03-24:
You can try a fresh install.

vempire-ghost, 2024-03-24:
> You can try a fresh install.

not worked on a fresh install
cp: cannot create regular file '/etc/shorewall/interfaces': No such file or directory

vempire-ghost, 2024-03-24:
Fresh install with RC script
root@instance-1:/home/debian# wget -O - http://www.openmptcprouter.com/server-rc/debian.sh | sh
--2024-03-24 16:55:53--  http://www.openmptcprouter.com/server-rc/debian.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 193.162.131.1
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|193.162.131.1|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 88768 (87K) [application/octet-stream]
Saving to: ‘STDOUT’

-                                                                     0%[
heck user...
Check Linux version...
Check architecture...
Check virtualized environment
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Remove lock and update packages list...

Fresh install with snapshot scrip
root@instance-1:/home/debian# wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | sh
--2024-03-24 16:57:47--  https://www.openmptcprouter.com/server-test/debian-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 193.162.131.1
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|193.162.131.1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 88869 (87K) [application/octet-stream]
Saving to: ‘STDOUT’

-                                                                     0%[                                                                                                                                                                   ]       0  --.-KB/s               Check user...
Check Linux version...
Check architecture...
Check virtualized environment
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Remove lock and update packages list...
Reading package lists... Done

Always thinking its a update, and adding the line you suggested don't fixed the installation.

Ysurac, 2024-03-24:
The string "update mode" is always displayed, it means that update can be done and not that update will be done. It's an error, it's now fixed and will be displayed only if update is really done.
To force disable update mode: ``` wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPDATE="no" sh```
You should have all keys displayed at the end of the script, if already run it will erase previous files.

vempire-ghost, 2024-03-24:
> The string "update mode" is always displayed, it means that update can be done and not that update will be done. It's an error, it's now fixed and will be displayed only if update is really done. To [...]

The keys are showed at the end of the script, but still no success.
debian@instance-1:~$ journalctl -u omr-admin
Mar 24 18:17:39 instance-1 systemd[1]: Started omr-admin.service - OMR-Admin.
Mar 24 18:17:46 instance-1 omr-admin.py[8428]: Traceback (most recent call last):
Mar 24 18:17:46 instance-1 omr-admin.py[8428]:   File "/usr/bin/omr-admin.py", line 73, in <module>
Mar 24 18:17:46 instance-1 omr-admin.py[8428]:     FILE = open('/etc/shorewall/params.net', "r")
Mar 24 18:17:46 instance-1 omr-admin.py[8428]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 24 18:17:46 instance-1 omr-admin.py[8428]: FileNotFoundError: [Errno 2] No such file or directory: '/etc/shorewall/params.net'
Mar 24 18:17:48 instance-1 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Mar 24 18:17:48 instance-1 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Mar 24 18:17:48 instance-1 systemd[1]: omr-admin.service: Consumed 5.101s CPU time.
Mar 24 18:17:49 instance-1 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 1.
Mar 24 18:17:49 instance-1 systemd[1]: Stopped omr-admin.service - OMR-Admin.
Mar 24 18:17:49 instance-1 systemd[1]: omr-admin.service: Consumed 5.101s CPU time.
Mar 24 18:17:49 instance-1 systemd[1]: Started omr-admin.service - OMR-Admin.
Mar 24 18:17:54 instance-1 omr-admin.py[8524]: Traceback (most recent call last):
Mar 24 18:17:54 instance-1 omr-admin.py[8524]:   File "/usr/bin/omr-admin.py", line 73, in <module>
Mar 24 18:17:54 instance-1 omr-admin.py[8524]:     FILE = open('/etc/shorewall/params.net', "r")
Mar 24 18:17:54 instance-1 omr-admin.py[8524]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 24 18:17:54 instance-1 omr-admin.py[8524]: FileNotFoundError: [Errno 2] No such file or directory: '/etc/shorewall/params.net'

Ysurac, 2024-03-24:
What do you have in ```/etc/motd``` or ```/etc/motd.head``` ?

vempire-ghost, 2024-03-24:
> What do you have in `/etc/motd` or `/etc/motd.head` ?

in motd
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
< OpenMPTCProuter VPS 0.1029-test >

This is after running the script.

Ysurac, 2024-03-24:
Here OpenMPTCProuter is installed, so the script only update. If you remove the "OpenMPTCProuter" line, and the ```/root/openmptcprouter_config.txt``` it should do a fresh install.
There is a specific reason to use kernel 5.4 release ?

vempire-ghost, 2024-03-24:
> Here OpenMPTCProuter is installed, so the script only update. If you remove the "OpenMPTCProuter" line, and the `/root/openmptcprouter_config.txt` it should do a fresh install. There is a specific reason to use kernel 5.4 release ?

This installation was performed on a fresh Debian install with the script wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPDATE="no" sh. Shouldn't it have installed instead of trying to update?

I need the redundant scheduler, which for now, only exists in the 5.4 kernel.

Ysurac, 2024-03-24:
Yes it should have installed and no updated in initial install or with the UPDATE="no". 
Try again on a fresh OS install.

vempire-ghost, 2024-03-24:
> Yes it should have installed and no updated in initial install or with the UPDATE="no". Try again on a fresh OS install.

omr vps admin is now running
debian@instance-1:/etc/openmptcprouter-vps-admin$ sudo systemctl status omr-admin
● omr-admin.service - OMR-Admin
     Loaded: loaded (/lib/systemd/system/omr-admin.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-03-24 19:17:07 UTC; 9min ago
   Main PID: 424 (omr-admin.py)
      Tasks: 6 (limit: 1146)
     Memory: 85.4M
        CPU: 14.152s
     CGroup: /system.slice/omr-admin.service
             └─424 /usr/bin/python3 /usr/bin/omr-admin.py

Mar 24 19:23:36 instance-1 omr-admin.py[424]:   File "/usr/local/lib/python3.11/dist-packages/fastapi/routing.py", line 191, in run_endpoint_function
Mar 24 19:23:36 instance-1 omr-admin.py[424]:     return await dependant.call(**values)
Mar 24 19:23:36 instance-1 omr-admin.py[424]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 24 19:23:36 instance-1 omr-admin.py[424]:   File "/usr/bin/omr-admin.py", line 1768, in config
Mar 24 19:23:36 instance-1 omr-admin.py[424]:     ss_go_txrx = get_bytes_ss_go(username)
Mar 24 19:23:36 instance-1 omr-admin.py[424]:                  ^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 24 19:23:36 instance-1 omr-admin.py[424]:   File "/usr/bin/omr-admin.py", line 174, in get_bytes_ss_go
Mar 24 19:23:36 instance-1 omr-admin.py[424]:     for userdata in r.json()['users']:
Mar 24 19:23:36 instance-1 omr-admin.py[424]:                     ~~~~~~~~^^^^^^^^^
Mar 24 19:23:36 instance-1 omr-admin.py[424]: KeyError: 'users'

But OMR still cannot retrieve the keys fom VPS.
Mar 24 19:23:53 OpenMPTCProuter daemon.info xray[30930]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.22.0 linux/amd64)
Mar 24 19:23:53 OpenMPTCProuter daemon.info xray[30930]: A unified platform for anti-censorship.
Mar 24 19:23:53 OpenMPTCProuter daemon.info xray[30930]: Failed to start: main: failed to load config files: [/var/etc/xray/xray.main.json] > infra/conf: Failed to build TLS config. > infra/conf: failed to parse certificate > open /etc/luci-uploads/client.crt: no such file or di

Ysurac, 2024-03-24:
This time it was a bug in the API from latest changes I made.
It's fixed now, you can run the script again and this will update the API.

vempire-ghost, 2024-03-25:
Thank you its now working.

closed 2024-03-25 by vempire-ghost