Ysurac/openmptcprouter #3224: Wifi clients connect but can't reach LAN network

ambro95, 2024-03-25:
## Expected Behavior
Clients connect via wifi to the wlan0 interface on Rpi4 and can access LuCi interface and navigate on internet

## Current Behavior
Clients can connect on the wireless AP but can't reach anything, not even the administration interface.
I had everything setted up and devices were working flawlessy, I then changed shadowsocks to v2ray for testing purposes and from then I lost access via wifi. I even reflashed the SD card with a clean install.
Logs apparently doesn't say anything relavant to this case

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028-5.4.207
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: RPI4

Ysurac, 2024-03-25:
Change of proxy shouldn't block wifi. What is your wifi configuration ?

ambro95, 2024-03-25:
Hi Ysurac, sure, this is the wireless settings:
[image]
[image]
[image]
[image]
[image]

And this is the interface settings:
[image]
[image]
[image]

ambro95, 2024-03-25:
I got it to work again. I set the IPv4 gateway to 192.168.100.0 and now it's working.
I don't understand why on the LAN interface there is no IPv4 GW but it's working fine

ambro95, 2024-03-25:
When something is changed on the Advanced Settings of the OpenMPTCProuter wizard, interfaces are being reset. Is this an expected behaviour?

Ysurac, 2024-03-25:
Yes it's expected.

ambro95, 2024-03-25:
Ok thank you. Can you tell me why it's working fine on the default LAN interface (eth0) even without the GW setted?

ambro95, 2024-03-25:
Found the problem on IPv4 GW being reset on the interfaces when changing proxy type

closed 2024-03-25 by ambro95