Ysurac/openmptcprouter #3225: OMR not aggregating

YB1200, 2024-03-25:
## Expected Behavior
I expect to achieve accurate aggregated throughput(nearly 300Mbps).

## Current Behavior
I want to aggregate two modems (4G and 5G). The installation seems okay, but I'm not achieving the expected result(I got nearly 30Mbps). I tested with speed test and omr-test-speed : I got same result. The status page shows all green indicators.
## Specifications

  - OpenMPTCProuter version: v0.60rc Latest snapshot
  - OpenMPTCProuter VPS version: 0.1209-test 6.1 (Latest update)
  - OpenMPTCProuter platform: Virtual Box(x86_64)

Ysurac, 2024-03-25:
What is the result, via SSH on the router, with ```omr-test-speed```, ```omr-test-speed wan1``` (if the interface is wan1, and for each interface), and what is the result if you do ```omr-test-speed wan1``` and ```omr-test-speed wan2``` at the same time ?

YB1200, 2024-03-25:
For omr-test-speed, I got this result: 
[image]

For omr-test-speed eth1 and omr-test-speed eth2: I got an error:
[image]

Ysurac, 2024-03-31:
Can you put a screenshot of System->OpenMPTCProuter, "Status" page ?

hhaniel, 2024-04-13:
I have same problem -upgraded to the lates router image and vps - I have even started from scratch and manually configured everything again today with the latest image of the generic ext3 vmware image and a brand new vps running debian 12 and still no luck and i can not see any erroors

hhaniel, 2024-04-13:
Forgot to mention - Same setup on OMR 59 and Debian 10 was working fine - I am considering downgrading again as i still have a snapshot of the OMR VM but on the VPS side i will have difficulty as I can not rebuild my VPS to debian 10

closed 2024-07-18 by github-actions[bot]