Ysurac/openmptcprouter #3226: Glorytun failing on OMR 0.60RC1 kernel 5.4

moi6576, 2024-03-25:
Hello, 

I have installed OMR 0.60RC1 kernel 5.4 on a new NUC (to replace a failing one that was on 0.59 kernel 5.4). VPS is on 1028. All seems to work but when I try to access my network from outside, glorytun TCP disconnects and I lose internet (and cannot connect from outside). Not sure what to do in this situation ? 

Here is the log when it happens : 
```
Mar 25 16:45:42 OpenMPTCProuter syslog.info syslogd started: BusyBox v1.33.2
Mar 25 16:45:42 OpenMPTCProuter user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mar 25 16:45:42 OpenMPTCProuter user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mar 25 16:45:42 OpenMPTCProuter user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: started, version 2.85 cachesize 150
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: DNS service limited to local subnets
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: using only locally-known addresses for domain test
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: using only locally-known addresses for domain onion
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: using only locally-known addresses for domain localhost
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: using only locally-known addresses for domain local
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: using only locally-known addresses for domain invalid
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: using only locally-known addresses for domain bind
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: using only locally-known addresses for domain use-application-dns.net
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: using nameserver 127.0.0.1#5353
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: using only locally-known addresses for domain lan
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: read /etc/hosts - 4 addresses
Mar 25 16:45:42 OpenMPTCProuter daemon.info dnsmasq[4348]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Mar 25 16:45:42 OpenMPTCProuter authpriv.info dropbear[4411]: Not backgrounding
Mar 25 16:45:42 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Failed to connect to ubus
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_lan' cannot resolve device of network 'lan'
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan1'
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan2'
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omrvpn'
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omr6in4'
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[0] (30250) does not specify a protocol, assuming TCP+UDP
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[1] (30251) does not specify a protocol, assuming TCP+UDP
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[3] (30258) does not specify a protocol, assuming TCP+UDP
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[4] (30259) does not specify a protocol, assuming TCP+UDP
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[5] (30255) does not specify a protocol, assuming TCP+UDP
Mar 25 16:45:42 OpenMPTCProuter user.notice firewall.omr-bypass: reload omr-bypass rules
Mar 25 16:45:42 OpenMPTCProuter user.notice Shadowsocks: Rules UP
Mar 25 16:45:42 OpenMPTCProuter user.notice Shadowsocks: Reload omr-bypass rules
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Failed to connect to ubus
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_lan' cannot resolve device of network 'lan'
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan1'
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan2'
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omrvpn'
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omr6in4'
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[0] (30250) does not specify a protocol, assuming TCP+UDP
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[1] (30251) does not specify a protocol, assuming TCP+UDP
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[3] (30258) does not specify a protocol, assuming TCP+UDP
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[4] (30259) does not specify a protocol, assuming TCP+UDP
Mar 25 16:45:42 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[5] (30255) does not specify a protocol, assuming TCP+UDP
Mar 25 16:45:43 OpenMPTCProuter daemon.notice wpa_supplicant[4917]: Successfully initialized wpa_supplicant
Mar 25 16:45:43 OpenMPTCProuter user.notice : Added device handler type: bonding
Mar 25 16:45:43 OpenMPTCProuter user.notice : Added device handler type: 8021ad
Mar 25 16:45:43 OpenMPTCProuter user.notice : Added device handler type: 8021q
Mar 25 16:45:43 OpenMPTCProuter user.notice : Added device handler type: macvlan
Mar 25 16:45:43 OpenMPTCProuter user.notice : Added device handler type: veth
Mar 25 16:45:43 OpenMPTCProuter user.notice : Added device handler type: bridge
Mar 25 16:45:43 OpenMPTCProuter user.notice : Added device handler type: Network device
Mar 25 16:45:43 OpenMPTCProuter user.notice : Added device handler type: tunnel
Mar 25 16:45:43 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is enabled
Mar 25 16:45:43 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is setting up now
Mar 25 16:45:43 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now up
Mar 25 16:45:43 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is enabled
Mar 25 16:45:43 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is setting up now
Mar 25 16:45:43 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is now up
Mar 25 16:45:43 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is enabled
Mar 25 16:45:43 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is setting up now
Mar 25 16:45:43 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is now up
Mar 25 16:45:43 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is enabled
[... 227 lines cut ...]
Mar 25 16:46:40 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99v2ray: Flush terminated
Mar 25 16:46:40 OpenMPTCProuter daemon.notice ttyd[26861]: ttyd 1.6.3 (libwebsockets 3.1.0)
Mar 25 16:46:40 OpenMPTCProuter daemon.notice ttyd[26861]: tty configuration:
Mar 25 16:46:40 OpenMPTCProuter daemon.notice ttyd[26861]:   start command: /bin/login
Mar 25 16:46:40 OpenMPTCProuter daemon.notice ttyd[26861]:   close signal: SIGHUP (1)
Mar 25 16:46:40 OpenMPTCProuter daemon.notice ttyd[26861]:   terminal type: xterm-256color
Mar 25 16:46:40 OpenMPTCProuter daemon.notice ttyd[26861]:  Using foreign event loop...
Mar 25 16:46:40 OpenMPTCProuter daemon.notice ttyd[26861]: Creating Vhost 'default' port 7681, 2 protocols, IPv6 off
Mar 25 16:46:40 OpenMPTCProuter daemon.notice ttyd[26861]:  Listening on port: 7681
[... cut ...]
Mar 25 16:56:08 OpenMPTCProuter daemon.info glorytun[7553]: STOPPED tun0
Mar 25 16:56:09 OpenMPTCProuter daemon.err glorytun[7553]: 134.59.116.4.65001: connected
Mar 25 16:56:19 OpenMPTCProuter daemon.err glorytun[7553]: 134.59.116.4.65001: key exchange failed
Mar 25 16:56:19 OpenMPTCProuter daemon.info glorytun[7553]: STOPPED tun0
Mar 25 16:56:20 OpenMPTCProuter daemon.err glorytun[7553]: 134.59.116.4.65001: connected
```
Thanks a lot for any help and even more thanks to Icarus for this magic solutions !!!

moi6576, 2024-03-25:
PS : same issue using alternative VPN

moi6576, 2024-03-25:
If I stop the firewall (all accept), all work perfectly !!

Ysurac, 2024-03-25:
What you mean with "all accept" ?

moi6576, 2024-03-25:
The ones on top in general settings
[image]

moi6576, 2024-03-25:
Not sure whether it is the solution or a bug in the port definition. I need to define the protocol several times before the mention "TCP and UDP" appear (it was selected right from the start but was not showing) : 
[image]

moi6576, 2024-03-25:
I confirm that the issue came from the port forwarding definition. By default TCP and UDP are selected. If you don't change it and define the port, the mentions TCP and UDP do not appear and external access/VPN fails. 
The solution is to go back to the port definition, enter any, save, come back, enter TCP and UPD, re-enter destination and ports and save : the mentions appear and all work beautifully !!

Ysurac, 2024-03-25:
For the redirection, try to only accept VPN input instead of general settings.
For the ports, I would need the result of ```uci show firewall``` via SSH.

moi6576, 2024-03-25:
Yes, it was just for testing purpose.
For the ports, here you go. Note that the last one (port 34000) was entered a second ago for testing : I did not change the protocol (should be TCP and UDP by default) and it does not appear as TCP and UDP. That was my problem. 
```
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='REJECT'
firewall.@defaults[0].output='REJECT'
firewall.@defaults[0].forward='REJECT'
firewall.zone_lan=zone
firewall.zone_lan.name='lan'
firewall.zone_lan.network='lan'
firewall.zone_lan.input='ACCEPT'
firewall.zone_lan.output='ACCEPT'
firewall.zone_lan.forward='ACCEPT'
firewall.zone_lan.mtu_fix='1'
firewall.zone_lan.auto_helper='1'
firewall.zone_wan=zone
firewall.zone_wan.name='wan'
firewall.zone_wan.input='REJECT'
firewall.zone_wan.output='ACCEPT'
firewall.zone_wan.forward='REJECT'
firewall.zone_wan.masq='1'
firewall.zone_wan.mtu_fix='1'
firewall.zone_wan.auto_helper='1'
firewall.zone_wan.network='wan1' 'wan2'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
firewall.@rule[0].proto='udp'
firewall.@rule[0].dest_port='68'
firewall.@rule[0].target='ACCEPT'
firewall.@rule[0].family='ipv4'
firewall.@rule[1]=rule
firewall.@rule[1].name='Allow-Ping'
firewall.@rule[1].src='wan'
firewall.@rule[1].proto='icmp'
firewall.@rule[1].icmp_type='echo-request'
firewall.@rule[1].family='ipv4'
firewall.@rule[1].target='ACCEPT'
firewall.@rule[2]=rule
firewall.@rule[2].name='Allow-IGMP'
firewall.@rule[2].src='wan'
firewall.@rule[2].proto='igmp'
firewall.@rule[2].family='ipv4'
firewall.@rule[2].target='ACCEPT'
firewall.@rule[3]=rule
firewall.@rule[3].name='Allow-DHCPv6'
firewall.@rule[3].src='wan'
firewall.@rule[3].proto='udp'
firewall.@rule[3].dest_port='546'
firewall.@rule[3].family='ipv6'
firewall.@rule[3].target='ACCEPT'
firewall.@rule[4]=rule
firewall.@rule[4].name='Allow-MLD'
firewall.@rule[4].src='wan'
firewall.@rule[4].proto='icmp'
firewall.@rule[4].src_ip='fe80::/10'
firewall.@rule[4].icmp_type='130/0' '131/0' '132/0' '143/0'
firewall.@rule[4].family='ipv6'
firewall.@rule[4].target='ACCEPT'
firewall.@rule[5]=rule
firewall.@rule[5].name='Allow-ICMPv6-Forward'
firewall.@rule[5].src='wan'
firewall.@rule[5].dest='*'
firewall.@rule[5].proto='icmp'
firewall.@rule[5].icmp_type='echo-request' 'echo-reply' 'destination-unreachable' 'packet-too-big' 'time-exceeded' 'bad-header' 'unknown-header-type'
firewall.@rule[5].limit='1000/sec'
[... 216 lines cut ...]
firewall.@redirect[8].src='vpn'
firewall.@redirect[8].dest='lan'
firewall.@redirect[8].proto='tcp' 'udp'
firewall.@redirect[8].dest_port='8000'
firewall.@redirect[8].src_dport='30257'
firewall.@redirect[8].name='30257 routeur kids'
firewall.@redirect[8].dest_ip='192.168.100.239'
firewall.@redirect[9]=redirect
firewall.@redirect[9].target='DNAT'
firewall.@redirect[9].name='30256  tautulli'
firewall.@redirect[9].src='vpn'
firewall.@redirect[9].dest='lan'
firewall.@redirect[9].proto='tcp' 'udp'
firewall.@redirect[9].src_dport='30256'
firewall.@redirect[9].dest_ip='192.168.100.44'
firewall.@redirect[9].dest_port='8181'
firewall.@redirect[10]=redirect
firewall.@redirect[10].target='DNAT'
firewall.@redirect[10].name='test'
firewall.@redirect[10].src='vpn'
firewall.@redirect[10].src_dport='34000'
firewall.@redirect[10].dest='vpn'
firewall.@redirect[10].dest_port='34000'

```

moi6576, 2024-03-25:
Damn : I stil have the issue when remote accessing ...

moi6576, 2024-03-25:
[image]

moi6576, 2024-03-25:
Mar 25 20:37:47 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar 25 20:37:50 OpenMPTCProuter daemon.err glorytun[14734]: x.x.x.x.65001: key exchange failed
Mar 25 20:37:50 OpenMPTCProuter daemon.info glorytun[14734]: STOPPED tun0
Mar 25 20:37:51 OpenMPTCProuter daemon.err glorytun[14734]: x.x.x.x.65001: connected
Mar 25 20:38:00 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
Mar 25 20:38:01 OpenMPTCProuter daemon.err glorytun[14734]: x.x.x.x.65001: key exchange failed
Mar 25 20:38:01 OpenMPTCProuter daemon.info glorytun[14734]: STOPPED tun0
Mar 25 20:38:02 OpenMPTCProuter daemon.err glorytun[14734]: x.x.x.x.65001: connected
Mar 25 20:38:03 OpenMPTCProuter user.notice firewall.omr-bypass: reload omr-bypass rules
Mar 25 20:38:03 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Mar 25 20:38:03 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Mar 25 20:38:12 OpenMPTCProuter daemon.err glorytun[14734]: x.x.x.x.65001: key exchange failed
Mar 25 20:38:12 OpenMPTCProuter daemon.info glorytun[14734]: STOPPED tun0
Mar 25 20:38:13 OpenMPTCProuter daemon.err glorytun[14734]: x.x.x.x.65001: connected
Mar 25 20:38:23 OpenMPTCProuter daemon.err glorytun[14734]: x.x.x.x.65001: key exchange failed
Mar 25 20:38:23 OpenMPTCProuter daemon.info glorytun[14734]: STOPPED tun0
Mar 25 20:38:24 OpenMPTCProuter daemon.err glorytun[14734]: x.x.x.x.65001: connected
Mar 25 20:38:34 OpenMPTCProuter daemon.err glorytun[14734]: x.x.x.x.65001: key exchange failed
Mar 25 20:38:34 OpenMPTCProuter daemon.info glorytun[14734]: STOPPED tun0
Mar 25 20:38:35 OpenMPTCProuter daemon.err glorytun[14734]: x.x.x.x.65001: connected
Mar 25 20:38:45 OpenMPTCProuter daemon.err glorytun[14734]: x.x.x.x.65001: key exchange failed
Mar 25 20:38:45 OpenMPTCProuter daemon.info glorytun[14734]: STOPPED tun0

Ysurac, 2024-03-26:
Do you have same issue when using OpenVPN as VPN in System->Wizard, "Advanced settings" checkbox ?

moi6576, 2024-03-26:
Yes and no : I get "VPN down" with OpenVPN and never a VPN working (while with glorytun, all works beautifully with a nice VPN up and running until I use remote access which kills the VPN).

Ysurac, 2024-03-26:
Hum yes, you have 0.1028 VPS release, it can explain why OpenVPN is not working correctly.
The VPN is down when you are doing some high traffic or even small external traffic ?
You should update VPS side in any case.

moi6576, 2024-03-26:
Not really. I can download massively (500MBPS), no problem. The only issue occurs when I go to IP-Router:port_forward (ie. 200.200.200.200:3312) -> VPN goes down and no access. I ll update the VPS (tonight as I am not on site and I have no external access - Arghhh ;-) ). Thanks !

moi6576, 2024-03-26:
I managed to update the VPS to 0.1029-test. Should I use a snapshot or RC1 for the router in order to test tonight ? I tried a snapshot from 15th of March if I remember well.

Ysurac, 2024-03-26:
Use a snapshot.

moi6576, 2024-03-26:
Ok, I ll get the march 26 one then.

moi6576, 2024-03-26:
In between, could the macvlan protocol create some issues ???
[image]

moi6576, 2024-03-26:
I don't dare to write it down but all seems to work beautifully with glorytun TCP, VPS 1029 and the latest snapshot of OMR ! I'll update in the next 24h to check whether it lasts. Can we show our appreciation of the great help/tool provided for free with a donation or something ?

moi6576, 2024-03-26:
Argh ... still seing VPN going down. Interestingly, most of my issues occur when I try to connect Plex which seemingly has its own certificates : could it be an issue ? I see Plex loading and around 50%, it stops and kills the VPN. Any idea ? Can I by bypass OMR and directly connect from outside my Plex server (not sure if this works in this direction) ?

Ysurac, 2024-03-26:
Certificates is not an issue. Maybe a network loop is created ? Can you try to launch a ping to 10.255.255.1 from the router when you try plex ?
Did you try another VPN now that VPS is updated ?
For the bypass from VPS to OMR, not possible.

moi6576, 2024-03-27:
All is now working well, not a single interruption of the VPN. A big thanks for the top help (and the snapshots) !

closed 2024-03-27 by moi6576