Ysurac/openmptcprouter #3227: SRT stream not working

ambro95, 2024-03-25:
## Expected Behavior
SRT traffic expected to route like normal UDP traffic

## Current Behavior
I'm trying to use OMR for sending a SRT stream, but until now I had no results. I tried to use V2Ray, as someone suggested, for UDP traffic, but without luck, the stream can't reach OBS.
Can someone help me?

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028-5.4.207
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: RPi4

Ysurac, 2024-03-26:
SRT use UDP as transport protocol, so if VPN between the router and the VPS is working then it should work. You may try to set manually MTU on the sender application.
V2Ray can also provide UDP transport, in System->OpenMPTCProuter, "Advanced settings" tab, check "
"When proxy V2Ray is used, use it for UDP" box.

ambro95, 2024-03-26:
I already tried SS+GT TCP and V2Ray, but without results. 
Why are you proposting to change the MTU on sender side? Even if there is fragmentation, it's beeing handled between OMR and VPS, right? The receiver has no idea that the packets are beeing fragmented on the source

Palexxxxx, 2024-03-29:
I have the same problem… Does anyone know how to fix it?

Palexxxxx, 2024-03-30:
> > I have the same problem… Does anyone know how to fix it?
> 
> I resolved this by ditching OMR in favor of SRTLA done with the belabox code. Self hosting the server for the srt bonding is still possible even if it's not updated anymore from the developers, or you can pay 10$/month for their servers.

Could you explain your setup?

ambro95, 2024-03-30:
Search for belabox, there's a github repository and a website with well explained instructions on how to setup it. Keep in mind that RPi are not supported

Ysurac, 2024-04-04:
After some research, I think that SRT packets have a DF (don't fragment) flag set. I will check what I can do about that if client can't disable it...

ambro95, 2024-04-04:
I already came across the fact that SRT packets can't be fragmented, but I thought that it didn't matter because the packets are "reconstructed" anyway once on the VPS. If I knew that wasn't a silly hypothesis I would definitely wrote it here.
Btw thank you @Ysurac for looking at it.

Ysurac, 2024-04-05:
Do you have issues when streaming to any service from OBS (would be easier to test than streaming from my home to my home) ?

ambro95, 2024-04-05:
I think I'm not following you, sorry.
After the srt stream arrives on my pc with OBS, I can re-stream it to any service I want without problems.
The phone with IRL pro is an OMR client, and OBS on my pc is receiving the video from my home wan. OMR is connected with a tethering phone just to testing.

Ysurac, 2024-04-05:
As I said I don't know anything to SRT.
So you have OBS on a WAN (not using OMR) and your phone using OMR to reach OBS via Internet ?
How I can put OBS as SRT listener ?

tiz47, 2024-04-05:
Hi, I also use omr with srt. I tried a setup similar to the one above; obs/vmix as srt listener behind omr and another obs sending srt over internet.
current configuration:
OMR wan are two wired connections (about 80down/20 up both) with MTU 1492 because they use PPPoE.
OMR is configured to use shadowsocks and glorytun TCP.
By connecting directly through the open port on the OMR public IP I have a costant packet loss of around 10% (I'm sending an SRT flow at 30mbps)
using a wireguard VPN between the remote OBS and the one behind OMR the packet loss drops to 0%.
OMR seems to always use only one of the two connections (in both tests it used the same one)

If more information is needed I can do other tests.

Ysurac, 2024-04-11:
@ambro95 ambro I think I'm really not good on this... OBS doesn't open a port with the srt://... configuration.
I can see that some packets are coming put without server...
Can you check with a ```tcpdump -i any -p port <portyouuse>``` on your computer that packets are coming ? If you are on windows, tcpdump should be available or wireguard can also be used.

@tiz47 you checked with another VPN ? Still same problem with a flow using less bandwidth ? What release/kernel of OpenMPTCProuter are you using ?

tiz47, 2024-04-11:
Hi, I didn't try with another VPN, the tests were done with OMR 0.59.1 and kernel 5.4. I saw that version 0.60 is available, is it better to try the one with kernel 6.1?

ambro95, 2024-04-11:
@Ysurac I can do a tcpdump this weekend, but I already looked on the firewall and I didn't see any traffic coming through the pc with obs.
Anyway if you want I'm available on discord for helping you with the obs test on your side. Let me know

ambro95, 2024-04-15:
Tested and I didn't see any traffic reaching OBS

closed 2024-07-19 by github-actions[bot]