Ysurac/openmptcprouter #3229: openmptcprouter don't start

Gregg03, 2024-03-27:
## Expected Behavior

When I install the openmptcprouter on a HP serveur Proliant DL360p Gen8 x86-64 it's not starting.

## Current Behavior
The system stop in the start
And just tell me the name of my keyboard

## Specifications

  - OpenMPTCProuter version: [openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined.img.gz
  - OpenMPTCProuter platform: x86_64

patimages, 2024-03-28:
Your gear is most likely not supported by 5.4 kernel ...

Gregg03, 2024-04-05:
I try to install the old version on the 5.1 kernel and a I have same thing

Gregg03, 2024-04-05:
I have this result 
[image]

Gregg03, 2024-04-05:
I can't attach a file ?

patimages, 2024-04-05:
Tou may rather try 6.1 .... https://wiki.debian.org/HP/ProLiant

patimages, 2024-04-05:
[image]
Yes you can, but you file seems to not upload

Gregg03, 2024-04-05:
when i try to upload the picture i have that

Gregg03, 2024-04-05:
[image]

patimages, 2024-04-05:
Internet issues ? You need Openmptcprouter ;-D !

Gregg03, 2024-04-05:
[image]

Gregg03, 2024-04-05:
i have that when i try to start the server

moi6576, 2024-04-05:
Try 6.1

Gregg03, 2024-04-05:
this version ? 
https://download.openmptcprouter.com/release/v0.60beta1-6.1/x86_64/targets/x86/64/openmptcprouter-v0.60beta1-6.1-r0+23789-ce6ad123e7-x86-64-generic-ext4-combined-efi.img.gz

patimages, 2024-04-05:
try : https://download.openmptcprouter.com/release/v0.60rc2-5.4/
and if no luck : https://download.openmptcprouter.com/release/v0.60rc2-6.1/

Seems as if you are mixing versions : Efi, non Efi ...

Gregg03, 2024-04-05:
yes sorry it's not support efi

Gregg03, 2024-04-05:
I tried both and it's still the same

Gregg03, 2024-04-05:
for 5.4 
[image]

Gregg03, 2024-04-05:
and for 6.1 
[image]

Gregg03, 2024-04-05:
and for the 6.1 I have this message at the end : 
random: crng init done

patimages, 2024-04-05:
Not sure whether it is a failure for the latter ... did you connect the different lines to aggregate ? can you access OMR ?

Ysurac, 2024-04-06:
If you press "Enter" key you have nothing ?

Gregg03, 2024-04-06:
No i just have a internet connection, what is ORM

Gregg03, 2024-04-06:
if I press "enter", it's just make a line break like \n

Ysurac, 2024-04-06:
OMR = OpenMPTCProuter

Gregg03, 2024-04-06:
no i don't have access

patimages, 2024-04-06:
Meaning that you tried to connect http://192.168.100.1/ once all is connected properly ? Did you install the VPS ?

Gregg03, 2024-04-06:
I connect my PC on the server and try to connect on the http://192.168.100.1/ and i don't have access

Gregg03, 2024-04-06:
no I haven't installed the VPS server yet

patimages, 2024-04-06:
Did you try to connect on the http://192.168.100.1/ ? Do you know OMR well ?

Gregg03, 2024-04-06:
yes i tried to connect but it doesn't work

patimages, 2024-04-06:
Hard to help then ... Lots of possible causes ...

Gregg03, 2024-04-06:
sometime I have this message : "random: crng init done"

Ysurac, 2024-04-06:
it's only initialization of random number generator.
Try to remove all USB device as it seems to end on USB.

Gregg03, 2024-04-22:
I tried to revome all USB equipment and i have same result

Gregg03, 2024-04-22:
[image]

closed 2024-07-27 by github-actions[bot]