Ysurac/openmptcprouter #3230: OMR does not aggregate

aladdinshammr, 2024-03-28:
## Expected Behavior

I am trying to aggregate 2 LTE modems (for now), so I can get the maximum speed.
internet is really bad here and speed is volatile, especially in LTE modems. so I don't expect a specific speed.
I just want all the connections to work in parallel. 

## Current Behavior
When I test with only one modem, it works fine. but as soon as I add the second one, things go wrong. 
I tested with omr-test-speed wan1 or wan2, I get an acceptable speed.
but when I try omr-test-speed, I get a bad result.
it's even less than one connection alone.

I do not know how things would be when I try with the other 15 adsl connections .

system log is very long and I  do not know how to locate the issue.

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028.5.15.0-1054-gcp
  - OpenMPTCProuter VPS provider:GCP
  - OpenMPTCProuter platform: First I tried with RPI4 and now x86_64 in proxmox.

[image]

- thank you in advance, you are doing a great job.
- and I salute your patience if you reached this line

Ysurac, 2024-03-28:
The VPS is using a 5.15.0 kernel, it's not supported. Did you reboot the VPS after installation ?

aladdinshammr, 2024-03-29:
sorry I am late.

Yes I Did reboot the server after installation.
I tried ubuntu as well as Debian the same result.
 
do you recommend any updated tutorial, because I think all tutorials out there are outdated..

Ysurac, 2024-03-29:
Tutorials are not outdated, but you can try latest 0.60rc2: https://github.com/Ysurac/openmptcprouter/issues/2961

aladdinshammr, 2024-03-29:
Thank you for your help.
I will try this version.

by the way , how many connections can I add?
you mentioned in the doc "" from up to 8 internet connections""
so 8 connections is the maximum?

Ysurac, 2024-03-29:
You can add up to 8 connections.
With versions using kernel 5.4 on x86/x86_64 you can add up to 32 connections.
With versions using kernel 6.1 I have a not yet tested patch to add up to 64 connections.

aladdinshammr, 2024-03-30:
Fair enough. 
I appreciate your effort

lotus10, 2024-03-30:
@Ysurac Hi, my dear.
 I had the same problem. 
 router version 0.59-1-54 and vps version 0.1028 5.4.207.
i have  3 LTE modems (a b  c ):
When using A alone, the network speed download can reach about 25Mb/S,upload 16Mb/s,
 using B alone, the network speed download can reach about 20Mb/S,upload 8Mb/s,
 using C alone, the network speed download can reach about 40Mb/S,upload 20Mb/s
When the network is aggregated, the  network speed can only reach 12Mb/S upload and 30M download.

aladdinshammr, 2024-03-31:
@lotus10 
to be honest, I just reran VPS script then rebooted and it worked as expected.
I also got a notification that the connections block MPTCP, but when I enabled MPTCP over VPN, it disappeared.
try doing that

closed 2024-04-03 by aladdinshammr