Ysurac/openmptcprouter #3234: Gateway Down Error

pablodnaarg, 2024-03-31:
Gateway Working

## Current Behavior

The modem IP is 192.90.2.1
DHCP Modem is Disable
DIrect conection Modem to PC works fine and i can reach internet
What are the possible causes for this error ? How could I indagate?

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4]
  - OpenMPTCProuter VPS version: none
  - OpenMPTCProuter VPS provider: none
  - OpenMPTCProuter platform: RP4 x64

Ysurac, 2024-03-31:
Check that you can ping the gateway you defined via SSH from the router.

pablodnaarg, 2024-03-31:
Hello Ysurac, thank you very much for responding so quickly. I just solved the problem, it was a LAN port setting on the modem.

pablodnaarg, 2024-03-31:
For those using a Huawei B311 modem, it must be set up like this. Otherwise, it will not work.

[image]

closed 2024-04-02 by pablodnaarg