Ysurac/openmptcprouter #3235: Preventing a LAN client access to WAN

IVCrossway, 2024-03-31:
Hello, I currently have OMR installed on my main router. I also have an AP used for the Garage WIFI, this AP is doing nat. So my main network is 192.168.188.0/24 while the Garage AP is giving 192.168.1.x IPs and I would like to keep it this way.
I tried to add to OMR two firewall rules to prevent the AP (and so, the clients connected to it) access to Internet, but it is not working as intended.
Would you help me to fix this?

I also tried to do this rule with the AP static ip address but it didn't work either.

```
config rule
        option name 'test1'
        list proto 'all'
        option dest 'wan'
        option target 'REJECT'
        option src 'lan'
        list src_mac '[redacted]'

config rule
        option name 'test2'
        list proto 'all'
        option dest 'vpn'
        option target 'REJECT'
        option src 'lan'
        list src_mac '[redacted]'
```

IVCrossway, 2024-03-31:
Following https://github.com/Ysurac/openmptcprouter/wiki/Block-client-Internet-access does not work.
This is what I created: 
`config rule
        option name 'test1'
        option target 'REJECT'
        option src 'lan'
        list src_ip '192.168.188.208'
        list src_ip '192.168.1.0/24'
        list proto 'tcp'

config rule
        option name 'test2'
        option target 'REJECT'
        list src_ip '192.168.188.208'
        list src_ip '192.168.1.0/24'
        list proto 'all'
        option src '*'`

Ysurac, 2024-03-31:
Can you use the template ? It's not the same on 0.59.1 and 0.60 based on 6.1 kernel.

IVCrossway, 2024-03-31:
I added the specifications in the first post, if that's what you meant.

closed 2024-07-07 by github-actions[bot]