Ysurac/openmptcprouter #3238: OMR aggregating with 1028 but not 1029-test

SpectroHA, 2024-04-01:
## Expected Behavior
I have 2 VPS (vps & vps2) installations with ovhcloud (the 2nd intended to replace the 1st).  Debian 10 and Debian 12 respectively, where vps installed a few years ago is 5.4 and vps2 installed about a month ago is 6.1.  They are both configured on my router and I can switch between which is master successfully.  The expectation is that they would be functionally identical.

## Current Behavior
When vps is master everything works normally, as it has done for a few years.  When vps2 is master traffic does not aggregate, it only passes through the MPTCP master.   "MPTCP Support Check" is successful for vps but not successful (returns -TCPOptionMPTCPCapable) for vps2.
My hunch is that vps2 is not using a mptcp compatible kernel.  I've tried running the install script multiple times, and I have rebooted each time.

On the working vps
`debian@vps-f81e2723:~$ uname -a
Linux vps-f81e2723 5.4.207-mptcp #1 SMP Sun Jul 24 14:39:44 UTC 2022 x86_64 GNU/Linux`

On the broken vps
`debian@vps-c70185dd:~$ uname -a
Linux vps-c70185dd 6.1.0-18-cloud-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.76-1 (2024-02-01) x86_64 GNU/Linux`
 
I examined /boot/grub/grub.cfg on vps2 and there is no menuentry for mptcp, there are only normal and recovery entries for cloud-amd64 for 6.1.0-13 and 6.1.0-18

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - Working OpenMPTCProuter VPS version: 0.1028.5.4.207-mptcp
  - NOT Working OpenMPTCProuter VPS version: 0.1029-test 6.1.0-18-cloud-amd64
  - NOT Workign OpenMPTCProuter VPS provider: OVHCloud
  - OpenMPTCProuter platform: x86_64 virtual machine

Ysurac, 2024-04-01:
It's normal. 0.59.1 with 5.4 kernel can't work with kernel 6.1 on server.

SpectroHA, 2024-04-01:
Thank you.  I assume if I upgrade the router then it should start working with the 6.1 vps and then not work with the 5.4 vps?

Also, is it normal that kernel 6.1 on vps does not have an "mptcp" grub entry?

Ysurac, 2024-04-01:
Yes you can upgrade router to also use 6.1 (you will also need to update the VPS using 5.4 kernel)
For 6.1, default Debian kernel is used, MPTCP is now in Linux kernel by default and enabled in default Debian kernel so the name is not modified.

SpectroHA, 2024-04-01:
That makes sense!  Thank you very much!

closed 2024-04-01 by SpectroHA