Ysurac/openmptcprouter #3239: MANAGEMENT: Client disconnected the lastest VPS

fareign, 2024-04-01:
Apr 02 05:03:16 iZwz9ecl90hv4wg2bwwqqmZ ovpn-tun0[739]: MANAGEMENT: Client disconnected
Apr 02 05:03:16 iZwz9ecl90hv4wg2bwwqqmZ ovpn-tun0[739]: MANAGEMENT: CMD 'status'
Apr 02 05:03:16 iZwz9ecl90hv4wg2bwwqqmZ ovpn-tun0[739]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:46529

Apr 02 05:02:49 iZwz9ecl90hv4wg2bwwqqmZ ovpn-tun0[739]: MANAGEMENT: Client disconnected
Apr 02 05:02:49 iZwz9ecl90hv4wg2bwwqqmZ ovpn-tun0[739]: MANAGEMENT: CMD 'status'
Apr 02 05:02:49 iZwz9ecl90hv4wg2bwwqqmZ ovpn-tun0[739]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:40895

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: snap
  - OpenMPTCProuter VPS version: the lastest 0.1029 test 6.1
  - OpenMPTCProuter VPS provider:aliyun
  - OpenMPTCProuter platform: x86_64
  - Country: cn

fareign, 2024-04-03:
@Ysurac but openvpn ,can aggrate SRT.

Ysurac, 2024-04-04:
It's not related to this.
I will centralize SRT problems in https://github.com/Ysurac/openmptcprouter/issues/3227

closed 2024-04-04 by Ysurac