Ysurac/openmptcprouter #3242: dnsmasq fails to start if OMR-Bypass creates config lines over 1024 characters long

SpectroHA, 2024-04-02:
## Expected Behavior
Entries in OMR-Bypass should not be able to interfere with other critical services running

## Current Behavior

Entires in OMR-Bypass result in updates to /var/etc/dnsmasq.conf.cfg###### where I suspect the # characters will be different for different installations.  In my installation the full filename is dnsmasq.conf.cfg01411c.  The updates look to be lines that begin with "nftset="  I have found that if enough entries are added in OMR-Bypass such that one of these lines (inclusive of nftset=) exceeds 1024 characters then dnsmasq will fail to start with errors like thie following
`Apr  2 15:50:08 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr  2 15:50:08 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Apr  2 15:50:09 OpenMPTCProuter user.notice omr-bypass: Restart dnsmasq...
Apr  2 15:50:09 OpenMPTCProuter daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Apr  2 15:50:09 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Apr  2 15:50:09 OpenMPTCProuter daemon.crit dnsmasq[1]: bad option at line 48 of /var/etc/dnsmasq.conf.cfg01411c
Apr  2 15:50:09 OpenMPTCProuter daemon.crit dnsmasq[1]: FAILED to start up
Apr  2 15:50:14 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
Apr  2 15:50:14 OpenMPTCProuter daemon.crit dnsmasq[1]: bad option at line 48 of /var/etc/dnsmasq.conf.cfg01411c
Apr  2 15:50:14 OpenMPTCProuter daemon.crit dnsmasq[1]: FAILED to start up
Apr  2 15:50:19 OpenMPTCProuter daemon.crit dnsmasq[1]: bad option at line 48 of /var/etc/dnsmasq.conf.cfg01411c
Apr  2 15:50:19 OpenMPTCProuter daemon.crit dnsmasq[1]: FAILED to start up`

## Possible Solution

I think there's significance to the fact that this occurs when a line exceeds 1024 characters, since that is a clean cutoff at 2^10

## Steps to Reproduce the Problem

  1. start monitoring DNS i.e. 'watch nslookup yahoo.com 192.168.100.1'
  2. Making sure to configure for the same interface (specified or unspecified), add the following protocols/services in OMR-Bypass: amazonvideo, disneyplus, hulu, netflix, teams, apple
  3. monitor dnsmasq config i.e. 'watch tail dnsmasq.conf.cfg01411c'
  4. Observe DNS queries will being to fail when the nftset= line is modified with the above protocols/services
  5. See syslog that dnsmasq failed to start 

## Context (Environment)

The issue is that this breaks DNS on OMR, but using OMR as DNS provider is necessary for OMR-Bypass to work. 

## Specifications

  - OpenMPTCProuter version: 0.60rc2-6.1
  - OpenMPTCProuter VPS version: 0.1029-test 6.1.0-18-cloud-amd64
  - OpenMPTCProuter VPS provider: OVHCloud
  - OpenMPTCProuter platform: x86_64) virtual machine
  - Country: USA

closed 2024-04-02 by Ysurac

Ysurac, 2024-04-02:
This is fixed now, will be in next snapshot image.

SpectroHA, 2024-04-02:
That was fast!  Thank you!