Ysurac/openmptcprouter #3247: NAT Loopback not working

KaWiSo, 2024-04-03:
## Expected Behavior
I am using nginx reverse proxy for making certain webservices available via internet.
The ports 443 and 80 are forwarded to 4443 and 8080 on nginx proxy manager and the services are all reachable via internet, even without an additional snat rule.
NAT-Loopback should make these services available to be reached on the same external domains without leaving the lan.

## Current Behavior
Regardless of what I configure I always get connection closed when accessing the services from lan.
It does not matter if I use external or internal address for NAT Loopback.
I think that the problem is likely, that the OMR is not directly connected to the WAN-IP of the VPS thats used for accessing these webservices, so it cant successfully do NAT-Loopback.
It also seems like all traffic, thats directed at the VPS-IP gets offloaded to the local wan directly, instead of going through the tunnel. 
Even if I disable NAT Loopback, I dont see any 443-Traffic on tun0 if I try to access one of the services.
Contrary I instantly see 443 traffic when accessing from another connection or even with a client thats directly connected to wan1-network.

I cant really debug the traffic flow for tcp correctly as I only get 1 single hop when doing a tcp traceroute to any web-address.
Is that expected behaviour because of using shadowsocks?

Thanks in advance!

## Context (Environment)
OMR on Unraid as x86_64-VM
2 WANs connected via VLANs (VSDL+5G)
Both WAN-Routers do NAT themselves, I am unable to bridge the WAN IP directly to OMR.
Router VDSL: FritzBox 7590ax
Router 5G: ZTE MC801a

## Specifications

  - OpenMPTCProuter version: v0.60rc2-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 0.1029-test 6.1.0-18
  - OpenMPTCProuter VPS provider: webtropia
  - OpenMPTCProuter platform: x86_64
  - Country: Germany

Ysurac, 2024-04-03:
All traffic to VPS IP use a WAN directly without using proxy or VPN. It's for now expected.
You can set a custom DNS resolution via dnsmasq for example.

KaWiSo, 2024-04-03:
Thanks, thats the workaround I got working for now.

closed 2024-07-08 by github-actions[bot]