Ysurac/openmptcprouter #3248: BPI Support

Yessine1200, 2024-04-03:
Are the Banana Pi BPI-R64 and Banana Pi BPI-R2 Pro supported in the latest version (0.6rc2) of OMR? If so, which one do you recommend using?

Yessine1200, 2024-04-03:
And BPI-R2, not tested also ? If I purchase one, what are the chances of successfully installing OMR on it?

Ysurac, 2024-04-03:
BPI-R2 not tested also, older than BPI-R64.
What is your need ? Which speed ? 4G/5G ? small ? cheap ?

Yessine1200, 2024-04-03:
My need is to aggregate various types of links (fiber, 4G/5G, xDSL, etc.) for clients, without focusing on any specific scenario. Currently, I am exploring suitable hardware to install OMR on, and cost is not a concern as long as the hardware is capable of running OMR effectively.

Ysurac, 2024-04-03:
Best hardware is based on x86_64 with AES hardware support.
Else RPI5 can give up to 950Mb/s (limited by ethernet interface), RPI4 550Mb/s and RPI3 250Mb/s (this can give an idea about speed on arch based on same CPU)
It's always better to have a CPU with AES hardware support.

Yessine1200, 2024-04-05:
Hello Ysurac, 
RPI devices are known to have stability issues over the long term. Additionally, one of my requirements is to find hardware that supports two modems, possibly even a miniPC. Do you have any suggestions for a miniPC that supports two modems? Alternatively, do you know of any internal cards with a PCI-X port to which a SIM modem could be added?
Also, I found this BPI similar to RPI-R64 , it seems support 2 modems: [https://wiki.banana-pi.org/5G_%2B_4G_LTE%2BWifi_AC%2BGigabit_Multiplex_aggregate_route]() and also OMR can be installed on it. 
I found this video also : [https://www.youtube.com/watch?v=lhZIa-L85dk](url)

Ysurac, 2024-04-05:
This is still based on BPI-R64.
I can't really help on hardware, I don't have the ideal one...

Yessine1200, 2024-04-05:
Yes, and it supports two modems, correct?
Maybe I will buy one and test it. If it works, I can add it to the documentation.

closed 2024-07-10 by github-actions[bot]