Ysurac/openmptcprouter #3249: Multiple disconnection

greatdie, 2024-04-03:
## Expected Behavior
Connections should stay up

## Current Behavior
I've configured multiple WAN connections but i've continuos disconnection.
On logs:
Wed Apr  3 15:47:31 2024 daemon.err openvpn(omr)[14472]: Connection reset, restarting [0]
Wed Apr  3 15:47:31 2024 daemon.warn openvpn(omr)[14472]: sitnl_send: rtnl: generic error (-3): No such process
Wed Apr  3 15:47:31 2024 daemon.warn openvpn(omr)[14472]: ERROR: Linux route delete command failed
Wed Apr  3 15:47:31 2024 daemon.notice openvpn(omr)[14472]: net_addr_ptp_v4_del: 10.255.252.6 dev tun0
Wed Apr  3 15:47:31 2024 daemon.notice netifd: Network device 'tun0' link is down
Wed Apr  3 15:47:31 2024 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Apr  3 15:47:31 2024 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Apr  3 15:47:31 2024 daemon.notice netifd: Interface 'omrvpn' is disabled
Or:
Wed Apr  3 15:52:41 2024 daemon.err rmmod: module is not loaded
Wed Apr  3 15:52:41 2024 daemon.err rmmod: module is not loaded
Wed Apr  3 15:52:41 2024 daemon.err rmmod: module is not loaded
Wed Apr  3 15:52:41 2024 daemon.info vnstatd[13380]: Info: SIGHUP received, flushing data to disk and reloading config.
Wed Apr  3 15:52:41 2024 user.notice omr-tracker: Launching...
Wed Apr  3 15:52:49 2024 user.notice omr-tracker: Launched
Wed Apr  3 15:52:49 2024 user.notice Shadowsocks: Rules DOWN
Wed Apr  3 15:52:49 2024 authpriv.info dropbear[32054]: Child connection from 185.201.65.86:32205
Wed Apr  3 15:52:49 2024 authpriv.info dropbear[32054]: Exit before auth from <185.201.65.86:32205>: Disconnect received
Wed Apr  3 15:52:49 2024 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Wed Apr  3 15:52:49 2024 user.notice Shadowsocks: Rules UP
Wed Apr  3 15:52:50 2024 user.notice Shadowsocks: Reload omr-bypass rules
Wed Apr  3 15:52:50 2024 user.notice omr-bypass: Starting OMR-ByPass...
Wed Apr  3 15:52:51 2024 user.notice omr-bypass: OMR-ByPass is running
Wed Apr  3 15:52:51 2024 daemon.info /usr/bin/ss-local[1501]: using tcp fast open
Wed Apr  3 15:52:51 2024 daemon.info /usr/bin/ss-local[1501]: enable TCP no-delay
Wed Apr  3 15:52:51 2024 daemon.info /usr/bin/ss-local[1501]: initializing ciphers... aes-256-gcm
Wed Apr  3 15:52:51 2024 daemon.info /usr/bin/ss-local[1501]: listening at 127.0.0.1:1111
Wed Apr  3 15:52:51 2024 daemon.info /usr/bin/ss-local[1501]: tcp port reuse enabled
Wed Apr  3 15:52:51 2024 daemon.info /usr/bin/ss-local[1501]: udprelay enabled
Wed Apr  3 15:52:51 2024 daemon.info /usr/bin/ss-local[1501]: udp port reuse enabled
Wed Apr  3 15:52:51 2024 daemon.info /usr/bin/ss-local[1501]: running from root user
Wed Apr  3 15:52:51 2024 user.info : using tcp fast open

I've tried different settings: BBR2 or cubic, "master interface selection" Balacing or Dynamic change", different type of VPN... But nothing.

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1028
  - OpenMPTCProuter VPS provider: host.it
  - OpenMPTCProuter platform: x86_64 mini pc

Ysurac, 2024-04-03:
It's too partial log to find anything that can help to diagnose. All is green in System->OpenMPTCProuter, Status page ?
You should update to 0.60rc2 and VPS 0.1029

greatdie, 2024-04-03:
Yes, it's all green (only 1 wan it's red but i've disable it from multipath).
How can i update to newest version VPS and OMR?
Here it's a complete System Log:
Wed Apr  3 15:35:42 2024 daemon.notice netifd: Network device 'tun0' link is down
Wed Apr  3 15:35:42 2024 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Apr  3 15:35:42 2024 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Apr  3 15:35:42 2024 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Apr  3 15:35:42 2024 daemon.err /usr/bin/ss-redir[11365]: remote recv: Connection reset by peer
Wed Apr  3 15:35:42 2024 daemon.err /usr/bin/ss-redir[11365]: remote recv: Connection reset by peer
Wed Apr  3 15:35:42 2024 daemon.err /usr/bin/ss-redir[11365]: remote recv: Connection reset by peer
Wed Apr  3 15:35:42 2024 daemon.notice openvpn(omr)[14472]: /usr/libexec/openvpn-hotplug down omr tun0 1500 1626 10.255.252.6 10.255.252.5 init
Wed Apr  3 15:35:42 2024 daemon.notice openvpn(omr)[14472]: SIGUSR1[soft,connection-reset] received, process restarting
Wed Apr  3 15:35:45 2024 user.notice post-tracking-post-tracking: omrvpn down
Wed Apr  3 15:35:46 2024 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.192.168 dev pppoe-Netoip
Wed Apr  3 15:35:47 2024 daemon.warn openvpn(omr)[14472]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Wed Apr  3 15:35:47 2024 daemon.warn openvpn(omr)[14472]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Wed Apr  3 15:35:47 2024 daemon.notice openvpn(omr)[14472]: TCP/UDP: Preserving recently used remote address: [AF_INET]185.201.65.86:65301
Wed Apr  3 15:35:47 2024 daemon.notice openvpn(omr)[14472]: Attempting to establish TCP connection with [AF_INET]185.201.65.86:65301 [nonblock]
Wed Apr  3 15:35:47 2024 daemon.notice openvpn(omr)[14472]: TCP connection established with [AF_INET]185.201.65.86:65301
Wed Apr  3 15:35:47 2024 daemon.notice openvpn(omr)[14472]: TCP_CLIENT link local: (not bound)
Wed Apr  3 15:35:47 2024 daemon.notice openvpn(omr)[14472]: TCP_CLIENT link remote: [AF_INET]185.201.65.86:65301
Wed Apr  3 15:35:47 2024 daemon.warn openvpn(omr)[14472]: WARNING: 'link-mtu' is used inconsistently, local='link-mtu 1551', remote='link-mtu 1559'
Wed Apr  3 15:35:47 2024 daemon.warn openvpn(omr)[14472]: WARNING: 'auth' is used inconsistently, local='auth [null-digest]', remote='auth SHA1'
Wed Apr  3 15:35:47 2024 daemon.notice openvpn(omr)[14472]: [server] Peer Connection Initiated with [AF_INET]185.201.65.86:65301
Wed Apr  3 15:35:48 2024 daemon.notice openvpn(omr)[14472]: TUN/TAP device tun0 opened
Wed Apr  3 15:35:48 2024 daemon.notice openvpn(omr)[14472]: net_iface_mtu_set: mtu 1500 for tun0
Wed Apr  3 15:35:48 2024 daemon.notice openvpn(omr)[14472]: net_iface_up: set tun0 up
Wed Apr  3 15:35:48 2024 daemon.notice openvpn(omr)[14472]: net_addr_ptp_v4_add: 10.255.252.6 peer 10.255.252.5 dev tun0
Wed Apr  3 15:35:48 2024 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Apr  3 15:35:48 2024 daemon.notice netifd: Network device 'tun0' link is up
Wed Apr  3 15:35:48 2024 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Apr  3 15:35:48 2024 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Apr  3 15:35:48 2024 daemon.notice openvpn(omr)[14472]: /usr/libexec/openvpn-hotplug up omr tun0 1500 1626 10.255.252.6 10.255.252.5 init
Wed Apr  3 15:35:48 2024 daemon.notice netifd: Interface 'omrvpn' is now up
Wed Apr  3 15:35:48 2024 daemon.warn [8526]: <warn>  [SQM_IFB_633b4] invalid sysfs path read for net/SQM_IFB_633b4
Wed Apr  3 15:35:48 2024 daemon.warn [8526]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_633b4 not found
Wed Apr  3 15:35:48 2024 daemon.warn [8526]: <warn>  [SQM_IFB_8face] invalid sysfs path read for net/SQM_IFB_8face
Wed Apr  3 15:35:48 2024 daemon.warn [8526]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_8face not found
Wed Apr  3 15:35:48 2024 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Wed Apr  3 15:35:48 2024 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Wed Apr  3 15:35:50 2024 user.notice post-tracking-post-tracking: Set firewall on server vps
Wed Apr  3 15:35:53 2024 daemon.notice openvpn(omr)[14472]: Initialization Sequence Completed
Wed Apr  3 15:35:53 2024 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Wed Apr  3 15:35:59 2024 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.252.5 dev tun0
Wed Apr  3 15:37:54 2024 daemon.err /usr/bin/ss-redir[11365]: remote recv: Connection reset by peer
Wed Apr  3 15:38:05 2024 daemon.err /usr/bin/ss-redir[11365]: remote recv: Connection reset by peer
Wed Apr  3 15:38:09 2024 daemon.err /usr/bin/ss-redir[11365]: remote recv: Connection reset by peer
Wed Apr  3 15:38:46 2024 user.notice omr-tracker: Launching...
Wed Apr  3 15:38:53 2024 user.notice omr-tracker: Launched
Wed Apr  3 15:38:53 2024 user.notice omr-bypass: Starting OMR-ByPass...
Wed Apr  3 15:38:54 2024 user.notice omr-bypass: OMR-ByPass is running
Wed Apr  3 15:39:14 2024 daemon.err /usr/bin/ss-redir[11365]: remote recv: Connection reset by peer
Wed Apr  3 15:40:16 2024 daemon.err /usr/bin/ss-redir[11365]: remote recv: Connection reset by peer
[... cut ...]
Wed Apr  3 17:38:57 2024 daemon.err /usr/bin/ss-redir[15216]: remote recv: Connection reset by peer
Wed Apr  3 17:39:09 2024 daemon.err /usr/bin/ss-redir[15216]: remote recv: Connection reset by peer
Wed Apr  3 17:39:44 2024 daemon.err /usr/bin/ss-redir[15216]: remote recv: Connection reset by peer
Wed Apr  3 17:41:18 2024 daemon.err /usr/bin/ss-redir[15216]: remote recv: Connection reset by peer
Wed Apr  3 17:41:34 2024 daemon.err /usr/bin/ss-redir[15216]: remote recv: Connection reset by peer
Wed Apr  3 17:42:01 2024 user.info : luci: accepted login on / for root from 10.10.14.197

Ysurac, 2024-04-03:
To update: https://github.com/Ysurac/openmptcprouter/issues/2961
And revert "master interface selection" to balancing.

greatdie, 2024-04-03:
Ok, i've installed it on Proxmox, so the command for update is:
wget https://download.openmptcprouter.com/release/v0.60rc2-5.4/x86_64/targets/x86/64/openmptcprouter-v0.60rc2-5.4-r0+16862-170d9e447d-x86-64-generic-squashfs-combined-efi.img.gz
Correct?
After install new version i'll testing it.
Thanks.

greatdie, 2024-04-04:
Can i update from 5.4 to 6.1 with a non efi image or not?

Ysurac, 2024-04-04:
Yes if you disable EFI on the VM. But if it's working you can keep it, not a real problem, only that EFI make sometimes problem at boot (but if it boot, then it's ok)

greatdie, 2024-04-04:
Ok, i'll update directly from 5.4 to 6.1 and i'll tell you if the problem persist or not.
Thanks a lot!

greatdie, 2024-04-04:
I've problem with new version on recognize phisical LAN on my mini pc (it has 6 gigabit LAN).
It's possible via cmd to set a DHCP on eth0?

Ysurac, 2024-04-04:
DHCP server or DHCP client ?
Default is DHCP server and static ip 192.168.100.1
To set LAN as DHCP client:
```uci del network.lan.ipaddr && uci del network.lan.netmask && uci set network.lan.proto='dhcp' && uci commit network && /etc/init.d/network restart```

greatdie, 2024-04-05:
How can i update a Proxmox instance directly from 0.59 5.4 to 6.0 6.1?

Ysurac, 2024-04-05:
Like any others device: via System->Flash Firmware, "Flash Image" button and selecting a sysupgrade image.

greatdie, 2024-04-05:
Sorry but I've never done this... Where can i find sysupgrade image? It's .gz file (https://download.openmptcprouter.com/release/v0.60rc2-6.1/x86_64/targets/x86/64/openmptcprouter-v0.60rc2-6.1-r0+24843-acf40c022e-x86-64-generic-squashfs-combined.img.gz)?

Ysurac, 2024-04-08:
Yes

greatdie, 2024-04-09:
I've installed new version without Proxmox and now seems to work correctly.
With Proxmox i had multiple disconnection of eth2 (i've other 2 installed version with Proxmox, but none has this problem...), now it works.

closed 2024-07-14 by github-actions[bot]