Ysurac/openmptcprouter #3251: OMR-Bypass issues with MAC-Address filtering

jackwakefield, 2024-04-05:
## Expected Behavior
Adding the following rule would route that device's traffic through lan1 (wan1).

[image]

## Current Behavior

When I view my IP address through ipify.org or whatismyip.com, I can see the traffic for that device is still being routed through the VPS.

## Possible Solution

## Steps to Reproduce the Problem

  1. Create a wireless network under Network -> Wireless with "Network" set to "lan"
  2. Go to Network -> Interfaces and allow OpenWRT to create the br-lan interface
  3. Go to Services -> OMR-Bypass and add a MAC-address filter for a device
  4. Connect to WiFi using that device

## Context (Environment)

I have 2 WAN interfaces configured, and using XRay VLESS as the default proxy.

`nft list ruleset`: https://gist.github.com/jackwakefield/404d2649ff6dfae4d5c0b6e396f298c5

`uci show network`: https://gist.github.com/jackwakefield/e927329e44371cb69b44aa1eca545a25

`uci show firewall`: https://gist.github.com/jackwakefield/b8933d2017094f20b880790f42e73817

`uci show omr-bypass`: https://gist.github.com/jackwakefield/9d96f62c102e3ac2cb7270feb9d70257

`uci show openmptcprouter.*.metric`: https://gist.github.com/jackwakefield/1fc1a0045a09e6622beb10f1ea7f68b6

## Specifications

  - OpenMPTCProuter version: https://github.com/Ysurac/openmptcprouter/commit/1e7d76672c8e158b99cd3f797ff45768e5739df7
  - OpenMPTCProuter VPS version: https://github.com/Ysurac/openmptcprouter-vps/commit/96775408d7d40e917f190cf9d60744f0bcb83fdb
  - OpenMPTCProuter VPS provider: ionos.co.uk
  - OpenMPTCProuter platform: GL.iNet GL-MT6000 (Flint 2)
  - Country: United Kingdom

closed 2024-04-05 by Ysurac

jackwakefield, 2024-04-06:
That's fixed it. Thanks for all your work on this project, it's truly incredible.

kpma1985, 2024-04-18:
@jackwakefield may u share the image? i have a mt6000,too but it always fail at compile