Ysurac/openmptcprouter #3252: Less data consuming as possible, if possible.

aladdinshammr, 2024-04-05:
**Is your feature request related to a problem? Please describe.**

I tested OMR and it worked well but as I am in one of stone age countries that restrict the amount of data  the user can consume ( bandwidth cap ) I ended up paying the bill twice this month.
for example, when I download 10G, I see 20G deducted from my plan.

**Describe the solution you'd like**

there would be many solutions, The difficulty lies in whether it can be applied or not.
- if OMR can aggregate connections for those who want the aggregation to enhance internet speed and are not keen on MPTCP.
- or redirect the necessary ports if the MPTCP is critical for the aggregation to work.
- (I think this is impossible but I will mention it anyway) maybe OMR can do data compression or something like that, so we can escape from data cap restrictions.

**Describe alternatives you've considered**

I am currently using load balance , but things get bad when you have connections with different characteristics.

**Additional context**

I do appreciate your time and effort you invested on this project, it's great for those people who live in modern times.

vempire-ghost, 2024-04-05:
Check if you are not using the redundant scheduler in the MPTCP settings. If you are, change it to another one because it doubles the bandwidth usage.

aladdinshammr, 2024-04-05:
@vempire-ghost I did not change anything in the default setting .
does it work good for you?
you did not notice any increase in the bandwidth usage?

Ysurac, 2024-04-05:
Please use the full template. What is the OpenMPTCProuter version you use ?
You can configure OMR-Tracker to check the connections less often.

aladdinshammr, 2024-04-05:
@Ysurac edited, Although the template did not ask for the router version 😢

Ysurac, 2024-04-05:
@aladdinshammr yes as feature request there is no version request. Maybe I should add that sorry.
You should try 0.60rc2 it should call the VPS less often so use also less data.

aladdinshammr, 2024-04-06:
@Ysurac

aladdinshammr, 2024-04-06:
Thanks you a lot, I will try that..
so there is no way to aggregate connections without a VPS??

aladdinshammr, 2024-04-06:
@vempire-ghost 
I am referring to my ISP, not to the VPS.
when I use MPTCP I see a noticeable difference between how much my ISP gives me and how much I consume.
say for example I consume 70G, my ISP charges me for 100G!!
I wish you got what I mean.

is it the same with the VPS??
I think it is only about the resources you choose, I did not think about the traffic

Ysurac, 2024-04-06:
@aladdinshammr the VPS is here to get all connections packets, aggregate them and give Internet access with this full connection. So no way without a VPS. The only way would be that all website you use support MPTCP, not yet the case.

aladdinshammr, 2024-04-06:
@Ysurac 
what is the cause of this error?

Apr  6 10:00:58 OpenMPTCProuter user.notice cake-autorate.unknown: ERROR: 1712397658.920114 The value of 'dl_owd_delta_thr_ms' in config file: '/root/cake-autorate/config.wan2.sh' is not a valid value of type: 'float'.
Apr  6 10:00:58 OpenMPTCProuter user.notice cake-autorate.unknown: ERROR: 1712397658.927539 The value of 'dl_owd_delta_thr_ms' in config file: '/root/cake-autorate/config.wan1.sh' is not a valid value of type: 'float'.
Apr  6 10:00:58 OpenMPTCProuter user.notice cake-autorate.unknown: ERROR: 1712397658.963967 The value of 'ul_owd_delta_thr_ms' in config file: '/root/cake-autorate/config.wan2.sh' is not a valid value of type: 'float'.
Apr  6 10:00:58 OpenMPTCProuter user.notice cake-autorate.unknown: ERROR: 1712397658.970475 The value of 'ul_owd_delta_thr_ms' in config file: '/root/cake-autorate/config.wan1.sh' is not a valid value of type: 'float'.
Apr  6 10:00:59 OpenMPTCProuter user.notice cake-autorate.unknown: ERROR: 1712397659.004871 The value of 'dl_avg_owd_delta_thr_ms' in config file: '/root/cake-autorate/config.wan2.sh' is not a valid value of type: 'float'.
Apr  6 10:00:59 OpenMPTCProuter user.notice cake-autorate.unknown: ERROR: 1712397659.012183 The value of 'dl_avg_owd_delta_thr_ms' in config file: '/root/cake-autorate/config.wan1.sh' is not a valid value of type: 'float'.
Apr  6 10:00:59 OpenMPTCProuter user.notice cake-autorate.unknown: ERROR: 1712397659.044625 The value of 'ul_avg_owd_delta_thr_ms' in config file: '/root/cake-autorate/config.wan2.sh' is not a valid value of type: 'float'.
Apr  6 10:00:59 OpenMPTCProuter daemon.warn openvpn(omr)[12678]: sitnl_send: rtnl: generic error (-17): File exists
Apr  6 10:00:59 OpenMPTCProuter user.notice cake-autorate.unknown: ERROR: 1712397659.053101 The value of 'ul_avg_owd_delta_thr_ms' in config file: '/root/cake-autorate/config.wan1.sh' is not a valid value of type: 'float'

even though the wizard accept only integer in download and upload speed when I check "SQM autorate" 

router version :  0.60rc2

vempire-ghost, 2024-04-06:
> @vempire-ghost I am referring to my ISP, not to the VPS. when I use MPTCP I see a noticeable difference between how much my ISP gives me and how much I consume. say for example I consume 70G, my ISP [...]

Google Cloud, which I use, charges for egress; it depends on the VPS service as they have different charges. As for your internet service provider, they usually sum up the download and upload to count your quota. See if your consumption, summed between upload and download, doesn't approach the value charged by them.

closed 2024-04-06 by Ysurac

aladdinshammr, 2024-04-07:
@vempire-ghost 

sure upload and download are included.
maybe it's because I always get disconnected from the VPS.
Do you think if my ISP blocked shadowsocks therefore I always get disconnected from the VPS???
I also use google cloud service, I noticed that they charge me more when there is a heavy load on the CPU.
Does that happen to you as well???

vempire-ghost, 2024-04-07:
> @vempire-ghost
>
> sure upload and download are included. maybe it's because I always get disconnected from the VPS. Do you think if my ISP blocked shadowsocks therefore I always get disconnected from [...]

I can't say for sure; I don't think it's a blockage, but I prefer to use Xray as a proxy; it gave me the best performance. 
Google charges a fixed amount per hour, at least in my VPS configuration. I don't know about yours, but they've never charged me for CPU usage, only for outgoing internet traffic.

aladdinshammr, 2024-04-07:
@vempire-ghost 

What made me think of that is when I try to browse "https://shadowsocks.org" , it won't open unless I'm using a VPN.
so I thought maybe there is a filtering or something behind the scene

vempire-ghost, 2024-04-07:
> @vempire-ghost
> 
> What made me think of that is when I try to browse "https://shadowsocks.org" , it won't open unless I'm using a VPN. so I thought maybe there is a filtering or something behind the scene

I imagine that blocking the address is easier than trying to block the traffic that Shadowsocks emits. If it drops more easily, it might be an internet routing issue.

aladdinshammr, 2024-04-07:
@vempire-ghost 
forgive me for bothering you, but this is the last question.
which one should I choose for this xRay (xray socks - xray trojan ................)

vempire-ghost, 2024-04-08:
> @vempire-ghost forgive me for bothering you, but this is the last question. which one should I choose for this xRay (xray socks - xray trojan ................)

Don't mention it :)

If you live in a country that tends to block content, Xray VLESS Reality is the most recommended, but all the others also work to bypass these restrictions.

aladdinshammr, 2024-04-08:
@vempire-ghost 
Thank you a million.
I'll try that