Ysurac/openmptcprouter #3255: Routing VPS traffic to Residential IP Proxy

danausx, 2024-04-07:
Routing to a standard cheap VPS alone usually cripples the overall internet usage experience as the IP it gives are mostly blacklisted by mayor service like Netflix, Reddit and many other day to day sites seem to behave erratically when they detect these types of IPs. It has gotten worse with web scraping being even more common nowadays with AI chatbots.

Bypassing these sites might seem helpful but it adds a whole lot of manual setup overall and likely wont cover most edge cases.

Is there any way to route the VPS traffic through another residential IP proxy and thus obtaining a clean ip address in the process?

I am not fully aware of how the VPS operates but I have been wondering if such thing is even possible, maybe with custom routing, not sure of the latency impact as well but might still be worth it in some cases. Probably TCP/UDP might work but ICMP wouldn't maybe?

Residential IP VPS are quire rare and a lot more expensive as well. Any input on this?

Thanks!

Ysurac, 2024-04-07:
You need to find a provider that have IPs not blacklisted, you can use OMR-ByPass that will exit directly via a WAN or use the new exit via a VPN set on the VPS that I need to document.

Ysurac, 2024-04-08:
I mean exit via a VPN on the VPS. So if correctly configured, it will use this as exit.
But if you can put a wireguard exit on a residential IP, why not put a VPS here ?

danausx, 2024-04-08:
> I mean exit via a VPN on the VPS. So if correctly configured, it will use this as exit.

Got it!

> But if you can put a wireguard exit on a residential IP, why not put a VPS here ?

There is a Wireguard VPN offered as VPN only service at a way cheaper price than getting the whole VPS and includes Residential IP. I believe Ivacy does this too. Many of the cheaper residential IP VPS are out of stock and the only ones left even raised prices recently. Even notoriously hard to route services like Netflix seem to work with these type of IPs.

Oracle Cloud free tier is a very good option to use this with as its a 4c/24GB Ampere Arm instance with 200GB storage and unlimited ingress bandwidth.

nobodyatroot, 2024-05-02:
> use the new exit via a VPN set on the VPS that I need to document.

@Ysurac do you have any more info on this? i'm assuming it's some additional source policy routing that needs to be configured on the VPS? i've noticed more and more sites are starting to crackdown on visitors with datacenter IPs, so this would be very useful information. thanks!

sojibhossain171, 2024-07-09:
@danausx I know a provider who provides residential VPS and residential proxy services I am a user of them and their VPS quality is good. you can contact them and ask them how they maintain that thing. the mentioned website is ipmela.com

closed 2024-10-13 by github-actions[bot]