Ysurac/openmptcprouter #3259: Z8102ax

wouam31200, 2024-04-09:
Hi,
I just saw that there is starting to be a version for the Z8102ax.
I tried to install it on mine, but I think I'm not doing it correctly, and the file name seems weird to me.

[openmptcprouter-v0.60beta1-5.4-r0+16862-170d9e447d-mediatek-mt7622-bpi_bananapi-r64-squashfs-sysupgrade](https://snapshots.openmptcprouter.com/5.4/z8102ax_128m/targets/mediatek/mt7622/)

Attached is the latest serial log.

```
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258000
[    0.000000] Kernel command line: earlycon=uart8250,mmio32,0x11002000 console=ttyS0,115200n1 swiotlb=512
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1017852K/1048384K available (7038K kernel code, 440K rwdata, 1992K rodata, 384K init, 300K bss, 30532K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC CPU mask not found - kernel will fail to boot.
[    0.000000] ------------[ cut here ]------------
[    0.000000] GICv3 system registers enabled, broken firmware!
[    0.000000] WARNING: CPU: 0 PID: 0 at gic_cpu_init+0x184/0x190
[    0.000000] Modules linked in:
[    0.000000] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.238 #0
[    0.000000] Hardware name: Bananapi BPI-R64 (DT)
[    0.000000] pstate: 60000085 (nZCv daIf -PAN -UAO)
[    0.000000] pc : gic_cpu_init+0x184/0x190
[    0.000000] lr : gic_cpu_init+0x184/0x190
[    0.000000] sp : ffffffc0109c3d70
[    0.000000] x29: ffffffc0109c3d70 x28: 0000000044960018
[    0.000000] x27: 0000000000000003 x26: ffffff803fdec7f8
[    0.000000] x25: ffffffc01035d580 x24: ffffffc01035d750
[    0.000000] x23: ffffffc010005000 x22: ffffffc01000d000
[    0.000000] x21: 0000000000000000 x20: ffffffc0109c9c90
[    0.000000] x19: ffffffc0109c9c90 x18: 000000000000003f
[    0.000000] x17: 000000000000000a x16: 0000000000000010
[    0.000000] x15: 0000000000000c00 x14: ffffffc010a36d98
[    0.000000] x13: 0000000000000000 x12: ffffffc010a36000
[    0.000000] x11: ffffffc0109d7000 x10: 0000000000000010
[    0.000000] x9 : 0000000000000000 x8 : 6b6f7262202c6465
[    0.000000] x7 : 6c62616e65207372 x6 : 0000000000000022
[    0.000000] x5 : 000000000000000d x4 : 0000000000000000
[    0.000000] x3 : 0000000000000000 x2 : 00000000ffffffff
[    0.000000] x1 : ffffffc0109d7dd8 x0 : 0000000000000030
[    0.000000] Call trace:
[    0.000000]  gic_cpu_init+0x184/0x190
[    0.000000]  gic_init_bases+0xdc/0x178
[    0.000000]  __gic_init_bases+0x1bc/0x1e8
[    0.000000]  gic_of_init+0x274/0x344
[    0.000000]  of_irq_init+0x174/0x29c
[    0.000000]  irqchip_init+0x14/0x1c
[    0.000000]  init_IRQ+0xb4/0xe8
[    0.000000]  start_kernel+0x298/0x410
[    0.000000] ---[ end trace ca47383eecc8c06e ]---
[    0.000000] GIC CPU mask not found - kernel will fail to boot.
[    0.000000] arch_timer: cp15 timer(s) running at 13.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns
[    0.000003] sched_clock: 56 bits at 13MHz, resolution 76ns, wraps every 4398046511101ns
[    0.008249] Calibrating delay loop (skipped), value calculated using timer frequency.. 26.00 BogoMIPS (lpj=52000)
[    0.018562] pid_max: default: 32768 minimum: 301
[    0.023286] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.030718] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.039492] ASID allocator initialised with 65536 entries
[    0.044969] rcu: Hierarchical SRCU implementation.
[    0.050155] smp: Bringing up secondary CPUs ...
```

If further manipulation needs to be done, I would be delighted.

Ysurac, 2024-04-09:
This is not the good image, the good image is https://snapshots.openmptcprouter.com/6.1/z8102ax_128m/targets/mediatek/filogic/openmptcprouter-v0.60-snapshot-6.1-r0+24843-acf40c022e-mediatek-filogic-z8102ax-128m-squashfs-sysupgrade.bin

wouam31200, 2024-04-09:
I tried :

[openmptcprouter-v0.60rc2-6.1-r0+24843-acf40c022e-mediatek-filogic-z8102ax-128m-squashfs-sysupgrade](https://download.openmptcprouter.com/release/v0.60rc2-6.1/z8102ax_128m/targets/mediatek/filogic/)

```
[   15.786197] usbcore: registered new interface driver usb_serial_simple
[   15.792034] mtdblock: MTD device 'Factory' is NAND, please consider using UBI block devices instead.
[   15.792867] usbserial: USB Serial support registered for carelink
[   15.808001] usbserial: USB Serial support registered for flashloader
[   15.814438] usbserial: USB Serial support registered for funsoft
[   15.820544] usbserial: USB Serial support registered for google
[   15.826525] usbserial: USB Serial support registered for hp4x
[   15.832342] usbserial: USB Serial support registered for kaufmann
[   15.838485] usbserial: USB Serial support registered for libtransistor
[   15.845079] usbserial: USB Serial support registered for moto_modem
[   15.851413] usbserial: USB Serial support registered for motorola_tetra
[   15.858076] usbserial: USB Serial support registered for nokia
[   15.863971] usbserial: USB Serial support registered for novatel_gps
[   15.870386] usbserial: USB Serial support registered for siemens_mpi
[   15.876786] usbserial: USB Serial support registered for suunto
[   15.882771] usbserial: USB Serial support registered for vivopay
[   15.888849] usbserial: USB Serial support registered for zio
[   15.900658] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[   15.908482] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[   15.938344] usbcore: registered new interface driver cdc_mbim
[   15.945353] usbcore: registered new interface driver ipw
[   15.950773] usbserial: USB Serial support registered for IPWireless converter
[   15.960968] usbcore: registered new interface driver option
[   15.966619] usbserial: USB Serial support registered for GSM modem (1-port)
[   15.976853] usbcore: registered new interface driver qcserial
[   15.982698] usbserial: USB Serial support registered for Qualcomm USB modem
[   15.998381] usbcore: registered new interface driver rt2800usb
[   16.016439] usbcore: registered new interface driver rtl8192cu
[   16.031116] usbcore: registered new interface driver ath9k_htc
[   16.037308] kmodloader: done loading kernel modules from /etc/modules.d/*
[   62.572796] gpio-457 (watchdog): gpiod_request: status -16
[   63.602728] gpio-457 (watchdog): gpiod_request: status -16
[   64.619988] gpio-457 (watchdog): gpiod_request: status -16
[   65.632589] gpio-457 (watchdog): gpiod_request: status -16
[   66.644065] gpio-457 (watchdog): gpiod_request: status -16
[   67.655444] gpio-457 (watchdog): gpiod_request: status -16
[   68.666890] gpio-457 (watchdog): gpiod_request: status -16
[   69.687144] gpio-457 (watchdog): gpiod_request: status -16
[   70.709125] gpio-457 (watchdog): gpiod_request: status -16
[   71.736824] gpio-457 (watchdog): gpiod_request: status -16
[   72.753498] gpio-457 (watchdog): gpiod_request: status -16
[   73.765558] gpio-457 (watchdog): gpiod_request: status -16
[   74.778394] gpio-457 (watchdog): gpiod_request: status -16
[   75.790429] gpio-457 (watchdog): gpiod_request: status -16
[   76.802341] gpio-457 (watchdog): gpiod_request: status -16
[   77.815009] gpio-457 (watchdog): gpiod_request: status -16
[   78.827044] gpio-457 (watchdog): gpiod_request: status -16
[   79.838984] gpio-457 (watchdog): gpiod_request: status -16
[   80.850487] gpio-457 (watchdog): gpiod_request: status -16
[   81.861936] gpio-457 (watchdog): gpiod_request: status -16
[   82.873414] gpio-457 (watchdog): gpiod_request: status -16
[   83.884814] gpio-457 (watchdog): gpiod_request: status -16
[   84.896248] gpio-457 (watchdog): gpiod_request: status -16
[   85.907664] gpio-457 (watchdog): gpiod_request: status -16
[   86.919304] gpio-457 (watchdog): gpiod_request: status -16
[   87.930722] gpio-457 (watchdog): gpiod_request: status -16
```

Ysurac, 2024-04-09:
There is no error here. Press "Enter" to see the login.

wouam31200, 2024-04-09:
Thank you for the answer!
I just tested the correct version.

On the wan in dhcp, I cannot access 192.168.100.1 (neither by the web nor by ping)

I blocked my network card's ip address to 192.168.100.10 and try to ping 192.168.100.1 on all ports (lan and wan)
I have nothing that answers.

Ysurac, 2024-04-09:
Check via ```ip a``` what you have on the ZBT

Ysurac, 2024-04-09:
I just checked with the image with a fresh config on a Z8102AX, and I have correctly "192.168.100.1" set on eth1 interface, it's WAN interface. I can access the web interface.

wouam31200, 2024-04-10:
Hi,
I just tried again, I think I have a little problem.
I will try to put the stock firmware and put openmptcp back.
I'll keep you informed.

```
1: lo: <LOOPBACK> mtu 65536 qdisc noop state DOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP group default qlen 1000
    link/ether f8:5e:3c:4c:72:4e brd ff:ff:ff:ff:ff:ff
    inet6 fe80::fa5e:3cff:fe4c:724e/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether f8:5e:3c:4c:72:50 brd ff:ff:ff:ff:ff:ff
4: lan1@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether f8:5e:3c:4c:72:4e brd ff:ff:ff:ff:ff:ff
5: lan2@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether f8:5e:3c:4c:72:4e brd ff:ff:ff:ff:ff:ff
6: lan3@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether f8:5e:3c:4c:72:4e brd ff:ff:ff:ff:ff:ff
7: lan4@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether f8:5e:3c:4c:72:4e brd ff:ff:ff:ff:ff:ff
8: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr 425c:2768:9e0b::
9: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
10: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
11: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
12: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
13: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr 3af4:479a:f5e3::
14: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
root@(none):/
```

Ysurac, 2024-04-10:
When you update from ZBT OpenWRT you must not keep settings. This may be the issue...

wouam31200, 2024-04-12:
Hi,
I think I understand the problem.
I had installed a firmware ROOTER, except when I tried to install Openmptcp it didn't work.

I put the firmware back to stock then installed Openmptcp.

At the beginning, I had duplicate lans (lan1 x 2, lan 2 x 2...)

So, I re-installed openmptcp in tftp, and magically everything was back to normal.

wouam31200, 2024-04-12:
Hi,

I wanted to update to:
openmptcprouter-v0.61-snapshot-6.1-r0+24843-acf40c022e-mediatek-filogic-z8102ax-128m-squashfs-sysupgrade

It went very badly. I no longer had any access.

I put the original version back and since then I have tried to install OpenMPTCP whether via the web or via TFTP, it does not work.

I have a ZBT-Z8102AX V01

[image]

[image]

```
ubi0: attaching mtd6
ubi0: scanning is finished
ubi0: empty MTD device detected
ubi0: attached mtd6 (name "ubi", size 64 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 512, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 0, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 0/0, WL threshold: 4096, image sequence number: 0
ubi0: available PEBs: 489, total reserved PEBs: 23, PEBs reserved for bad PEB handling: 19
Reading from volume 'kernel' to 0x46000000, size 0x0 ... *** Cannot find volume 'kernel' ***
MT7981>
```

Ysurac, 2024-04-12:
Some versions of z8102ax have a dirty u-Boot image that doesn't allocate 128M, you may need to update it.
Should be > BL31

wouam31200, 2024-04-12:
Thank you for the feedback.
Can you guide me to put on the right boot?

Ysurac, 2024-04-12:
You checked if you have BL31 ?
You will need to contact ZBT to get BL32. Else I can compile another image for BL31...

wouam31200, 2024-04-12:
```
NOTICE:  EMI: Detected DRAM size: 1024MB
NOTICE:  EMI: complex R/W mem test passed
NOTICE:  CPU: MT7981 (1300MHz)
NOTICE:  SPI_NAND parses attributes from parameter page.
NOTICE:  SPI_NAND Detected ID 0xef
NOTICE:  Page size 2048, Block size 131072, size 134217728
NOTICE:  Initializing NMBM ...
NOTICE:  Signature found at block 1022 [0x07fc0000]
NOTICE:  First info table with writecount 0 found in block 960
NOTICE:  Second info table with writecount 0 found in block 963
NOTICE:  NMBM has been successfully attached in read-only mode
NOTICE:  BL2: Booting BL31
NOTICE:  BL31: v2.6(release):1e58996-dirty
NOTICE:  BL31: Built : 03:33:06, Feb 14 2023
NOTICE:  Hello BL31!!!
```

If you can make an image for BL31 that would be great!

Ysurac, 2024-04-12:
ok, this will be in next image snapshot as z8102ax-64m (the BL31 reduce flash size I can use).

Ysurac, 2024-04-13:
The image is available here: https://snapshots.openmptcprouter.com/6.6/z8102ax_64m/targets/mediatek/filogic/

hairy-dog, 2024-04-13:
> Note that LAN ports are used as WAN ports and WAN port is used as LAN port and answer on 192.168.100.1 This to be able to 
> connect the maximum WAN connections, but not sure if this will be keep as default on this device...

Does it also use the 4G/5G connections as WAN connections?

Ysurac, 2024-04-14:
yes, internal modems are also used as WAN connections, identified as modem1 and modem2

wouam31200, 2024-04-14:
Hi! That's great ! Thank you so much ! I'm going to try this version today and get back to you soon!

wouam31200, 2024-04-14:
Hi, I just installed it. All went well .
Great job!

On the other hand, I have a high average load: 5.26, 3.97, 2.14

It's normal?
Thank you so much

Ysurac, 2024-04-14:
No, it's not normal, but possible if WAN have no SIMs (it try to restart them).

wouam31200, 2024-04-15:
Hi!

That's right, I removed the interfaces that are not needed and the CPU consumption has dropped enormously!

For this model, do you think it makes sense to keep the LANs in WAN and vice versa?
How can we put two lte modules?!

Thank you for your help and for firmware that matches my model.

davidohne, 2024-04-15:
Is OpenMPTCRouter working with the Z8102AX-M2-S V2 version? According to the ROOTer forums they are changing things to the device and do not publish the changes. Don't want to buy a device and it's not working with openmptcprouter :)

Ysurac, 2024-04-15:
@wouam31200 You do what you what with LANs/WAN port, the default config with LANs ports used as WAN is when you want to use multiple ethernet connections. Modems are included in the box and there is many different modems used.

@davidohne I think all Z8102AX share same platform, others part are about metal/plastic box and modems (4G/5G/eSIM support or not,...).
But I can't guarantee anything, only the manufacturer, ZBT, can really know.

kyawgyikwa, 2024-04-15:
@wouam31200 Which z8102ax version did you use bro? Plastic case with foldable antenna version or metal case with detectable antenna version?

wouam31200, 2024-04-16:
Hi, I have both! There is no difference, you just have to look on the motherboard which version you have.

davidohne, 2024-04-16:
> Hi, I have both! There is no difference, you just have to look on the motherboard which version you have.

Perfect. Than I will order one. I was a little bit afraid, that they made some changes, like people were saying in the ROOter forums: https://whrl.pl/RgHCNO

wouam31200, 2024-04-16:
Yes apparently there are several versions.
As indicated above, I have V1s and so far no problems.

Before ordering I advise you to ask ZBT

davidohne, 2024-04-16:
> @wouam31200 Which z8102ax version did you use bro? Plastic case with foldable antenna version or metal case with detectable antenna version?

I am unsure if it affects the reception, but the antennas of the foldable version share both Wi-Fi and mobile signals. This is an image provided by the manufacturer:

[image]

Ysurac, 2024-04-16:
For now, all Z8102AX version should work. The V2 add eSIM support only I think. OpenMPTCProuter is already tested on at least 4 different versions.
For the antennas, modems antennas should be connected to the modems modules directly.

davidohne, 2024-04-18:
Perfect. I just ordered one, but it will take some time from china. For me, it looks like a perfect device for digital nomads... I'm also thinking about adding another USB-M.2 Adapter with a 3rd 5G modem in the inside of the router case and add four more antennas to the sides of the case to support three mobile networks in sum. We'll see if that works with good with the usb 3 port.

davidohne, 2024-05-06:
Is there any way to find out the BL version without connecting a serial cable?

davidohne, 2024-05-06:
Serial console output with 64mb firmware:
```
[    2.351540] mtk_soc_eth 15100000.ethernet eth0: entered promiscuous mode
[    2.358302] DSA: tree 0 setup
[    2.362197] UBI: auto-attach mtd4
[    2.365542] ubi0: default fastmap pool size: 25
[    2.370062] ubi0: default fastmap WL pool size: 12
[    2.374866] ubi0: attaching mtd4
[    2.805337] ubi0: scanning is finished
[    2.818480] ubi0 error: ubi_read_volume_table: not enough PEBs, required 895, available 512
[    2.826976] ubi0 error: ubi_attach_mtd_dev: failed to attach mtd4, error -28
[    2.834054] UBI error: cannot attach mtd4
[    2.838275] clk: Disabling unused clocks
[    2.842981] /dev/root: Can't open blockdev
[    2.847110] VFS: Cannot open root device "" or unknown-block(0,0): error -6
[    2.854058] Please append a correct "root=" boot option; here are the available partitions:
[    2.862410] 1f00            1024 mtdblock0
[    2.862415]  (driver?)
[    2.868941] 1f01             512 mtdblock1
[    2.868945]  (driver?)
[    2.875469] 1f02            2048 mtdblock2
[    2.875473]  (driver?)
[    2.881990] 1f03            2048 mtdblock3
[    2.881994]  (driver?)
[    2.888516] 1f04           65536 mtdblock4
[    2.888520]  (driver?)
[    2.895039] List of all bdev filesystems:
[    2.899035]  ext3
[    2.899038]  ext2
[    2.900951]  ext4
[    2.902864]  squashfs
[    2.904783]  f2fs
[    2.907044]
[    2.910437] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    2.918683] SMP: stopping secondary CPUs
[    2.922595] Kernel Offset: disabled
[    2.926070] CPU features: 0x0,00000000,00000000,1000400b
[    2.931369] Memory Limit: none
[    2.939672] pstore: backend (ramoops) writing error (-28)
[    2.945062] Rebooting in 1 seconds..
```

Serial output with 128mb firmware:
```
[    3.244121] ubi0: available PEBs: 0, total reserved PEBs: 913, PEBs reserved for bad PEB handling: 14
[    3.253331] ubi0: background thread "ubi_bgt0d" started, PID 665
[    3.261385] block ubiblock0_1: created from ubi0:1(rootfs)
[    3.266905] ubiblock: device ubiblock0_1 (rootfs) set to be root filesystem
[    3.274061] clk: Disabling unused clocks
[    3.286293] VFS: Mounted root (squashfs filesystem) readonly on device 254:0.
[    3.296991] devtmpfs: mounted
[    3.300200] Freeing unused kernel memory: 512K
[    3.304783] Run /sbin/init as init process
[    3.308871]   with arguments:
[    3.311826]     /sbin/init
[    3.314529]   with environment:
[    3.317657]     HOME=/
[    3.320005]     TERM=linux
[    3.492311] SQUASHFS error: xz decompression failed, data probably corrupt
[    3.499214] SQUASHFS error: Failed to read block 0x209666d: -5
[    3.505050] SQUASHFS error: Unable to read fragment cache entry [209666d]
[    3.511874] SQUASHFS error: Unable to read fragment cache entry [209666d]
[    3.518662] SQUASHFS error: Unable to read page, block 209666d, size 6120
Error loading shared library libjson-c.so.5: I/O error (needed by /sbin/init)
[    3.578360] SQUASHFS error: xz decompression failed, data probably corrupt
[    3.585269] SQUASHFS error: Failed to read block 0x209666d: -5
[    3.591089] SQUASHFS error: Unable to read fragment cache entry [209666d]
[    3.597869] SQUASHFS error: Unable to read page, block 209666d, size 6120
Error loading sh[    3.605383] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00
[    3.613710] SMP: stopping secondary CPUs
[    3.617625] Kernel Offset: disabled
[    3.621100] CPU features: 0x0,00000000,00000000,1000400b
[    3.626400] Memory Limit: none
[    3.634897] pstore: backend (ramoops) writing error (-28)
[    3.640287] Rebooting in 1 seconds..
```

Ysurac, 2024-05-06:
1st part is what you have when you are on BL31.
For the other part, try to download the image again and update it again: image is corrupted, so I think it's not correctly write on the flash.

davidohne, 2024-05-06:
Tried again (this time in Safari, before in Firefox). Flashed stock openwrt, then used the following file. Unfortunately again, bootloop :(

https://snapshots.openmptcprouter.com/6.6/z8102ax_64m/targets/mediatek/filogic/openmptcprouter-v0.61-snapshot-6.6-r0+26018-84a48ce400-mediatek-filogic-z8102ax-64m-squashfs-sysupgrade.bin

[image]

Ysurac, 2024-05-06:
You tried stable release or with 6.1 based release ?
And for the second part it was z8102ax-128m, here it's the 64m version that, from your log, will not work.

davidohne, 2024-05-06:
1. Since I'm on BL31 2.7 I thought I need the 64m version, is that right?
2. I tried snapshots -> 6.6 -> 64m version in the last try. Is there another version I should try?

davidohne, 2024-05-06:
Hm. At ROOter they must know what the difference makes. IN February someone said, that his BL31 2.7 is not working with ROOter (it's the same version i own comparing his log files). Now the ROOter firmware is work, even with an explicit remark to OpenMTCPRouter - https://whrl.pl/RgHA2f:

```
The openmptcprouter firmware runs on 8102ax V2
And I thank you for that.

I used some of his router definition and combined it with a number of other things needed to make a proper firmware, like fixing an incorrect GPIO assignment and adding a working WAN port.

I finally have a working V2 firmware. And I went one step farther and packaged up both the V1 and V2 firmware in a single package with instructions on how tell which one you have.

I'm currently making up a PDF that shows how to change the V1 bootloader to the V2 and back again using a Serial cable. The V2 has the Recovery GUI which is handy. With the V1 you can use a number of other firmware from various sources. Other than the bootloader the V1 and V2 are identical.

One more problem solved. ```

There is also a manual how to downgrade the V2 bootloader to V1, do you think that would help?
https://www.ofmodemsandmen.com/upload/SerialRecovery.zip

Ysurac, 2024-05-06:
Yes, I solved the issue between previous BL31 and after u-boot release. I don't use same DTS as rooter, I use a custom one with nice name instead of using GPIO numbers. The SIM part is another story and don't change anything for boot. Only SIM is for now supported with OpenMPTCProuter (eSIM management is not so easy due to change in each modems models...).
It's already tested on about 5 different z8102ax models and I think a new one soon.
Don't downgrade.

davidohne, 2024-05-06:
Well.. It works now and I don't know what made it work... I read your message to late and used the fip and bl2 file out of the package from ROOter to flash the V2 bootloader on my V2 bootloader.. In my option: Nothing changed and also the Version number and date stayed the same.. During flashing I had errors with the tftp transmit all the time so I changed the computer and it worked out. After that I flashed the 128mb version (still BL31 release 2.7).

And it's working 🎉  Unfortunatley now I don't know if it was a problem with the ethernet connection between the first computer and the second one or if it was because of the "changed" fip and bl2...

Anyway: OpenMPTCProuter boots and is working so far. In the next days I will add the two 5G modems and keep testing..

Ysurac, 2024-05-06:
I'm actively working on z8102ax modems, so if you have problems tell me with modems models, I have some script not yet used in OpenMPTCProuter.

davidohne, 2024-05-07:
That sounds perfect. I've got two T99W175, which I soldered and brought to USB mode. Well one so far. The other one is still a little diva and I don't know what's the problem. Perhaps I made a mistake while soldering. Hopefully not.

The first T99W175 is detected as `Bus 001 Device 005: ID 1e2d:00b3 Cinterion Cinterion PID 0x00B3 USB Mobile Broadband`

I've installed all related mbim, qmi, modemmanager, mhi packages and kmods but modemmanager doesn't recognize it at the moment. Sometimes it's also withdrawn according to dmesg:

In the beginning:
```
[  164.358285] cdc_mbim 1-1.2:1.0 wwan0: register 'cdc_mbim' at usb-11200000.usb-1.2, CDC MBIM, 52:04:0f:bd:5b:b1
[  164.369361] option 1-1.2:1.2: no of_node; not parsing pinctrl DT
[  164.369480] option 1-1.2:1.2: GSM modem (1-port) converter detected
[  164.376034] option1 ttyUSB0: no of_node; not parsing pinctrl DT
[  164.376814] usb 1-1.2: GSM modem (1-port) converter now attached to ttyUSB0
[  164.384922] option 1-1.2:1.3: no of_node; not parsing pinctrl DT
[  164.385358] option 1-1.2:1.4: no of_node; not parsing pinctrl DT
[  164.385471] option 1-1.2:1.4: GSM modem (1-port) converter detected
[  164.392006] option1 ttyUSB1: no of_node; not parsing pinctrl DT
[  164.393346] usb 1-1.2: GSM modem (1-port) converter now attached to ttyUSB1
[  164.400843] option 1-1.2:1.5: no of_node; not parsing pinctrl DT
[  164.400964] option 1-1.2:1.5: GSM modem (1-port) converter detected
[  164.407405] option1 ttyUSB2: no of_node; not parsing pinctrl DT
[  164.420271] usb 1-1.2: GSM modem (1-port) converter now attached to ttyUSB2
```
And later on:
```
[  150.008062] mt7981-pinctrl 11d00000.pinctrl: request pin 2 (SYS_WATCHDOG) for pinctrl_moore:514
[  150.008089] gpio gpiochip0: Persistence not supported for GPIO 2
[  150.008104] gpio gpiochip0: registered chardev handle for line 2
[  150.008118] gpio gpiochip0: registered chardev handle for 1 lines
[  150.252562] usb 1-1.2: USB disconnect, device number 4
[  150.258043] cdc_mbim 1-1.2:1.0 wwan0: unregister 'cdc_mbim' usb-11200000.usb-1.2, CDC MBIM
[  150.317799] option1 ttyUSB0: GSM modem (1-port) converter now disconnected from ttyUSB0
[  150.326038] option 1-1.2:1.2: device disconnected
[  150.333644] option1 ttyUSB2: usb_wwan_open: submit read urb 0 failed: -19
[  150.340446] option1 ttyUSB2: usb_wwan_open: submit read urb 1 failed: -19
[  150.347268] option1 ttyUSB2: usb_wwan_open: submit read urb 2 failed: -19
[  150.354064] option1 ttyUSB2: usb_wwan_open: submit read urb 3 failed: -19
[  150.361422] option1 ttyUSB1: GSM modem (1-port) converter now disconnected from ttyUSB1
[  150.369599] option 1-1.2:1.4: device disconnected
[  150.375692] option1 ttyUSB2: GSM modem (1-port) converter now disconnected from ttyUSB2
[  150.383878] option 1-1.2:1.5: device disconnected
[  151.017781] mt7981-pinctrl 11d00000.pinctrl: request pin 2 (SYS_WATCHDOG) for pinctrl_moore:514
```

There is a massiv spam of these messages, which I think are not related to the modems:
```
[  152.031151] mt7981-pinctrl 11d00000.pinctrl: request pin 2 (SYS_WATCHDOG) for pinctrl_moore:514
[  152.031179] gpio gpiochip0: Persistence not supported for GPIO 2
[  152.031195] gpio gpiochip0: registered chardev handle for line 2
[  152.031209] gpio gpiochip0: registered chardev handle for 1 lines
```

The modem is working on another openwrt snapshot device, so I need to figure out why it's working on openwrt and not on openmptcprouter.

davidohne, 2024-05-07:
Nope. Unfortunatley the connection to the modem is no stable:

```
[  121.599040] cdc_mbim 1-1.2:1.0 wwan0: unregister 'cdc_mbim' usb-11200000.usb-1.2, CDC MBIM
[  121.649489] option1 ttyUSB0: GSM modem (1-port) converter now disconnected from ttyUSB0
[  121.657697] option 1-1.2:1.2: device disconnected
[  121.665587] option1 ttyUSB1: GSM modem (1-port) converter now disconnected from ttyUSB1
[  121.673781] option 1-1.2:1.4: device disconnected
[  121.678382] mt7981-pinctrl 11d00000.pinctrl: request pin 2 (SYS_WATCHDOG) for pinctrl_moore:514
[  121.678499] gpio gpiochip0: Persistence not supported for GPIO 2
[  121.678515] gpio gpiochip0: registered chardev handle for line 2
[  121.678530] gpio gpiochip0: registered chardev handle for 1 lines
[  121.684715] option1 ttyUSB2: GSM modem (1-port) converter now disconnected from ttyUSB2
[  121.692899] option 1-1.2:1.5: device disconnected
[  122.703978] mt7981-pinctrl 11d00000.pinctrl: request pin 2 (SYS_WATCHDOG) for pinctrl_moore:514
```

Ysurac, 2024-05-07:
Check what you have in Status->System log
For the gpio... part of the log, it's related to watchdog, I would need to check how I can remove that.

davidohne, 2024-05-07:
I will check later. Perhaps I also have to upgrade the modem firmware, to improve the stability, but with an APU board it worked stable so far. Thought it would be easier to "highjack" these modems...

closed 2024-08-20 by github-actions[bot]