Ysurac/openmptcprouter #3260: Configuration doesnt save

zRxnx, 2024-04-10:
## Expected Behavior
Save configuration right

## Current Behavior
If i try to save the admin passwort, it says "The syspassword cant be saved"
I cant delete wan1, wan2 in interfaces, it says "pending deletion"

Overall, it looks like i cant save my configuration due to it doesnt save

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version: NOT CONNECTED YET
  - OpenMPTCProuter VPS provider:NOT CONNECTED YET
  - OpenMPTCProuter platform: RPI4B 64Bit, ext4

zRxnx, 2024-04-10:
Alry tried to reconnect, how do i check if its readonly?

zRxnx, 2024-04-10:
It says "Remounting filesystem read-only"

Ysurac, 2024-04-10:
Then it's the issue, as it's a fresh install, flash again the SD card.
Make sure you have a good SD card and you can also try squashfs sometimes better.

zRxnx, 2024-04-10:
Cant i just readd some permissions?

Ysurac, 2024-04-10:
easier to reinstall as it's a fresh installation...
Else do something like ```fsck.ext4 -y /dev/mmcblk0p2``` via SSH on the router.

zRxnx, 2024-04-10:
Thanks it works now, just to make sure.

DHCP Server has to be disabled on all isps right?
And they need all diffrent ips?

Ysurac, 2024-04-10:
If you are not using VLAN or a device with multiples ethernet ports, yes DHCP server must be disabled on all modems and modems set on different subnet.
The status page display the "IP was not leased..." because you use a static IP on your client computer (or IP was not yet renewed on OpenMPTCProuter DHCP server)

zRxnx, 2024-04-12:
Ok thank you!
The switch to the other image worked.

I use a managed switch with 8 ports.
I used this [tutorial](https://kamrul.dev/aggregate-multiple-internet-with-openmptcprouter/) to setup everything. I got it perfectly working.

But i have one big problem, and i think thats on windows side.
When i connect/disconnect the ethernet port multiple times, my pc doesnt detect the internet anymore. It says "No internet" but i can still access the openmptcprouter panel.
I always delete all internet profiles, flash the tcp ports and restart my internet adapter, but that doesnt help 100%.

One some question, while the router runs with one isp connected, can i connect the 2nd isp while the router is running, or should i restart him to work?

Can you help me there?

Ysurac, 2024-04-12:
Which ethernet port do you connect/disconnect ?
You can connect another ISP while the router is running.

What you mean with internet loss ? high latency or packets missing ?

If you use VLAN, you you only need to have differents IPs for each OpenMPTCProuter interfaces.

zRxnx, 2024-04-12:
> Which ethernet port do you connect/disconnect ?
The one from my pc, that connects to the openmptcprouter, sometimes if i reconnect that, windows says i have "No internet"

> What you mean with internet loss ? high latency or packets missing ?
Internet speed, normal i have ~600mbits but then i have only ~300mbits

Ysurac, 2024-04-13:
For the ethernet port so yes it's related to Windows.

For speed, how do you test it ? You can use ```omr-test-speed```
You can also check VPS speed between each link speed to this VPS, for example ```omr-test-speed-server wan1``` will test speed between wan1 and VPS.
A RPI4B can reach up to 550Mb/s.

zRxnx, 2024-04-13:
I tested the speed on an end device. (PC)

Wait, a rpi4b can only do 550mb/s instead of 1 gbit?

550mb/s is mbits right? So divide it by 8 and you get bytes right?

Ysurac, 2024-04-13:
Yes, a RPI5 can go up to 900Mb/s, limited by ethernet port speed.

zRxnx, 2024-04-13:
> Yes, a RPI5 can go up to 900Mb/s, limited by ethernet port speed.

Can you recommend anything that has >1gbit?

Ysurac, 2024-04-13:
x86_64 devices with AES supported by CPU.

zRxnx, 2024-04-14:
Thank you!
Everything solved now.

closed 2024-04-14 by zRxnx