Ysurac/openmptcprouter #3261: Support for Mikrotik RBM33G

dnwk, 2024-04-10:
If possible, I would love to see support for Mikrotik RBM33G
https://openwrt.org/toh/hwdata/mikrotik/mikrotik_rbm33g
https://openwrt.org/toh/mikrotik/rbm33g

**Describe why you are asking for this support?**
Mikrotik RBM33G has dual slot for Cellular modem and another slot for wifi card. It is ideally setup for the use case of OpenMPTCPRouter and convenient to aggregate 2 cellular connection

Ysurac, 2024-04-11:
16 MB Flash, 256 MB RAM and a 800Mhz CPU. It's quite the same CPU as Edgerouter X, it's really too slow and there is not enough flash for a complete image.

dnwk, 2024-04-11:
This device has microSD card slot which could potentially allow for full image. But yeah, CPU is the problem. Thanks for commenting.

aladdinshammr, 2024-04-16:
@dnwk 

Is it possible for openwrt to be installed in any router board or just in these types in the link??

dnwk, 2024-06-18:
> @dnwk
> 
> Is it possible for openwrt to be installed in any router board or just in these types in the link??

Since this model is specifically mentioned, I think it is possible

closed 2024-09-22 by github-actions[bot]