Ysurac/openmptcprouter #3262: Cannot be used when there are two identical IPV4 addresses

lkyddyy, 2024-04-11:
When there are two identical IPV4 addresses, it cannot be used and conflicts may occur. This issue does not exist in version 0.59.1

  - OpenMPTCProuter version: <openmptcprouter-v0.60-6.1>
  - OpenMPTCProuter VPS version: <0.1029>
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: <RPI4>

Ysurac, 2024-04-11:
Please use the template.
Do you have some logs ? More information ?

lkyddyy, 2024-04-11:
[image]
The first image is from version 0.59.1 and can be used normally
[image]
The second image is from version 0.60 and there is a conflict

The OpenMPTCP router platform is RPI4, and X86 also has the same issue

lkyddyy, 2024-04-11:
Apr 11 10:38:55 OpenMPTCProuter daemon.info [5944]: <inf> [modem0/sim0] loaded GID1: FFFFFFFF
Apr 11 10:38:55 OpenMPTCProuter daemon.info [5944]: <inf> [modem0/sim0] loaded GID2: FFFFFFFF
Apr 11 10:38:55 OpenMPTCProuter daemon.info procd: - init complete -
Apr 11 10:38:58 OpenMPTCProuter user.notice MPTCP: Set eth2 to on from eth2 is deactivated
Apr 11 10:38:59 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (usb1) switched up
Apr 11 10:38:59 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Routes not correctly set for wan2 (usb1 table 7) with IPs 192.168.42.2 
Apr 11 10:38:59 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of wan2 (usb1)
Apr 11 10:39:00 OpenMPTCProuter user.notice MPTCP: Set usb0 to master from usb0 is deactivated
Apr 11 10:39:00 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Apr 11 10:39:01 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath usb0 (wan1) switched to on (from off)
Apr 11 10:39:01 OpenMPTCProuter user.notice MPTCP: Set usb1 to on from usb1 is deactivated
Apr 11 10:39:02 OpenMPTCProuter user.notice post-tracking-020-status: New public ip detected for wan1 (usb0): 39.144.67.39 (previous: 39.144.134.49)
Apr 11 10:39:03 OpenMPTCProuter daemon.notice [5944]: <msg> [device /sys/devices/platform/soc/fe980000.usb/usb1/1-1/1-1.4] creating modem with plugin 'quectel' and '6' ports
Apr 11 10:39:03 OpenMPTCProuter daemon.warn [5944]: <wrn> [plugin/quectel] could not grab port ttyUSB9: Cannot add port 'tty/ttyUSB9', unhandled port type
Apr 11 10:39:03 OpenMPTCProuter daemon.warn [5944]: <wrn> [plugin/quectel] could not grab port ttyUSB6: Cannot add port 'tty/ttyUSB6', unhandled port type
Apr 11 10:39:03 OpenMPTCProuter daemon.warn [5944]: <wrn> [plugin/quectel] could not grab port ttyUSB5: Cannot add port 'tty/ttyUSB5', unhandled port type
Apr 11 10:39:03 OpenMPTCProuter daemon.info [5944]: <inf> [modem1] tty/ttyUSB7: at (primary)
Apr 11 10:39:03 OpenMPTCProuter daemon.info [5944]: <inf> [modem1] tty/ttyUSB8: at (secondary)
Apr 11 10:39:03 OpenMPTCProuter daemon.info [5944]: <inf> [modem1] tty/ttyUSB7: at (data primary)
Apr 11 10:39:03 OpenMPTCProuter daemon.info [5944]: <inf> [modem1] net/usb1: net (data)
Apr 11 10:39:03 OpenMPTCProuter daemon.notice [5944]: <msg> [base-manager] modem for device '/sys/devices/platform/soc/fe980000.usb/usb1/1-1/1-1.4' successfully created
Apr 11 10:39:03 OpenMPTCProuter user.notice MPTCP: Set usb0 to master from usb0 is deactivated
Apr 11 10:39:03 OpenMPTCProuter daemon.info [5944]: <inf> [modem1] SIM hot swap setup succeeded
Apr 11 10:39:03 OpenMPTCProuter daemon.info [5944]: <inf> [modem1] SIM is ready, and no need for the after SIM unlock step...
Apr 11 10:39:03 OpenMPTCProuter user.notice MPTCP: Set usb1 to on from usb1 is deactivated
Apr 11 10:39:03 OpenMPTCProuter daemon.info [5944]: <inf> [modem1/sim1] loaded SIM identifier: ###
Apr 11 10:39:03 OpenMPTCProuter daemon.info [5944]: <inf> [modem1/sim1] loaded IMSI: ###
Apr 11 10:39:03 OpenMPTCProuter daemon.info [5944]: <inf> [modem1/sim1] loaded operator identifier: 46007
Apr 11 10:39:03 OpenMPTCProuter daemon.info [5944]: <inf> [modem1/sim1] loaded operator name: CMCC
Apr 11 10:39:04 OpenMPTCProuter daemon.info [5944]: <inf> [modem1/sim1] loaded list of preferred networks: 
Apr 11 10:39:04 OpenMPTCProuter daemon.info [5944]: <inf> [modem1/sim1] loaded GID1: FFFFFFFF
Apr 11 10:39:04 OpenMPTCProuter daemon.info [5944]: <inf> [modem1/sim1] loaded GID2: FFFFFFFF
Apr 11 10:39:04 OpenMPTCProuter user.notice post-tracking-020-status: New public ip detected for wan2 (usb1): 111.55.10.166 (previous: 39.144.67.12)
Apr 11 10:39:12 OpenMPTCProuter daemon.notice [5944]: <msg> [modem0] state changed (unknown -> disabled)

Ysurac, 2024-04-11:
Via SSH on the router, what is the result of ```ip mptcp endpoint show``` and of ```multipath``` ?

lkyddyy, 2024-04-11:
[image]

Ysurac, 2024-04-11:
I would need more log, I think the log is only after boot and I need to find why usb1 is deactivated.

lkyddyy, 2024-04-11:
Apr 11 20:17:28 OpenMPTCProuter syslog.info syslogd started: BusyBox v1.36.1
Apr 11 12:17:28 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S13openssl: Generating engines.cnf
Apr 11 12:17:28 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S13openssl: Generating providers.cnf
Apr 11 12:17:28 OpenMPTCProuter daemon.info ModemManager[2147]: hotplug: add network interface gretap0: event processed
Apr 11 12:17:29 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='erspan0'
Apr 11 12:17:29 OpenMPTCProuter daemon.info ModemManager[2281]: hotplug: add network interface erspan0: event processed
Apr 11 12:17:29 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='ip6gre0'
Apr 11 12:17:29 OpenMPTCProuter daemon.info ModemManager[2341]: hotplug: add network interface ip6gre0: event processed
Apr 11 12:17:29 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='teql0'
Apr 11 12:17:29 OpenMPTCProuter user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Apr 11 12:17:29 OpenMPTCProuter user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Apr 11 12:17:29 OpenMPTCProuter user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Apr 11 12:17:29 OpenMPTCProuter daemon.info ModemManager[2398]: hotplug: add network interface teql0: event processed
Apr 11 12:17:29 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='eth1'
Apr 11 12:17:29 OpenMPTCProuter daemon.info ModemManager[2519]: hotplug: add network interface eth1: event processed
Apr 11 12:17:29 OpenMPTCProuter daemon.err ModemManager[2519]: hotplug: Couldn't report kernel event: error: couldn't get bus: Could not connect: No such file or directory
Apr 11 12:17:29 OpenMPTCProuter daemon.err procd: Could not find firmware /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-compute-module.bin: No such file or directory
Apr 11 12:17:29 OpenMPTCProuter daemon.err procd: Could not find firmware /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-compute-module.clm_blob: No such file or directory
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 114.114.114.114#53
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Apr 11 20:17:29 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Apr 11 12:17:29 OpenMPTCProuter authpriv.info dropbear[2897]: Not backgrounding
Apr 11 12:17:29 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='wlan0'
Apr 11 12:17:29 OpenMPTCProuter daemon.info ModemManager[2944]: hotplug: add network interface wlan0: event processed
Apr 11 12:17:29 OpenMPTCProuter daemon.err ModemManager[2944]: hotplug: Couldn't report kernel event: error: couldn't get bus: Could not connect: No such file or directory
Apr 11 12:17:30 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Apr 11 12:17:30 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 11 12:17:30 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 11 12:17:30 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 11 12:17:30 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 11 12:17:30 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 11 12:17:30 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 11 12:17:30 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 11 12:17:30 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
[... cut ...]
Apr 11 12:46:58 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for usb1
Apr 11 12:46:59 OpenMPTCProuter user.notice MPTCP: Set usb1 to on from usb1 is deactivated
Apr 11 12:47:00 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath usb1 (wan2) switched to on (from off)
Apr 11 12:47:03 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for usb0
Apr 11 12:47:03 OpenMPTCProuter user.notice MPTCP: Set usb0 to master from usb0 is deactivated

lkyddyy, 2024-04-11:
As long as the IPV4 address is the same, not only USB1 is disabled, but USB0 will also stop. The two interfaces keep switching and stopping. If the interface is set to a static address and different IPV4 are set, it can work normally

lkyddyy, 2024-04-11:
Two IP addresses cannot be the same 192.168.42.2, but this situation works properly in version 0.59.1，The x86 version also has the same issue

lkyddyy, 2024-04-11:
Thank you very much for your efforts. Version 0.60 has updated exciting features

closed 2024-07-16 by github-actions[bot]