Ysurac/openmptcprouter #3264: cascade NAT  causes kernal crash

aladdinshammr, 2024-04-11:
## Expected Behavior

I expect OMR to work well whether I connected to it directly or I plug in  a router that uses OMR as internet source

## Current Behavior

OMR works fine when I connect to it directly, but when I connect a mikrotik router internet slows down then gradually stops.

I see in log something like this

Apr 8 23:28:06 OpenMPTCProuter daemon.err omr-tracker[25131]: sh: 66.6667: bad number
Apr 8 23:35:23 OpenMPTCProuter daemon.err omr-tracker[25130]: sh: 66.6667: bad number

I don't have much knowledge in networking nor in  linux, but I assume that is because of enabling net.ipv4.tcp_mtu_probing in OMR in a network that has multiple nat causes this problem as mentioned in this https://github.com/multipath-tcp/mptcp/issues/108

## Specifications

  - OpenMPTCProuter version: v0.60 rc2
  - OpenMPTCProuter VPS version: 5.4.207-mptcp and today I set up 0.1029-test 6.1.0-18-cloud-amd64, but I haven't tested yet
  - OpenMPTCProuter VPS provider: google cloud 
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-04-11:
Please update to v0.60 first
What is the kernel used ? 5.4 or 6.1 ?
```net.ipv4.tcp_mtu_probing``` is disabled in OpenMPTCProuter.

aladdinshammr, 2024-04-11:
@Ysurac 

I tested it in kernal 5.4, and today I will test kernal 6.1.
I use everything mentioned in this link https://github.com/Ysurac/openmptcprouter/issues/2961

but these notifications never go, and when I update the VPS using the advance setting tab in OMR router it changes the kernal to 5.4.

[image]
[image]

Ysurac, 2024-04-11:
As I said, 0.60 is now available, you need to download it and update.
Here you are using rc2.

aladdinshammr, 2024-04-11:
@Ysurac 

this https://download.openmptcprouter.com/release/v0.60/x86_64/targets/x86/64/?

Ysurac, 2024-04-11:
Use https://www.openmptcprouter.com/download

aladdinshammr, 2024-04-11:
That's weird.
chrome until now shows me v0.59.1-5.4.
it worked only when I changed the browser.
maybe there is caching or something.
[image]

aladdinshammr, 2024-04-13:
I upgraded now 
router version: 0.60-6.1
VPS version: 0.1030.6.1.0-18-cloud-amd64.

It seem stable until now , but I haven't tested it yet.

something different  in this version:
1- when I check in http://amiusingmptcp.de/ , it complains that I don't use MPTCP.
2- I always get this errors in the log:

Apr 13 15:16:00 OpenMPTCProuter daemon.err omr-tracker[17486]: sh: 66.6667: bad number
Apr 13 15:17:09 OpenMPTCProuter daemon.err dnsmasq[48]: nftset inet fw4 omr_dscp_cs2_4 netlink: Error: cache initialization failed: Invalid argument
Apr 13 15:17:09 OpenMPTCProuter daemon.err dnsmasq[50]: nftset inet fw4 omr_dscp_cs2_4 netlink: Error: cache initialization failed: Resource busy
Apr 13 15:19:24 OpenMPTCProuter daemon.err dnsmasq[105]: nftset inet fw4 omr_dscp_cs2_4 netlink: Error: cache initialization failed: Protocol error
Apr 13 15:19:24 OpenMPTCProuter daemon.err dnsmasq[105]: nftset inet fw4 omr_dscp_cs2_4 netlink: Error: cache initialization failed: Protocol error
Apr 13 15:27:16 OpenMPTCProuter daemon.err dnsmasq[580]: nftset inet fw4 omr_dscp_cs2_4 netlink: Error: cache initialization failed: Resource busy
Apr 13 15:27:16 OpenMPTCProuter daemon.err dnsmasq[578]: nftset inet fw4 omr_dscp_cs2_4 netlink: Error: cache initialization failed: Invalid argument
Apr 13 15:34:29 OpenMPTCProuter daemon.err omr-tracker[17486]: sh: 66.6667: bad number

Ysurac, 2024-04-13:
http://amiusingmptcp.de/ only test MPTCPv0, MPTCPv1 is used with kernel 6.1
For DSCP I will check.

aladdinshammr, 2024-04-14:
This time the kernel didn't crash, but as usual, I lose the connection to the server.

system log:

Apr 14 16:24:40 OpenMPTCProuter user.notice OMR-VPS: Setting server config error, try again
Apr 14 16:25:08 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on 34.18.74.6, no server API answer on 34.18.74.6)
Apr 14 16:25:28 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Apr 14 16:25:38 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later

[image]

aladdinshammr, 2024-04-14:
can I put a load balance of 4 connections as a master?
maybe master connection needs to be with a high speed compared to the others

aladdinshammr, 2024-04-16:
@Ysurac 

is there anything useful in this ?

Apr 16 03:06:01 Aladdin user.notice post-tracking-001-post-tracking: wan3 (wan3) switched off because check error, interface have no IPv6, interface have no IPv6 gateway and ping from 192.168.4.52 error (1.0.0.1,114.114.115.115,1.2.4.8)
Apr 16 03:06:01 Aladdin user.notice post-tracking-001-post-tracking: Delete default route to 157.175.187.120 dev wan3
Apr 16 03:06:03 Aladdin user.notice post-tracking-001-post-tracking: Change in routes, set ip route replace default scope global nexthop via 192.168.2.1 dev wan1 weight 1 nexthop via 192.168.3.1 dev wan2 weight 100 (omrvpn_intf: tun0)
Apr 16 03:06:06 Aladdin user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on 157.175.187.120, no server API answer on 157.175.187.120)
Apr 16 03:06:15 Aladdin user.notice post-tracking-001-post-tracking: wan3 (wan3) switched up
Apr 16 03:06:15 Aladdin user.notice post-tracking-001-post-tracking: Interface route not yet set, set route ip r add default via 192.168.4.1 dev wan3 metric 9
Apr 16 03:06:23 Aladdin user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on 157.175.187.120, no server API answer on 157.175.187.120)
Apr 16 03:06:27 Aladdin user.notice post-tracking-001-post-tracking: wan2 (wan2) switched off because check error, interface have no IPv6, interface have no IPv6 gateway and ping from 192.168.3.10 error (1.0.0.1,114.114.115.115,1.2.4.8)
Apr 16 03:06:27 Aladdin user.notice post-tracking-001-post-tracking: Delete default route to 157.175.187.120 dev wan2
Apr 16 03:06:32 Aladdin user.notice post-tracking-020-status: Check API configuration...
Apr 16 03:06:32 Aladdin user.notice post-tracking-020-status: Check API configuration... Done
Apr 16 03:06:35 Aladdint user.notice post-tracking-001-post-tracking: wan2 (wan2) switched up
Apr 16 03:06:35 Aladdin user.notice post-tracking-001-post-tracking: Change in routes, set ip route replace default scope global nexthop via 192.168.2.1 dev wan1 weight 1 nexthop via 192.168.4.1 dev wan3 weight 1 nexthop via 192.168.3.1 dev wan2 weight 100 (omrvpn_intf: 
Apr 16 03:06:35 Aladdin user.notice post-tracking-001-post-tracking: Interface route not yet set, set route ip r add default via 192.168.3.1 dev wan2 metric 8
Apr 16 03:06:48 Aladdin user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on 157.175.187.120, no server API answer on 157.175.187.120)
Apr 16 03:06:57 Aladdin user.notice post-tracking-001-post-tracking: wan2 (wan2) switched off because check error, interface have no IPv6, interface have no IPv6 gateway and ping from 192.168.3.10 error (1.1.1.1,4.2.2.1,8.8.8.8)
Apr 16 03:06:57 Aladdin user.notice post-tracking-001-post-tracking: Delete default route to 157.175.187.120 dev wan2
Apr 16 03:07:05 Aladdin user.notice post-tracking-001-post-tracking: wan2 (wan2) switched up
Apr 16 03:07:05 Aladdin user.notice post-tracking-001-post-tracking: Change in routes, set ip route replace default scope global nexthop via 192.168.2.1 dev wan1 weight 1 nexthop via 192.168.4.1 dev wan3 weight 1 nexthop via 192.168.3.1 dev wan2 weight 100 (omrvpn_intf: 
Apr 16 03:07:05 Aladdin user.notice post-tracking-001-post-tracking: Interface route not yet set, set route ip r add default via 192.168.3.1 dev wan2 metric 8

closed 2024-07-21 by github-actions[bot]