Ysurac/openmptcprouter #3265: '0.9+20240324' for 'omr-vps-admin' was not found :-( when try OMR VPS upgrade to 0.1030 for 0.60 OMR

vinceducat, 2024-04-11:
update to 0.1030 from 0.1028

debian.sh stop with  '0.9+20240324' for 'omr-vps-admin' was not found message

 '0.9+20240324' for 'omr-vps-admin' was not found
## Possible Solution

  1. wget
  2.sh debian .sh
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.60
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: MilkyWan VPS
  - OpenMPTCProuter platform: x86_64
  - Country: France

lkyddyy, 2024-04-12:
I also encountered the same error during the upgrade process

vinceducat, 2024-04-12:
after install test version on vps + reinstall normal version... and many reinstlla beetwin omr router 5.4 6.1 ans same it seems to work in 5.4 omr and vps version.

do i need (retry) to install 6.1 on both omr and vps ?

Ysurac, 2024-04-12:
I forgot to update repo, should be ok now. Sorry.

vinceducat, 2024-04-12:
@Ysurac i'm on an OMR VPS of milkywan do i needs to try to update to kernel 6.1 ? or stay in 5.4 in this case what was the best way to do ? OMR-router from 0.60 5.4 to 0.61 6.1 and after OMR-Vps from  5.4 to 6.1 . Or first VPS and after Router ?

vinceducat, 2024-04-12:
When i try to update to kernel 6.1, the script ends with no error saying to reboot but the kernel stay in 5.15 version...

Ysurac, 2024-04-12:
Stay ? It's updated from 5.4 to 5.15 only ?
What do you have in /etc/os-release ?

vinceducat, 2024-04-12:
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Ysurac, 2024-04-12:
Did you reboot after install ? What do you have in /boot/ ?

vinceducat, 2024-04-12:
yes i reboot

vincentm@vincentm:/boot$ ls
config-5.10.0-28-amd64  config-5.4.207-mptcp  initrd.img-5.10.0-28-amd64  initrd.img-5.4.207-mptcp    System.map-5.15.57-mptcp  vmlinuz-5.10.0-28-amd64  vmlinuz-5.4.207-mptcp
config-5.15.57-mptcp    grub

vinceducat, 2024-04-12:
came back in 5.4 at this time to be able to have a connexion ...

can retest 6.1 install if you want

Ysurac, 2024-04-12:
Yes. I just updated my Milkywan VPS and it's using 6.1.0 kernel.
you can also try a ```apt-get dist-upgrade``` and if still no 6.1.0, a ```apt-get install linux-image-6.1.0-18-amd64```

vinceducat, 2024-04-12:
Thanks seems to work ... with apt-get dist-upgrade install script and reboot ;-) 
seems ok now thanks omr vos and omr router in 6.1

vinceducat, 2024-04-12:
after some othe problem (vps down :-( ) but help of Hugues From Milkywan all is ok ( Tanks to you both

vinceducat, 2024-04-14:
@Ysurac i continu to have some probleme with OMR VPS, Vps fail when trying to put some ipv6 config..

closed 2024-07-18 by github-actions[bot]