Ysurac/openmptcprouter #3268: Can you tell me what the latest VPS script is?

lkyddyy, 2024-04-12:
[image]

Can you tell me what the latest VPS script is? I will use wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | sh, this script is installed with version 5.4.207 and cannot be used in v0.60-6.1 versions

Ysurac, 2024-04-12:
As indicated in the doc ( https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS#installupdate-on-debian-10-busterdebian-11-bullseyedebian-12-bookworm-x86_64 ): ```wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="6.1" sh```

lkyddyy, 2024-04-12:
Thank you very much for your prompt reply. I will give it a try now

vinceducat, 2024-04-12:
@Ysurac  i do this command but the OMR-VPS stay in version 5.15 of the kernel even if the install script ends well...

hairy-dog, 2024-04-13:
Did you reboot the VPS after updating?