Ysurac/openmptcprouter #3269: Bypass via VPN failed to reconnect server

jle1511, 2024-04-12:
I should expect to able to ssh in VPN server on original IP and OPENMTPCP still connect through after turn on vpn1.conf

## Current Behavior

as soon we have vpn1.conf on, everything disconnected including ssh, and openmptcp router, and can not get back in original IP, we also tried vpn1.conf remote IP but no success to get back in the vps, everything down after vpn1.conf turn on.

we followed instruction here:
https://github.com/Ysurac/openmptcprouter/wiki/ByPass-via-VPN-on-server

Yes.

## Specifications

  - OpenMPTCProuter version: 0.60-6.1
  - OpenMPTCProuter VPS version: 0.1030 6.1.0-18 amd64
  - OpenMPTCProuter VPS provider: buyvm
  - OpenMPTCProuter platform: x86_64

Apr 12 17:04:33 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on x.x.x.x., no server API answer on x.x.x.x)

Ysurac, 2024-04-12:
It's because default gateway is modified to use VPN.
The page was not added to index because I've not yet finished it.

jle1511, 2024-04-12:
no wonder, I found the page by Google to see how to set it up but Google indexed your wiki lol, 
I had to recover and remove the setting. thank you
looking forward to getting it done once you are finished

closed 2024-07-16 by github-actions[bot]

alpha197, 2025-07-18:
Hello, any updates on this issue? I tried Bypass via VPN with a Wireguard VPN (on VPS Server with interface named „vpn1). No success.

jle1511, 2025-07-18:
> Should work if the VPN is working on VPN without modifying default gateway.

Is there a proper instruction to set this up? On wiki docs, it said not ready yet.Thanks