Ysurac/openmptcprouter #3271: VPS install script breaks connectivity in Aruba Cloud

nohajc, 2024-04-13:
## Expected Behavior
I should be able to connect to the internet from the VPS both via IPv6 and IPv4
e.g.
`ping -6 google.com`
`ping -4 google.com`
should work

## Current Behavior
After running the VPS install script (kernel 6.1, OS Debian 12 or Ubuntu 22.04) and rebooting the server, IPv6 stops working.
`ping -6 google.com` hangs without any output, after Ctrl+C reports 100% packet loss. Same result if I ping google's IPv6 address directly.

## Possible Solution

## Steps to Reproduce the Problem

  1. Try connectivity on a fresh VPS (type O1I1 - the cheapest one) configured with both IPv4 and IPv6 public addresses.
  2. Run the debian or ubuntu VPS installation script
  3. Reboot
  4. Try IPv6 connectivity

## Context (Environment)
Tested with Debian 12 and Ubuntu 22.04.
There are two ethernet interfaces - one for IPv4 (enx5), the other for IPv6 (enx4). Both have static IP assigned.
IPv6 default route is `fe80::1 via enx4` (link-local router address it seems). The IPv6 prefix is /64.

At first I used the Debian script which upgraded Debian 11 to 12 (Debian 11 is the latest provided by Aruba at the moment).
To avoid running an OS "unsupported" by the VPS provider, I switched to Ubuntu 22.04 (supported by Aruba and OMR out-of-the-box) but it didn't help.

## Specifications

  - OpenMPTCProuter version: 0.60
  - OpenMPTCProuter VPS version: 0.1030
  - OpenMPTCProuter VPS provider: Aruba Cloud
  - OpenMPTCProuter platform: RPI5
  - Country: CZ

nohajc, 2024-04-13:
I did further tests and curiously, if I configure the server as IPv6 only, then run the VPS script (with some modifications because github doesn't have a IPv6 address), I don't lose internet connection (pinging an external hostname works). However, I cannot ssh into the VPS from outside (still using IPv6). Any connection attempt will timeout.

nohajc, 2024-04-13:
At this point, I can even configure the IPv4 interface manually and I have both gateways working. OMR is giving me `Can't contact Server Admin Script` though. I suppose I'd have to restart some services.

nohajc, 2024-04-13:
Hmm, after restarting the VPS (this time I let it configure both interfaces automatically), I still have fully working internet with both IP versions but my RPi router can't connect via IPv4. I assume this is because there was no IPv4 configuration at the time I ran the installation script.

Anyway, my use case is to bond multiple WANs, some of which do not have IPv6 (that's why I'm using IPv4 VPS address in OMR config) and then enable IPv6 for the clients through VPS. Couldn't get it to work yet. When the IPv6 pings were timing out for me, OMR would just log that IPv6 is not supported on the VPS and disable it again.

nohajc, 2024-04-13:
Out of curiosity, I tried to enable IPv6 in OMR now and with the working upstream IP, it was actually enabled.
However, now I'm getting `IPv6 tunnel DOWN` and my client hangs while connecting to WiFi (probably not getting any address from DHCPv6).

Well, installing in IPv6 only mode and enabling IPv4 later is probably not the answer. I'm going to start over and see what happens with a fresh VPS installation when I disable `GLORYTUN_UDP` again.

Ysurac, 2024-04-13:
For IPv6, the poblem is more related to shorewall6, as interface is not the same for IPv4 and IPv6 (really strange...), you need to set ```/etc/shorewall6/params.net``` to correct interface.

nohajc, 2024-04-13:
Are we talking about incoming or outgoing connections now? Because originally, when I ran the VPS script on a fresh installation, I couldn't access IPv6 internet from the VPS itself.

If there's an assumption that the VPS has IPv4 and IPv6 on the same interface, could I work around this by bridging those interfaces together or something like that?

nohajc, 2024-04-13:
> For IPv6, the poblem is more related to shorewall6, as interface is not the same for IPv4 and IPv6 (really strange...), you need to set `/etc/shorewall6/params.net` to correct interface.

Setting the correct interface in `/etc/shorewall6/params.net` helped. I can connect from VPS to the internet now (via IPv6). Strangely, I'm still getting connection refused on 65301 though (VPN tunnel DOWN). That's on the IPv4 address.

This is a fresh installation again.

nohajc, 2024-04-13:
The service is running but the output seems a bit odd.
> \# systemctl status openvpn@tun1
> ● openvpn@tun1.service - OpenVPN connection to tun1
>      Loaded: loaded (/lib/systemd/system/openvpn@.service; enabled; vendor preset: enabled)
>      Active: active (running) since Sat 2024-04-13 16:59:47 UTC; 14min ago
>        Docs: man:openvpn(8)
>              https://community.openvpn.net/openvpn/wiki/Openvpn24ManPage
>              https://community.openvpn.net/openvpn/wiki/HOWTO
>    Main PID: 729 (openvpn)
>      Status: "Initialization Sequence Completed"
>       Tasks: 1 (limit: 1007)
>      Memory: 1.8M
>         CPU: 50ms
>      CGroup: /system.slice/system-openvpn.slice/openvpn@tun1.service
>              └─729 /usr/sbin/openvpn --daemon ovpn-tun1 --status /run/openvpn/tun1.status 10 --cd /etc/ope>
> 
> Apr 13 16:59:47 debian-forpsi-cz ovpn-tun1[729]: net_iface_mtu_set: mtu 1420 for tun1
> Apr 13 16:59:47 debian-forpsi-cz ovpn-tun1[729]: net_iface_up: set tun1 up
> Apr 13 16:59:47 debian-forpsi-cz ovpn-tun1[729]: net_addr_v4_add: 10.255.250.1/24 dev tun1
> Apr 13 16:59:47 debian-forpsi-cz ovpn-tun1[729]: Could not determine IPv4/IPv6 protocol. Using AF_INET
> Apr 13 16:59:47 debian-forpsi-cz ovpn-tun1[729]: Socket Buffers: R=[212992->524288] S=[212992->524288]
> Apr 13 16:59:47 debian-forpsi-cz ovpn-tun1[729]: UDPv4 link local (bound): [AF_INET][undef]:65301
> Apr 13 16:59:47 debian-forpsi-cz ovpn-tun1[729]: UDPv4 link remote: [AF_UNSPEC]
> Apr 13 16:59:47 debian-forpsi-cz ovpn-tun1[729]: MULTI: multi_init called, r=256 v=256
> Apr 13 16:59:47 debian-forpsi-cz ovpn-tun1[729]: IFCONFIG POOL IPv4: base=10.255.250.2 size=253
> Apr 13 16:59:47 debian-forpsi-cz ovpn-tun1[729]: Initialization Sequence Completed

nohajc, 2024-04-13:
> > For IPv6, the poblem is more related to shorewall6, as interface is not the same for IPv4 and IPv6 (really strange...), you need to set `/etc/shorewall6/params.net` to correct interface.
> 
> Setting the correct interface in `/etc/shorewall6/params.net` helped. I can connect from VPS to the internet now (via IPv6 and IPv4). Strangely, I'm still getting connection refused on 65301 though (VPN tunnel DOWN). That's on the IPv4 address.
> 
> This is a fresh installation again.

I switched back from Ubuntu to Debian and now everything is green. With the shorewall6 modification you suggested, I can enable IPv6 but my Wi-Fi client does not get any IPv6 assigned.

I use the default ULA prefix (fd46:.../48). My VPS has a limitation of maximum 16 IPv6 addresses, so if I understand it correctly, it's safer to use this private range for OMR clients. I already tried to enable IPv6 DHCP for the LAN interface but I'm not sure what the desired configuration is.

nohajc, 2024-04-13:
Ok, nevermind. It's just Firefox prefers IPv4 for some reason. Everything's working fine now.

One question though, can the install script be modified to correctly detect the "two separate interfaces" server configuration?

Ysurac, 2024-04-14:
I will add that in server script.

nohajc, 2024-04-14:
Thank you. Don't know if you want to use this issue to track the change... other than that it can be closed.

Ysurac, 2024-04-16:
Support is now added in snapshot VPS script. Will be in next release.

closed 2024-04-16 by Ysurac