Ysurac/openmptcprouter #3274: OMR-bypass issues after upgrading to latest version

darthclide, 2024-04-14:
## Expected Behavior
After setting port or domain name in OMR-bypass, it should go over interface. And visa versa, if I disable an OMR-bypass rule, it should no longer send traffic on that interface alone.

## Current Behavior
When disabling OMR-bypass rules it does not change anything. Instead, I get this in the system log:
Apr 14 00:40:02 OpenMPTCProuter daemon.err omr-tracker[16495]: sh: 66.6667: bad number
Apr 14 00:40:05 OpenMPTCProuter daemon.err omr-tracker[16495]: sh: 66.6667: bad number
Apr 14 00:40:10 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Apr 14 00:40:10 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Apr 13 20:40:15 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:15 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:15 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:15 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:16 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:16 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:16 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:16 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:16 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:18 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:18 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:18 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:18 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:19 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Apr 13 20:40:19 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory

## Possible Solution
I wonder if this is related to using your sysupgrade image and you changing the code over to nftables? I would prefer not nuking my 50 rules and starting over, and frankly "save and apply" should correctly change the rules anyway.

## Steps to Reproduce the Problem

  1. Change anything in OMR-bypass
  2. Save and Apply
  3. Watch as incorrect interface is used depending on what I changed

## Context (Environment)
I can no longer access some websites that require a residential IP address.

## Specifications

  - OpenMPTCProuter version: Version 0.60-6.1
  - OpenMPTCProuter VPS version: Version 0.1030 6.1.0-18-amd64
  - OpenMPTCProuter VPS provider: Vultr
  - OpenMPTCProuter platform: RPI4
  - Country: USA

darthclide, 2024-04-14:
Extra info: I just tried deleting some rules, and there was no change. It is as if it is pulling OMR-bypass data from a different location than what is configured in Luci App.

jle1511, 2024-04-14:
im seeing the same issues, i also have over 50 rules, and bypass wont take effect anything, i did tried clean setup clear up everything and just apply on/off 1 rule only and still not effect.

darthclide, 2024-04-14:
Additional info: Bonding is no longer working. Not sure if this is related to this OMR-bypass issue, but all traffic is only going over Master interface.

darthclide, 2024-04-14:
Additional info: Bonding is resolved by changing master interface. However, I will leave the previous comment in the off chance it is related in some way to other issues you are seeing on this version.

darthclide, 2024-04-14:
Additional info: I am now wondering if I should delete this issue because now OMR-bypass seems to be working correctly, but these errors in the system log are still concerning.

darthclide, 2024-04-16:
Coming here to report that although OLD rules seem to be working. NEW rules don't seem to be working. I have added all ports/domains listed under CurrPorts application to my eth1 (DSL), and yet the ping inside of that application jumps around by 50-100ms. When using my DSL connection directly through a 2nd Ethernet port in my PC, the ping stays rock solid. Unless the application is able to make connections without a port required, this should be working.

darthclide, 2024-04-17:
That being said, I found some 0.0.0.0 on local address, but with no remote address or remote port? I am not an expert, but this should mean that there is nothing to "bypass" if there is no remote address or port?

darthclide, 2024-04-17:
On a whim I started putting those supposedly "local only" ports into OMR-bypass, and the game is running smoothly on pure interface connection with no VPN. Although every match the port changes, I just make the port range wider to cover that port number.

This still doesn't fix those weird errors showing up in system log, so I will keep this post up as "not resolved".

darthclide, 2024-04-17:
Well... Spoke too soon. Switched applications and now this one that was just working yesterday is now broken... CurrPorts shows only one connection being used by this application and it is port 443. This is set in OMR-bypass. Also getting a ray ID error in Chrome when accessing secure.runescape.com even though this is also added in the domain section of OMR-Bypass.

darthclide, 2024-04-17:
Apparently runescape lists their servers with nice domain names because in the "Remote Host Name" field in the CurrPorts, I see "world104.runescape.com". And yet even with this set in OMR-Bypass, it is still not connecting over my DSL. I guess I will just keep adding comments into the void while I lose my mind over OMR-bypass working for one game, but not another. Perhaps these errors are more serious than it seems:

nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
nftset inet fw4 omr_dscp_cs2_4 Error: No such file or directory

Ysurac, 2024-04-17:
Not sure if it's the best way to get help... There is many messages and only small part of a log in latest message, this doesn't really help...
As domain you only have to put "runescape.com", all subdomains will be also bypassed. 
I would need the result of ```uci show omr-bypass``` via SSH on the router.
Make sure that you use OpenMPTCProuter IP as DNS on client computer.
I would need more logs.

darthclide, 2024-04-17:
> Not sure if it's the best way to get help... There is many messages and only small part of a log in latest message, this doesn't really help... As domain you only have to put "runescape.com", all [...]

? I attached the logs in the first post and there has been no response about it. So I shifted my comments to talk about my troubleshooting process. I never touch DNS settings on the computer, I always leave that to any router I use (in this case openmptcprouter).

Here is the result of

omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='irc.chat.twitch.tv'
omr-bypass.@domains[0].interface='eth1'
omr-bypass.@domains[0].enabled='0'
omr-bypass.@domains[1]=domains
omr-bypass.@domains[1].name='www.bungie.net'
omr-bypass.@domains[1].interface='eth1'
omr-bypass.@domains[2]=domains
omr-bypass.@domains[2].name='discordapp.com'
omr-bypass.@domains[2].interface='eth1'
omr-bypass.@domains[2].enabled='0'
omr-bypass.@domains[3]=domains
omr-bypass.@domains[3].name='discord.com'
omr-bypass.@domains[3].interface='eth1'
omr-bypass.@domains[3].enabled='0'
omr-bypass.@domains[4]=domains
omr-bypass.@domains[4].name='discord.gg'
omr-bypass.@domains[4].interface='eth1'
omr-bypass.@domains[4].enabled='0'
omr-bypass.@domains[5]=domains
omr-bypass.@domains[5].name='delta.com'
omr-bypass.@domains[5].interface='eth1'
omr-bypass.@ips[0]=ips
omr-bypass.@ips[0].ip='167.114.26.81'
omr-bypass.@ips[0].interface='eth1'
omr-bypass.@ips[0].note='Psyber Minecraft'
omr-bypass.@ips[0].enabled='0'
omr-bypass.@src_port[0]=src_port
omr-bypass.@src_port[0].sport='9427'
omr-bypass.@src_port[0].proto='udp'
omr-bypass.@src_port[0].note='Teamspeak'
omr-bypass.@src_port[0].interface='eth1'
omr-bypass.@src_port[0].enabled='0'
omr-bypass.@src_port[1]=src_port
omr-bypass.@src_port[1].proto='udp'
omr-bypass.@src_port[1].interface='eth1'
omr-bypass.@src_port[1].sport='27000:27031'
omr-bypass.@src_port[1].note='Steam'
omr-bypass.@src_port[2]=src_port
omr-bypass.@src_port[2].sport='3478:3480'
omr-bypass.@src_port[2].proto='udp'
omr-bypass.@src_port[2].note='Steam Voice/War Thunder'
omr-bypass.@src_port[2].interface='eth1'
omr-bypass.@src_port[3]=src_port
omr-bypass.@src_port[3].sport='4379'
omr-bypass.@src_port[3].proto='udp'
omr-bypass.@src_port[3].note='Steam Voice'
omr-bypass.@src_port[3].interface='eth1'
omr-bypass.@src_port[4]=src_port
omr-bypass.@src_port[4].sport='4380'
omr-bypass.@src_port[4].proto='udp'
omr-bypass.@src_port[4].interface='eth1'
omr-bypass.@src_port[4].note='Steam Voice'
[... 243 lines cut ...]
omr-bypass.@src_port[27].proto='udp'
omr-bypass.@src_port[27].interface='eth1'
omr-bypass.@src_port[27].note='War Thunder'
omr-bypass.@src_port[28]=src_port
omr-bypass.@src_port[28].sport='80'
omr-bypass.@src_port[28].proto='tcp'
omr-bypass.@src_port[28].interface='eth1'
omr-bypass.@src_port[28].note='War Thunder'
omr-bypass.@src_port[28].enabled='0'
omr-bypass.@src_port[29]=src_port
omr-bypass.@src_port[29].sport='8111'
omr-bypass.@src_port[29].proto='tcp'
omr-bypass.@src_port[29].interface='eth1'
omr-bypass.@src_port[29].note='War Thunder Test'
omr-bypass.@src_port[30]=src_port
omr-bypass.@src_port[30].sport='50000:64000'
omr-bypass.@src_port[30].proto='udp'
omr-bypass.@src_port[30].interface='eth1'
omr-bypass.@src_port[30].note='War Thunder Test'
omr-bypass.@src_port[30].enabled='0'
omr-bypass.@src_port[31]=src_port
omr-bypass.@src_port[31].sport='1910'
omr-bypass.@src_port[31].proto='tcp'
omr-bypass.@src_port[31].interface='eth1'
omr-bypass.@src_port[31].note='War Thunder Test'

darthclide, 2024-04-17:
Before my changes for War Thunder yesterday, runescape.com was working just fine. Like you said, it should cover all subdomains. Since then I have been trying everything to get this to work.

Ysurac, 2024-04-17:
Many bypass if source port, not so used in most case.
What is the result of ```/etc/init.d/omr-bypass restart``` and ```/etc/init.d/firewall restart``` (this one should be verbose and should help to find some errors) ?

darthclide, 2024-04-17:
Result of /etc/init.d/omr-bypass restart
[image]

Result of /etc/init.d/firewall restart

Section @defaults[0] option 'disable_ipv6' is not supported by fw4
Section zone_lan (lan) fullcone in defaults not enabled, ignore zone fullcone settings
Section zone_wan (wan) fullcone in defaults not enabled, ignore zone fullcone settings
Section zone_vpn (vpn) fullcone in defaults not enabled, ignore zone fullcone settings
Section @rule[0] (Deny Twitch DSL) is disabled, ignoring section
Section @rule[8] (Support-UDP-Traceroute) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstip_4 (omr_dst_bypass_eth0_rule) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstip_4_accept (omr_dst_bypass_eth0_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_eth0_srcip_4 (omr_dst_bypass_eth0_srcip) is disabled, ignoring section
Section omr_dst_bypass_eth0_mac_4 (omr_dst_bypass_eth0_mac) is disabled, ignoring section
Section omr_dst_bypass_eth0_srcport_tcp_4 (omr_dst_bypass_eth0_srcport) is disabled, ignoring section
Section omr_dst_bypass_eth0_srcport_udp_4 (omr_dst_bypass_eth0_srcport) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstport_tcp_4 (omr_dst_bypass_eth0_dstport) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstport_udp_4 (omr_dst_bypass_eth0_dstport) is disabled, ignoring section
Section omr_dst_bypass_eth1_dstip_4_accept (omr_dst_bypass_eth1_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_eth1_srcip_4 (omr_dst_bypass_eth1_srcip) is disabled, ignoring section
Section omr_dst_bypass_eth1_mac_4 (omr_dst_bypass_eth1_mac) is disabled, ignoring section
Section omr_dst_bypass_eth1_dstport_tcp_4 (omr_dst_bypass_eth1_dstport) is disabled, ignoring section
Section omr_dst_bypass_eth1_dstport_udp_4 (omr_dst_bypass_eth1_dstport) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstip_4 (omr_dst_bypass_tun0_rule) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstip_4_accept (omr_dst_bypass_tun0_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_tun0_srcip_4 (omr_dst_bypass_tun0_srcip) is disabled, ignoring section
Section omr_dst_bypass_tun0_mac_4 (omr_dst_bypass_tun0_mac) is disabled, ignoring section
Section omr_dst_bypass_tun0_srcport_tcp_4 (omr_dst_bypass_tun0_srcport) is disabled, ignoring section
Section omr_dst_bypass_tun0_srcport_udp_4 (omr_dst_bypass_tun0_srcport) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstport_tcp_4 (omr_dst_bypass_tun0_dstport) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstport_udp_4 (omr_dst_bypass_tun0_dstport) is disabled, ignoring section
Section omr_dst_bypass_eth2_dstip_4 (omr_dst_bypass_eth2_rule) is disabled, ignoring section
Section omr_dst_bypass_eth2_dstip_4_accept (omr_dst_bypass_eth2_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_eth2_srcip_4 (omr_dst_bypass_eth2_srcip) is disabled, ignoring section
Section omr_dst_bypass_eth2_mac_4 (omr_dst_bypass_eth2_mac) is disabled, ignoring section
Section omr_dst_bypass_eth2_srcport_tcp_4 (omr_dst_bypass_eth2_srcport) is disabled, ignoring section
Section omr_dst_bypass_eth2_srcport_udp_4 (omr_dst_bypass_eth2_srcport) is disabled, ignoring section
Section omr_dst_bypass_eth2_dstport_tcp_4 (omr_dst_bypass_eth2_dstport) is disabled, ignoring section
Section omr_dst_bypass_eth2_dstport_udp_4 (omr_dst_bypass_eth2_dstport) is disabled, ignoring section
Section omr_dst_bypass_usb0_dstip_4 (omr_dst_bypass_usb0_rule) is disabled, ignoring section
Section omr_dst_bypass_usb0_dstip_4_accept (omr_dst_bypass_usb0_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_usb0_srcip_4 (omr_dst_bypass_usb0_srcip) is disabled, ignoring section
Section omr_dst_bypass_usb0_mac_4 (omr_dst_bypass_usb0_mac) is disabled, ignoring section
Section omr_dst_bypass_usb0_srcport_tcp_4 (omr_dst_bypass_usb0_srcport) is disabled, ignoring section
Section omr_dst_bypass_usb0_srcport_udp_4 (omr_dst_bypass_usb0_srcport) is disabled, ignoring section
Section omr_dst_bypass_usb0_dstport_tcp_4 (omr_dst_bypass_usb0_dstport) is disabled, ignoring section
Section omr_dst_bypass_usb0_dstport_udp_4 (omr_dst_bypass_usb0_dstport) is disabled, ignoring section
Section omr_dst_bypass_usb1_dstip_4 (omr_dst_bypass_usb1_rule) is disabled, ignoring section
Section omr_dst_bypass_usb1_dstip_4_accept (omr_dst_bypass_usb1_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_usb1_srcip_4 (omr_dst_bypass_usb1_srcip) is disabled, ignoring section
Section omr_dst_bypass_usb1_mac_4 (omr_dst_bypass_usb1_mac) is disabled, ignoring section
Section omr_dst_bypass_usb1_srcport_tcp_4 (omr_dst_bypass_usb1_srcport) is disabled, ignoring section
Section omr_dst_bypass_usb1_srcport_udp_4 (omr_dst_bypass_usb1_srcport) is disabled, ignoring section
Section omr_dst_bypass_usb1_dstport_tcp_4 (omr_dst_bypass_usb1_dstport) is disabled, ignoring section
Section omr_dst_bypass_usb1_dstport_udp_4 (omr_dst_bypass_usb1_dstport) is disabled, ignoring section
Section omr_dst_bypass_eth3_dstip_4 (omr_dst_bypass_eth3_rule) is disabled, ignoring section
Section omr_dst_bypass_eth3_dstip_4_accept (omr_dst_bypass_eth3_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_eth3_srcip_4 (omr_dst_bypass_eth3_srcip) is disabled, ignoring section
Section omr_dst_bypass_eth3_mac_4 (omr_dst_bypass_eth3_mac) is disabled, ignoring section
Section omr_dst_bypass_eth3_srcport_tcp_4 (omr_dst_bypass_eth3_srcport) is disabled, ignoring section
[... cut ...]
Section miniupnpd specifies unreachable path '/usr/share/miniupnpd/firewall.include', ignoring section
Automatically including '/usr/share/nftables.d/table-post/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/dstnat/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/forward/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/srcnat/20-miniupnpd.nft'
# Warning: table ip6 mangle is managed by iptables-nft, do not touch!
# Warning: table ip mangle is managed by iptables-nft, do not touch!

darthclide, 2024-04-17:
Ping results from OMR router to specific runescape world:
[image]

Ping results from OMR router using my eth1 (DSL) interface. That you can see is set in my OMR-bypass with "runescape.com":
[image]

Ysurac, 2024-04-17:
I would need ```nft list ruleset```

darthclide, 2024-04-17:
# Warning: table ip6 mangle is managed by iptables-nft, do not touch!
table ip6 mangle {
        chain PREROUTING {
                type filter hook prerouting priority mangle; policy accept;
                iifname "vtun*" meta l4proto tcp counter packets 0 bytes 0 xt target "MARK"
                iifname "eth1" xt match "dscp" counter packets 0 bytes 0 xt target "DSCP"
        }

        chain INPUT {
                type filter hook input priority mangle; policy accept;
        }

        chain FORWARD {
                type filter hook forward priority mangle; policy accept;
        }

        chain OUTPUT {
                type route hook output priority mangle; policy accept;
                meta l4proto udp xt match "multiport" counter packets 816 bytes 85807 xt target "DSCP"
        }

        chain POSTROUTING {
                type filter hook postrouting priority mangle; policy accept;
                oifname "eth1" meta mark & 0x000000ff == 0x00000000 counter packets 0 bytes 0 goto QOS_MARK_eth1
        }

        chain QOS_MARK_eth1 {
                counter packets 0 bytes 0 xt target "MARK"
                xt match "dscp" counter packets 0 bytes 0 xt target "MARK"
                xt match "dscp" counter packets 0 bytes 0 xt target "MARK"
                xt match "dscp" counter packets 0 bytes 0 xt target "MARK"
                xt match "dscp" counter packets 0 bytes 0 xt target "MARK"
                xt match "tos" counter packets 0 bytes 0 xt target "MARK"
        }
}
# Warning: table ip mangle is managed by iptables-nft, do not touch!
table ip mangle {
        chain PREROUTING {
                type filter hook prerouting priority mangle; policy accept;
                iifname "vtun*" meta l4proto tcp counter packets 0 bytes 0 xt target "MARK"
                iifname "eth1" xt match "dscp" counter packets 4051260 bytes 4623327306 xt target "DSCP"
        }

        chain INPUT {
                type filter hook input priority mangle; policy accept;
                counter packets 718059 bytes 798380720 jump omr-bypass-dpi
        }

        chain FORWARD {
                type filter hook forward priority mangle; policy accept;
                counter packets 697279 bytes 720679728 jump omr-bypass-dpi
        }

        chain OUTPUT {
                type route hook output priority mangle; policy accept;
                meta l4proto udp xt match "multiport" counter packets 59551 bytes 6449910 xt target "DSCP"
        }

        chain POSTROUTING {
                type filter hook postrouting priority mangle; policy accept;
                oifname "usb1" counter packets 1919659 bytes 211653513 xt target "TTL"
                oifname "eth1" meta mark & 0x000000ff == 0x00000000 counter packets 968208 bytes 232105709 goto QOS_MARK_eth1
        }

        chain QOS_MARK_eth1 {
                counter packets 968208 bytes 232105709 xt target "MARK"
                xt match "dscp" counter packets 0 bytes 0 xt target "MARK"
                xt match "dscp" counter packets 1517 bytes 193181 xt target "MARK"
                xt match "dscp" counter packets 0 bytes 0 xt target "MARK"
                xt match "dscp" counter packets 0 bytes 0 xt target "MARK"
[... 700 lines cut ...]
                iifname "eth0" ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 udp sport 1935 udp dport 1935 counter packets 0 bytes 0 ip dscp set cs4 comment "!fw4: omr_dscp_rule6"
                iifname "eth0" ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 udp sport 10000 udp dport 10000 counter packets 0 bytes 0 ip dscp set cs4 comment "!fw4: omr_dscp_rule7"
                iifname "eth0" ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 tcp sport 4444 tcp dport 4444 counter packets 0 bytes 0 ip dscp set cs3 comment "!fw4: omr_dscp_rule8"
        }

        chain mangle_output {
                type route hook output priority mangle; policy accept;
        }

        chain mangle_forward {
                type filter hook forward priority mangle; policy accept;
                iifname "eth0" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone lan IPv4/IPv6 ingress MTU fixing"
                iifname { "usb0", "usb1", "eth1", "eth2", "eth3" } tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 ingress MTU fixing"
                iifname "tun0" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone vpn IPv4/IPv6 ingress MTU fixing"
        }

        chain upnp_forward {
        }

        chain upnp_prerouting {
        }

        chain upnp_postrouting {
        }
}

darthclide, 2024-04-17:
I can put that in a pastebin if you need. I don't want to edit post if you are currently looking at it though.

Ysurac, 2024-04-17:
All seems ok...
What is the result of ```ip rule``` and ```ip route show table xxx``` (where xxx is the number after each word lookup in ip rule result)?

darthclide, 2024-04-17:
ip rule
0:      from all lookup local
0:      from all fwmark 0x1 lookup 100
0:      from 192.168.10.158 lookup 3
0:      from all oif tun0 lookup 1500
0:      from all oif usb0 lookup 5
0:      from 192.168.82.194 lookup 6
0:      from 172.20.10.3 lookup 4
0:      from all oif eth1 lookup 3
0:      from 10.255.255.2 lookup 1500
0:      from all oif eth2 lookup 4
0:      from 192.168.42.196 lookup 5
0:      from all oif usb1 lookup 6
1:      from all fwmark 0x45399999 lookup 9999
1:      from all fwmark 0x45393 lookup 3
1:      from all fwmark 0x45391500 lookup 1500
1:      from all fwmark 0x45394 lookup 4
1:      from all fwmark 0x45395 lookup 5
1:      from all fwmark 0x45396 lookup 6
1:      from all fwmark 0x45397 lookup 7
1:      from all fwmark 0x4539 lookup 991337
100:    from all lookup lan
10000:  from 192.168.100.1 lookup lan
20000:  from all to 192.168.100.1/24 lookup lan
32766:  from all lookup main
32767:  from all lookup default
90002:  from all iif lo lookup lan

root@OpenMPTCProuter:~# ip route show table 100
local default dev lo scope host

root@OpenMPTCProuter:~# ip route show table 3
default via 192.168.10.1 dev eth1
192.168.10.0/24 dev eth1 scope link

root@OpenMPTCProuter:~# ip route show table 4
default via 172.20.10.2 dev eth2
172.20.10.0/24 dev eth2 scope link

root@OpenMPTCProuter:~# ip route show table 5
default via 192.168.42.129 dev usb0
192.168.42.0/24 dev usb0 scope link

root@OpenMPTCProuter:~# ip route show table 6
default via 192.168.82.21 dev usb1
192.168.82.0/24 dev usb1 scope link

root@OpenMPTCProuter:~# ip route show table 7
Dump terminated

root@OpenMPTCProuter:~# ip route show table 1500
default via 10.255.255.1 dev tun0
10.255.255.2 dev tun0 scope link

root@OpenMPTCProuter:~# ip route show table 9999
Dump terminated

root@OpenMPTCProuter:~# ip route show table 991337
root@OpenMPTCProuter:~#

root@OpenMPTCProuter:~# ip route show table lan
192.168.100.0/24 dev eth0 proto static scope link metric 9999

root@OpenMPTCProuter:~# ip route show table local
local 10.255.255.2 dev tun0 proto kernel scope host src 10.255.255.2
local 127.0.0.0/8 dev lo proto kernel scope host src 127.0.0.1
local 127.0.0.1 dev lo proto kernel scope host src 127.0.0.1
broadcast 127.255.255.255 dev lo proto kernel scope link src 127.0.0.1
local 172.20.10.3 dev eth2 proto kernel scope host src 172.20.10.3
broadcast 172.20.10.255 dev eth2 proto kernel scope link src 172.20.10.3
local 192.168.10.158 dev eth1 proto kernel scope host src 192.168.10.158
broadcast 192.168.10.255 dev eth1 proto kernel scope link src 192.168.10.158
local 192.168.42.196 dev usb0 proto kernel scope host src 192.168.42.196
broadcast 192.168.42.255 dev usb0 proto kernel scope link src 192.168.42.196
local 192.168.82.194 dev usb1 proto kernel scope host src 192.168.82.194
broadcast 192.168.82.255 dev usb1 proto kernel scope link src 192.168.82.194
local 192.168.100.1 dev eth0 proto kernel scope host src 192.168.100.1
broadcast 192.168.100.255 dev eth0 proto kernel scope link src 192.168.100.1

root@OpenMPTCProuter:~# ip route show table main
default via 10.255.255.1 dev tun0
default via 192.168.10.1 dev eth1 metric 3
default via 172.20.10.2 dev eth2 metric 4
default via 192.168.42.129 dev usb0 metric 5
default via 192.168.82.21 dev usb1 metric 6
default via 10.255.255.1 dev tun0 metric 1500
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
10.255.255.2 dev tun0 scope link metric 1500
127.0.0.0/8 dev lo proto static scope link metric 8
***.**.***.*** metric 1
        nexthop via 192.168.10.1 dev eth1 weight 100
        nexthop via 172.20.10.2 dev eth2 weight 1
        nexthop via 192.168.42.129 dev usb0 weight 1
        nexthop via 192.168.82.21 dev usb1 weight 1
172.20.10.0/24 dev eth2 scope link metric 4
192.168.10.0/24 dev eth1 scope link metric 3
192.168.42.0/24 dev usb0 scope link metric 5
192.168.82.0/24 dev usb1 scope link metric 6

(I removed my public IP from the previous command)

Ysurac, 2024-04-17:
All seems to be ok.
If you are using Shadowsocks-libev, can you try with Shadowsocks-Rust (System->OpenMPTCProuter, Wizard tab, "advanced settings" checkbox) and reboot ?
I will make some tests tomorrow.

darthclide, 2024-04-17:
> All seems to be ok. If you are using Shadowsocks-libev, can you try with Shadowsocks-Rust (System->OpenMPTCProuter, Wizard tab, "advanced settings" checkbox) and reboot ? I will make some tests tomorrow.

Okay will do later when people are in bed.

darthclide, 2024-04-18:
Before I did the tests you wanted me to do tonight, in the middle of a match in War Thunder, my ping suddenly jumped back to 100ms (when usually I get 30-35ms when on pure non-VPN DSL). So I opened CurrPorts to see what new port came, but nothing new was there. All ports reported are within my scope 50000:66000 set in OMR-bypass. So what happened? Here is a screenshot of what CurrPorts shows. Let me know if you see a port I might have missed:

[image]

As for Shadowsocks Rust, it had no impact. Both on my Runescape game, as well as War Thunder.

darthclide, 2024-04-19:
Guess you got busy with other stuff, but just as a test I tried disabling all connections except for eth1, and as expected the ping didn't change. If anything the spikes got worse going up to 200ms at times. But as soon as I disable bonded internet and switch over to my 2nd ethernet port on pure DSL, my ping immediately drops and stays at 30-35ms.

Ysurac, 2024-04-20:
As War Thunder is free, I will test it. Before testing, I have something to know ? We get latency where ?

darthclide, 2024-04-20:
Oof, I refreshed page just as I went to bed. Just missed this message.

We have determined that overall ping issues can be mitigated with a different VPN (in the other github issue). The struggle in this github issue is that OMR-bypass seems to not work, or randomly quits working even after you have added all ports shown in CurrPorts or a straight up domain name (world104.runescape.com) to OMR-bypass.

I didn't realize you were offering to install the games, but if it isn't too late, I think runescape would be easier to test because it is much smaller in size. War Thunder even without HD textures is 26 GB I think. But if you are using War Thunder, you will know OMR-bypass it not working if the ping is jumping +25-50ms. Or you can do what I did and compare your best latency connection on its own, then compare it to OMR router.

Ysurac, 2024-04-20:
When you have bypass not working, what do you have in Status->System log at the same time ? Anything about firewall or  omr-bypass ?

darthclide, 2024-04-20:
From launch of game to middle of match this is the only thing that appears in system log:
Apr 20 22:10:04 OpenMPTCProuter daemon.err /usr/bin/ss-redir[31249]: send: Broken pipe
Apr 20 22:13:02 OpenMPTCProuter daemon.notice netifd: WAN3 (31064): udhcpc: sending renew to server 192.168.82.21
Apr 20 22:13:02 OpenMPTCProuter daemon.notice netifd: WAN3 (31064): udhcpc: lease of 192.168.82.32 obtained from 192.168.82.21, lease time 3599
Apr 20 22:13:52 OpenMPTCProuter daemon.notice netifd: wan2 (31045): udhcpc: sending renew to server 192.168.42.129
Apr 20 22:13:52 OpenMPTCProuter daemon.notice netifd: wan2 (31045): udhcpc: lease of 192.168.42.170 obtained from 192.168.42.129, lease time 3600

Have you had any luck with either Runescape or War Thunder?

darthclide, 2024-04-20:
Just happened to look at my clock in system log though. It says:
Apr 20 23:05:11

But in System -> System -> General Settings it says:
[image]

Perhaps the update from .59 to .60 caused it? Would be nice to fix though to help troubleshoot problems since the time is off by 4 hours.

darthclide, 2024-04-24:
I know this issue is far down on the list of priorities, but figured I would ask if you were able to test any games yet.

closed 2024-07-29 by github-actions[bot]