Ysurac/openmptcprouter #3275: Hetzner VPS and OMR script erreur with no secure backport repo

vinceducat, 2024-04-14:
## Expected Behavior
try to install omr vps on debian 12 isp Hetzner
## Current Behavior
error during script
## Possible Solution
update of the code script since mentioned in the log message

## Steps to Reproduce the Problem

  1. take a vps debian 12 at hetzner 
  2.  lunch script 
  3. error

## Context (Environment)
see log  

## Specifications

  - OpenMPTCProuter version: 0.60
  - OpenMPTCProuter VPS version:  0.1030
  - OpenMPTCProuter VPS provider: hetzner
  - OpenMPTCProuter platform: x86_64
  - Country:FR and gernany for vps

vinceducat, 2024-04-14:
root@debian-2gb-hel1-1:~# KERNEL="6.1"
root@debian-2gb-hel1-1:~# sh debian-x86_64.sh
Check user...
Check Linux version...
Check architecture...
Check virtualized environment
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Remove lock and update packages list...
Hit:1 http://security.debian.org/debian-security bookworm-security InRelease
Hit:2 http://deb.debian.org/debian bookworm InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://mirror.hetzner.com/debian/packages bookworm InRelease
Hit:5 http://mirror.hetzner.com/debian/packages bookworm-updates InRelease
Hit:6 http://mirror.hetzner.com/debian/security bookworm-security InRelease
Reading package lists... Done
Install apt-transport-https, gnupg and openssh-server...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gnupg is already the newest version (2.2.40-1.1).
openssh-server is already the newest version (1:9.2p1-2+deb12u2).
openssh-server set to manually installed.
The following NEW packages will be installed:
  apt-transport-https
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 25.2 kB of archives.
After this operation, 35.8 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main amd64 apt-transport-https all 2.6.1 [25.2 kB]
Fetched 25.2 kB in 0s (1472 kB/s)
Selecting previously unselected package apt-transport-https.
(Reading database ... 34130 files and directories currently installed.)
Preparing to unpack .../apt-transport-https_2.6.1_all.deb ...
Unpacking apt-transport-https (2.6.1) ...
Setting up apt-transport-https (2.6.1) ...
Add OpenMPTCProuter repo...
Explanation: Prefer OpenMPTCProuter provided packages over the Debian native ones
Package: *
Pin: origin repo.openmptcprouter.com
Pin-Priority: 1001
Explanation: Prefer libuv1 Debian native package
Package: libuv1
Pin: version *
Pin-Priority: 1003
--2024-04-14 01:01:57--  https://repo.openmptcprouter.com/openmptcprouter.gpg.key
Resolving repo.openmptcprouter.com (repo.openmptcprouter.com)... 2a01:4f9:4b:4b0b::2, 95.217.227.2
Connecting to repo.openmptcprouter.com (repo.openmptcprouter.com)|2a01:4f9:4b:4b0b::2|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1241 (1.2K) [application/octet-stream]
Saving to: '/etc/apt/trusted.gpg.d/openmptcprouter.gpg'

/etc/apt/trusted.gpg.d/openmptcprouter.gpg   100%[============================================================================================>]   1.21K  --.-KB/s    in 0s

2024-04-14 01:01:57 (59.3 MB/s) - '/etc/apt/trusted.gpg.d/openmptcprouter.gpg' saved [1241/1241]

Install mptcp kernel and shadowsocks...
Hit:1 http://security.debian.org/debian-security bookworm-security InRelease
Hit:2 http://deb.debian.org/debian bookworm InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Get:4 http://deb.debian.org/debian bullseye InRelease [116 kB]
Ign:5 http://deb.debian.org/debian buster-backports InRelease
Err:6 http://deb.debian.org/debian buster-backports Release
  404  Not Found [IP: 2a04:4e42:3a::644 80]
Hit:7 http://mirror.hetzner.com/debian/packages bookworm InRelease
Hit:8 http://mirror.hetzner.com/debian/packages bookworm-updates InRelease
Hit:9 http://mirror.hetzner.com/debian/security bookworm-security InRelease
Get:10 http://deb.debian.org/debian bullseye/main amd64 Packages [8068 kB]
Get:11 http://deb.debian.org/debian bullseye/main Translation-en [6236 kB]
Get:12 https://repo.openmptcprouter.com buster InRelease [2128 B]
Get:13 https://repo.openmptcprouter.com buster/main amd64 Packages [11.3 kB]
Reading package lists... Done
E: The repository 'http://deb.debian.org/debian buster-backports Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Ysurac, 2024-04-14:
Fixed

closed 2024-04-14 by Ysurac

vinceducat, 2024-04-14:
@Ysurac thanks, i will retry did you modify omr vps install script ? did i reload it ?

vinceducat, 2024-04-14:
@Ysurac  install is ok but VPN status is in orange at OMR router side ..

vinceducat, 2024-04-14:
[image]

vinceducat, 2024-04-14:
no prob with milkywan
[image]