Ysurac/openmptcprouter #3276: Can't find omr-bypass rules, restart omr-bypass v6.0

jle1511, 2024-04-14:
Expect omr-bypass to work but failed

## Current Behavior

fresh reinstall on x86, testing omrbypass:
1) domain ifconfig.me  > failed, as curl still return with VPS IP
2) mac address: failed, IP on mac still ping to VPS IP
3) all other services are all failed.

System log:
Apr 14 01:12:56 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Apr 14 01:13:00 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Apr 14 01:13:00 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Apr 14 01:13:02 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Apr 14 01:13:02 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Apr 14 01:13:07 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps

## Possible Solution

## Steps to Reproduce the Problem

  1. just set any omr-bypass setting and it doesn't apply

## Context (Environment)
 bypass VPS IP to master interface

## Specifications

  - OpenMPTCProuter version: [v0.60-6.1 r0+24843-acf40c022e]
  - OpenMPTCProuter VPS version: Version 0.1030 6.1.0-9-amd64
  - OpenMPTCProuter VPS provider: buyvm
  - OpenMPTCProuter platform:  x86_64
  - Country:  USA

jle1511, 2024-04-14:
additional log:

```
Apr 14 01:20:00 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Apr 14 01:20:00 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Apr 14 01:20:01 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Apr 14 01:20:02 OpenMPTCProuter daemon.info omr-tracker-xray: Reload xray rules
Apr 14 01:20:02 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Apr 14 01:20:02 OpenMPTCProuter user.notice omr-bypass: Restart dnsmasq...
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Apr 14 01:20:02 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5353
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Apr 14 01:20:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Apr 14 01:20:07 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
Apr 14 01:20:34 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth0) cc:50:e3:d8:be:50 
Apr 14 01:20:34 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth0) 192.168.100.109 cc:50:e3:d8:be:50 
Apr 14 01:20:34 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth0) 192.168.100.109 cc:50:e3:d8:be:50 
Apr 14 01:20:34 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth0) 192.168.100.109 cc:50:e3:d8:be:50 
Apr 14 01:21:06 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth0) 192.168.100.201 60:e8:5b:7e:89:82 
Apr 14 01:21:06 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth0) 192.168.100.201 60:e8:5b:7e:89:82 airthings-view
```

jle1511, 2024-04-14:
uci show omr-bypass and nft list ruleset

```
root@OpenMPTCProuter:~# uci show omr-bypass
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com' 'canalplus.com' 'canalplus-cdn.net' 'canalplus.pro' 'canal-plus.net'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.amazonvideo=proto
omr-bypass.amazonvideo.url='cloudfront.net' 'llnw.net'
omr-bypass.free=proto
omr-bypass.free.url='free.fr' 'freebox.fr' 'oqee.tv' 'oqee.net'
omr-bypass.orange=proto
omr-bypass.orange.url='orange.fr' 'sosh.fr' 'liveperson.net' 'liveperson.com' 'lpsn.net' 'lpsnmedia.net' 'francetelecom.fr'
omr-bypass.eth0=interface
omr-bypass.eth0.id='9999'
omr-bypass.eth1=interface
omr-bypass.eth1.id='8'
omr-bypass.eth2=interface
omr-bypass.eth2.id='9'
omr-bypass.eth3=interface
omr-bypass.eth3.id='10'
omr-bypass.tun0=interface
omr-bypass.tun0.id='1500'
omr-bypass.global=global
omr-bypass.global.vpn_ipv4_md5='68b329da9893e34099c7d8ad5cb9c940'
omr-bypass.global.vpn_ipv6_md5='68b329da9893e34099c7d8ad5cb9c940'
omr-bypass.@macs[0]=macs
omr-bypass.@macs[0].mac='F4:D4:88:6A:40:31'
omr-bypass.@macs[0].note='jimmy mac'
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='ifconfig.me'
omr-bypass.@domains[0].note='ifconfig.me'
omr-bypass.@domains[0].family='ipv4ipv6'
omr-bypass.@domains[0].interface='eth1'
omr-bypass.@domains[1]=domains
omr-bypass.@domains[1].name='suncoastcreditunion.com'
omr-bypass.@domains[1].note='suncoastcreditunion.com'
omr-bypass.@domains[1].family='ipv4ipv6'
omr-bypass.@domains[2]=domains
omr-bypass.@domains[2].name='reddit.com'
omr-bypass.@domains[2].note='reddit.com'
omr-bypass.@domains[2].family='ipv4ipv6'
omr-bypass.@dpis[0]=dpis
omr-bypass.@dpis[0].proto='amazonalexa'
omr-bypass.@dpis[0].family='ipv4ipv6'
omr-bypass.@dpis[1]=dpis
omr-bypass.@dpis[1].proto='amazonaws'
omr-bypass.@dpis[1].family='ipv4ipv6'
omr-bypass.@dpis[2]=dpis
omr-bypass.@dpis[2].proto='amazonvideo'
omr-bypass.@dpis[2].family='ipv4ipv6'
omr-bypass.@dpis[3]=dpis
omr-bypass.@dpis[3].proto='cloudflare'
omr-bypass.@dpis[3].family='ipv4ipv6'
omr-bypass.@dpis[4]=dpis
omr-bypass.@dpis[4].proto='cloudflarewarp'
omr-bypass.@dpis[4].family='ipv4ipv6'
omr-bypass.@dpis[5]=dpis
omr-bypass.@dpis[5].proto='hulu'
omr-bypass.@dpis[5].family='ipv4ipv6'
omr-bypass.@dpis[6]=dpis
omr-bypass.@dpis[6].proto='imap'
[... 717 lines cut ...]
                iifname "eth0" ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 udp sport 0-65535 udp dport { 65001, 65301 } counter packets 0 bytes 0 ip dscp set cs7 comment "!fw4: omr_dscp_rule6"
                iifname "eth0" ip saddr 0.0.0.0/0 ip daddr 0.0.0.0/0 tcp sport 0-65535 tcp dport { 65101, 65228 } counter packets 0 bytes 0 ip dscp set cs6 comment "!fw4: omr_dscp_rule7"
        }

        chain mangle_output {
                type route hook output priority mangle; policy accept;
        }

        chain mangle_forward {
                type filter hook forward priority mangle; policy accept;
                iifname "eth0" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone lan IPv4/IPv6 ingress MTU fixing"
                iifname { "eth1", "eth2", "eth3" } tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 ingress MTU fixing"
                iifname "tun0" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone vpn IPv4/IPv6 ingress MTU fixing"
        }

        chain upnp_forward {
        }

        chain upnp_prerouting {
        }

        chain upnp_postrouting {
        }
}
```

jle1511, 2024-04-14:
i think i found the problem, if i switch Default Proxy from Xray Trojan to Shadowsocks rust 2022, omrbypass start working again. Any ideas why ? and all the error disappear

Ysurac, 2024-04-14:
I think I found the issue: when you change the proxy via the Wizard, omr-bypass is not restarted and config is specific to the proxy.
If you change the proxy and reboot, all should be ok.
I will fix that for next minor release.

hle5128, 2024-04-18:
> I think I found the issue: when you change the proxy via the Wizard, omr-bypass is not restarted and config is specific to the proxy. If you change the proxy and reboot, all should be ok. I will fix that for next minor release.

Hi, 
I tried setting up some domain testing using ifconfig.me, saved and applied the settings, then rebooted. However, I'm still experiencing the same error and the omr-bypass is not working. I tried applying the same settings and rebooting on other proxies, but none of them worked except for shadowsocks. 

Here are sample of the logs:
`
Apr 18 01:42:47 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Apr 17 21:42:51 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) c8:c9:a3:54:3b:df 
Apr 17 21:42:51 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.110 c8:c9:a3:54:3b:df 
Apr 17 21:42:51 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) c8:c9:a3:54:3b:df 
Apr 17 21:42:51 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.110 c8:c9:a3:54:3b:df 
Apr 17 21:42:51 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) c8:c9:a3:54:3b:df 
Apr 17 21:42:51 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.110 c8:c9:a3:54:3b:df 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) cc:50:e3:d8:be:50 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.107 cc:50:e3:d8:be:50 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) cc:50:e3:d8:be:50 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.107 cc:50:e3:d8:be:50 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) c8:c9:a3:54:3b:df 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.110 c8:c9:a3:54:3b:df 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.110 c8:c9:a3:54:3b:df 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.110 c8:c9:a3:54:3b:df 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) cc:50:e3:d8:be:50 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.107 cc:50:e3:d8:be:50 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.110 c8:c9:a3:54:3b:df 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.110 c8:c9:a3:54:3b:df 
Apr 17 21:42:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPRELEASE(eth1) 192.168.100.119 7c:78:b2:87:84:b4 unknown lease
Apr 17 21:42:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) fc:45:c3:01:3a:6e 
Apr 17 21:42:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.115 fc:45:c3:01:3a:6e 
Apr 17 21:42:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.107 cc:50:e3:d8:be:50 
Apr 17 21:42:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.107 cc:50:e3:d8:be:50 
Apr 17 21:42:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) fc:45:c3:01:3a:6e 
Apr 17 21:42:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.115 fc:45:c3:01:3a:6e 
Apr 17 21:42:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.107 cc:50:e3:d8:be:50 
Apr 17 21:42:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.107 cc:50:e3:d8:be:50 
Apr 17 21:42:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.115 fc:45:c3:01:3a:6e 
Apr 17 21:42:59 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.115 fc:45:c3:01:3a:6e 
Apr 17 21:43:01 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) fc:45:c3:01:3a:6e 
Apr 17 21:43:01 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.115 fc:45:c3:01:3a:6e 
Apr 17 21:43:01 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.115 fc:45:c3:01:3a:6e 
Apr 17 21:43:01 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.115 fc:45:c3:01:3a:6e 
Apr 18 01:43:02 OpenMPTCProuter daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 212.27.48.10, 198.27.92.1, 151.101.129.164)
Apr 17 21:43:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) c8:c9:a3:54:3b:df 
Apr 17 21:43:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.110 c8:c9:a3:54:3b:df 
Apr 17 21:43:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPRELEASE(eth1) 192.168.100.177 2c:aa:8e:bb:0d:cb unknown lease
Apr 17 21:43:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) fc:45:c3:01:3a:6e 
Apr 17 21:43:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.115 fc:45:c3:01:3a:6e 
Apr 17 21:43:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.115 fc:45:c3:01:3a:6e 
Apr 17 21:43:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.115 fc:45:c3:01:3a:6e 
Apr 17 21:43:03 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) c8:c9:a3:54:3b:df 
Apr 17 21:43:03 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.110 c8:c9:a3:54:3b:df 
Apr 17 21:43:03 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.110 c8:c9:a3:54:3b:df 
[... cut ...]
                iifname "eth1" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone lan IPv4/IPv6 ingress MTU fixing"
                iifname { "eth0", "eth2" } tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 ingress MTU fixing"
                iifname "tun0" tcp flags syn / fin,syn,rst tcp option maxseg size set rt mtu comment "!fw4: Zone vpn IPv4/IPv6 ingress MTU fixing"
        }

        chain upnp_forward {
        }

        chain upnp_prerouting {
        }

        chain upnp_postrouting {
        }
}
root@OpenMPTCProuter:~# `

closed 2024-07-22 by github-actions[bot]