Ysurac/openmptcprouter #3277: Work incorrect. Wan interfaces are being disabled. Nanopi R4S 4gb ram. v0.60rc2-6.1, kernel 6.1.

Silvanskii, 2024-04-14:
Reinstalled debian VPS to version 12 and installed KERNEL="6.1" sh. During operation, the wan ports are interrupted at random all the time. Then I put version v0.59.1 back on and everything started working correctly. I am attaching a log file.
[image]
[image]
[LOG Core.txt](https://github.com/Ysurac/openmptcprouter/files/14970268/LOG.Core.txt)
[LOG openmptcprouter v0.60-6.1 r0+24843.txt](https://github.com/Ysurac/openmptcprouter/files/14970269/LOG.openmptcprouter.v0.60-6.1.r0%2B24843.txt)
[LOG openmptcprouter v0.60rc2-6.1.txt](https://github.com/Ysurac/openmptcprouter/files/14970270/LOG.openmptcprouter.v0.60rc2-6.1.txt)
P.S To remove the error message next to the VPS server - can't get public IP address from ShadowSoks Rust - you need to change the IP and key from the VPS on the 5.4 kernel in the settings, wait a couple of minutes and change the address to the key and IP vps 6.1 and the error disappears.

Silvanskii, 2024-04-14:
[image]
I also noticed that with the new version of nanopi r4s, it began to warm up to 80 degrees in 5-10 minutes of operation and the processor was loaded into idle by 50%. I returned the old version 59.1 and the processor is running about 60 degrees
[image]

Ysurac, 2024-04-14:
Please use the templae.
To check why an interface is show as red, check Status->System log

Silvanskii, 2024-04-15:
> Please use the templae. To check why an interface is show as red, check Status->System log
Bag only with 0.60+ versoin.
[15.04_dmesg.txt](https://github.com/Ysurac/openmptcprouter/files/14980425/15.04_dmesg.txt)
[15.04_logs.txt](https://github.com/Ysurac/openmptcprouter/files/14980426/15.04_logs.txt)

closed 2024-07-19 by github-actions[bot]