Ysurac/openmptcprouter #3279: Only one ISP do their work

zRxnx, 2024-04-14:
## Expected Behavior
Both ISP works at the best speed

## Current Behavior
I have 2 ISPs connected.
But only one is getting 779mbits and the other gets 1.6mbits
[image]

## Specifications
  - OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 0.1030 6.1.0-20-cloud-amd64
  - OpenMPTCProuter VPS provider: Exoscale
  - OpenMPTCProuter platform: RPI4B
  
  Both ISPs have "DHCP Client".
  ISP1 is Master.
  
[image]
[image]

zRxnx, 2024-04-14:
Now its reversed, ISP2 doesnt everything ISP1 just gives some kilobits

Ysurac, 2024-04-14:
I don't know what you mean exactly: it's not aggregated or only not using both links ?
Links are used based on latency, if you don't need more bandwidth, only one link is used.

zRxnx, 2024-04-14:
The two Internet connections doesnt "bond".

Normal both connections have ~600mbits, but in the network>mptcp>Bandwith i see that one connection does 99% and the other one has just some kilobits.

So it looks like one connection works, the other one not.
[image]

For example ISP1 has 161.32 mBits and ISP2 has 37.47 kBits

Ysurac, 2024-04-14:
I see that both connections seems to be in same IP range. What is the result of ```omr-test-speed eth0.10``` and ```omr-test-speed eth0.20``` launched at the same time via SSH on the router ?

zRxnx, 2024-04-14:
Will do the speed test in few mins.

But the two internet connections are from the same company in the same house, the same contract and the same modem, maybe this is important.

zRxnx, 2024-04-14:
Okay, i think i know the problem.

I disconnected one connection port, immediatly i get 520mbits down.

So ig its bottlenecking each other when i use them both?

zRxnx, 2024-04-15:
Problem with my ISP

closed 2024-04-15 by zRxnx