Ysurac/openmptcprouter #3280: ISP Milkywan OMR VPS and OMR router best setup for IPV6 - advice

vinceducat, 2024-04-14:
## Expected Behavior
setup the best IPV6 for OMR router to OMR vps and have ipv6 on lan to

## Current Behavior
want to have advise
## Specifications

  - OpenMPTCProuter version: 0.60 kernel 6.1 
  - OpenMPTCProuter VPS version: 0.1030 debian 12 kernel 6.1
  - OpenMPTCProuter VPS provider: milkywan
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-04-14:
More info is needed...
What is the problem ? What is the ip/routes ?...

vinceducat, 2024-04-14:
@Ysurac 
du to an old setup of omr i want to check the ipv6 config  : 

_ first :  for a milkywan vps withj ipv4 and ipv6 do i enter the both addresses in the vps setup definition in the wizard setup  
_ 2n : do i activate ipv6 in the advance setup / witch prefix do i use
_ 3rd : do i change ULA prefix in global interface option
_ 3rd : do i change any think in the lan interface to activate dhcpv6 ? 

the problem is that first :
when i try to put ipv6 OMR vps in the omr router config ... it doesnt work ... lost a link between omr

Ysurac, 2024-04-14:
So you mean you lost VPS access when IPv6 is enabled even without a custom ULA set on OMR ?
You also lost IPv4 access or only IPv6 ?

vinceducat, 2024-04-14:
@Ysurac for 

1 _ in fact when i add the ipv6 of the vps in the wizard menu for serveur, the vps hang up sometime :-( (i need ton call milkywan to reboot) 
so i want to check all the setup before doing any modification ...

to answered, i lost aggregation when ipv6 adresse is put. and sometime the vps hang :-(
i have a ULA custom, do i reput the "standard" ? so what is the ULA original ?

vinceducat, 2024-04-14:
[image]

Ysurac, 2024-04-14:
If the custom ULA conflict with the IP of the VPS, this may explain some issues.
It's always better to first test with a normal ULA ( something like fd58:8a43:2f89::/48, it's randomized private IP)

vinceducat, 2024-04-14:
so i put fd58:8a43:2f89::/48 in the global network option prefix ULA that right ?
at this time i have 2a0e:e701:xxxx:/48

Ysurac, 2024-04-14:
Yes

vinceducat, 2024-04-14:
its done addd omr reboot...

do i put the ipv6 server adresse in the 2nd adresse field of the wizard ? to continu the test

Ysurac, 2024-04-14:
yes if you have WANs with IPv6 connections.

vinceducat, 2024-04-14:
so you say if wans have ipv6 connexion ..

so i have freebox adsl (very old freebox ;-))  for wan 1 + freemobile 4g/5g for wan 2... but both are  ipv4 setup and perhaps ipv4 only enable at the Free.fr side ... 
do i check the two have ipv6 enable ?

so if i understand putting an ipv6 adresse for serveur is only necessary if i have wans support ipv6 ?

this is not link to the setup of the ipv6 setup of the vps ... and the goal to obtain ipv6 form the vps range  from milkywan

am i right ?

Ysurac, 2024-04-14:
Yes, you put IPv6 address of server only if WANs support IPv6. It's another way to contact server.
It's not needed if you want to only use server IPv6.

vinceducat, 2024-04-14:
ok, that clear.

but at this time do i lease the ULA prefix, or is it better to put the milkywan prefix 2a0e:e701:xxxx:/48 ?

(note for myself : in fact the wizard advance conf prefix and global network option prefix ULA is the same field ... ;-)

vinceducat, 2024-04-14:
i the lan interface at dhcp server ipv6 and ipv6 ra do i made so check ? or change ?

closed 2024-07-18 by github-actions[bot]