Ysurac/openmptcprouter #3281: VPS install fails with repository error

hairy-dog, 2024-04-14:
## Expected Behavior
I paste in the upgrade command to upgrade the VPS. 

wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="6.1" sh

## Current Behavior
I get: 

https://www.openmptcprouter.com/server/debian-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 2400:52e0:1e03::886:1, 143.244.38.136
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|2400:52e0:1e03::886:1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 89907 (88K) [application/octet-stream]
Saving to: ‘STDOUT’

-                                      0%[                                                                      ]       0  --.-KB/s               Check user...
Check Linux version...
Check architecture...
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Update repo.openmptcprouter.com key
--2024-04-14 17:26:23--  https://repo.openmptcprouter.com/openmptcprouter.gpg.key
Resolving repo.openmptcprouter.com (repo.openmptcprouter.com)... 2400:52e0:1e03::886:1, 143.244.38.136
Connecting to repo.openmptcprouter.com (repo.openmptcprouter.com)|2400:52e0:1e03::886:1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1241 (1.2K) [application/octet-stream]
Saving to: '/etc/apt/trusted.gpg.d/openmptcprouter.gpg'

/etc/apt/trusted.gpg.d/openmptcprout 100%[=====================================================================>]   1.21K  --.-KB/s    in 0s

2024-04-14 17:26:23 (45.9 MB/s) - '/etc/apt/trusted.gpg.d/openmptcprouter.gpg' saved [1241/1241]

Remove lock and update packages list...
Hit:1 http://security.debian.org/debian-security buster/updates InRelease
Hit:2 http://deb.debian.org/debian buster InRelease
Hit:3 http://deb.debian.org/debian buster-updates InRelease
Ign:4 http://deb.debian.org/debian buster-backports InRelease
Err:5 http://deb.debian.org/debian buster-backports Release
  404  Not Found [IP: 2a04:4e42:f::644 80]
Hit:6 https://repo.openmptcprouter.com buster InRelease
Reading package lists... Done
E: The repository 'http://deb.debian.org/debian buster-backports Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
-                                     81%[=======================================================>              ]  71.39K  --.-KB/s    in 0.08s

Cannot write to ‘-’ (Broken pipe).

## Specifications

  - OpenMPTCProuter version:  openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1028
  - OpenMPTCProuter VPS provider: intovps
  - OpenMPTCProuter platform: RPi4

vinceducat, 2024-04-14:
@hairy-dog i have the same problem with hetzner isp and @Ysurac  solve the probleme, i think with a new installation script ... but dont know if it is solve for all the isp ...

Ysurac, 2024-04-14:
The issue related to buster-backports is solved since a few hours. Should be ok now.

hairy-dog, 2024-04-14:
This failed a few minutes ago. I have renamed buster-backports.list out of the way and tried again. It seems to be working, but I won't know until it finishes.

hairy-dog, 2024-04-14:
Does it need buster-backports for anything?

hairy-dog, 2024-04-14:
Well, it installed after I renamed buster-backports.list away, but the install put it in again, and this time the install worked to the end.

The VPS install seems to have gone OK. Not so sure about the router install. That's for another discussion.

darthclide, 2024-04-15:
Same issue here. Going to wait until tomorrow and see if it is fixed. Or if Ysurac comes in and says an alternative method to get VPS installed I will follow instructions.

QuartConz, 2024-04-15:
Same here, got  server on IONOS and tried with debian & ubuntu.... always:.... failed: Connection timed out.

Ysurac, 2024-04-15:
@QuartConz I don't think it's same issue as there is no connection timeout in current log. What is the script output ?

QuartConz, 2024-04-15:
Works now, made a fresh install with server image. thx

closed 2024-07-19 by github-actions[bot]