Ysurac/openmptcprouter #3285: http://amiusingmptcp.de/ says I  am not using MPTCP - speeds are not as expected since upgrade

hhaniel, 2024-04-15:
## Expected Behavior

http://amiusingmptcp.de/ should say yes and I would expect the bandwith of all links aggregated or at least close to something like that.

## Current Behavior
I have upgraded to the latest version of OpenMPTCProuter on router and and on VPS and checking on http://amiusingmptcp.de/ it says "I am not using mptcp" also my speed tests are only showing speeds of the fastest link and aggregation does not seem to be working as expected. When I use tcpdump on my VPS I am am not seeing anything going via my slower link - There is only traffic going through links that are roughly the same speed.

## Specifications

  - OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022
  - OpenMPTCProuter VPS version: 0.1030 on Kernel 6.1 (Debian)
  - OpenMPTCProuter VPS provider: Fasthosts UK
  - OpenMPTCProuter platform: x86_64 on ESX

I originally attempted using upgrades on both sides and also by restoring configs on the router but I thought there may be some left overs from previous version that causes this issue I have rebuild and reconfigured everything from scratch now - I still have a snapshot of the old router but no longer the old VPS so I would have to rebuild my VPS with an older version to fall back. - I was previously running OpenMPTCProuter v0.59.1-5.4 where everything was working very well.

Ysurac, 2024-04-15:
http://amiusingmptcp.de/ Only check MPTCPv0, with v0.60 on kernel 6.1 MPTCPv1 is used.
To check aggregation use ```omr-test-speed``` and go to Network->MPTCP, "Bandwidth" page.

hhaniel, 2024-04-16:
Thanx for clarifying the MPTCPv0 vs MPTCPv1 issue on the test web page - It would be nice if there was a web page that would check for v1. The `curl http://www.multipath-tcp.org` way also does not work presumably for the same reason. 
I was not having a lot of luck with `omr-test-speed` either but it seems to work now although it lacks testing uploads but I have now worked around this now by running iperf between another vps and a system on my internal system and transferring large files and I can indead see on the bandwith page that aggregation is working. 
One thing that was having an impact was my VPS only having 512Mb of memory which was working fine on the older VPS version but with 6.0-6.1 that seems to make a difference as I noticed the VPS to occasionally start swapping small amounts of memory during speed tests although I have vm.swappiness set to 1.
The other problem which is having an impact is that one of my WiFi links seems to fluctuate in speed at the moment as there seems to be some inteference which beyond my control at the moment and I will have to do some tuning there but that seems confuse a speedtest with speedtest.net. (I have 1 ADSL link and 2 WiFi links) unfortunately that interference must have started at the same time when I was migrating to the new version. - I think this may also have confused SQM which I have disabled for the time being.

Ysurac, 2024-04-16:
1024Mb is recommended on VPS side. For me it was already not working with 512Mb.
Wifi links often give strange results, for now I've not really worked on this type of link (I will for next release).

hhaniel, 2024-04-16:
I have not put any wifi stuff in the OMR router bit as i did not want to break stuff there - only bit i replaced on the OMR router is syslogd with syslog-no because remote logging is not working with syslogd.
To collect to other WiFi networks i use separate  virtual machines with linux or openwrt and between them and the router i use subnets in the 172.16.0.0 range as that helps with clashes in the 192.168.0.0 range with my local network and wifi networks i may be connecting to.
One of the wifi’s i connect to is a public hotspot in my range and i do that with linux as i have scripts running on that that automatically log me in when the session times out.

closed 2024-07-21 by github-actions[bot]