Ysurac/openmptcprouter #3286: VPN isn't

DarthVaderFR, 2024-04-15:
Hello,

Usually, openmlptcprouter connects to the VPS and aggregate my different links with the VPS public ip.

I've decided, just before the week-end, to upgrade my system on the latest version. I've read enough, but the kernel of the VPS on firstheberg wasn't high enough so it didn't work. I've taken a news VPS with OVH, which support Debian 12 and reinstalled the new script for the repo, keeping my conf on the router side, and populate the different fields with the new ip and the new key. But, instead of the usual way of working, i now have a direct output through the wan1 which is default one, a VPN not responding, but the script on the VPS is reachable, and that type of logs on the router :

```
Apr 15 19:24:21 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:24:25 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:24:29 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:24:33 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:24:38 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:24:42 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:24:46 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:24:50 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:24:54 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:24:56 OpenMPTCProuter daemon.err /usr/bin/ss-redir[1088]: remote recv: Socket not connected
Apr 15 19:24:58 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:02 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:06 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:10 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:15 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:19 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:23 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:27 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:31 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:35 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:39 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:43 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:48 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:52 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:25:56 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:00 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:04 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:08 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:12 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:16 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:20 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:25 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:29 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:33 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:37 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:41 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:45 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:49 OpenMPTCProuter daemon.warn wpa_supplicant[3203]: phy1-sta0: Failed to initiate sched scan
Apr 15 19:26:49 OpenMPTCProuter daemon.notice wpa_supplicant[3203]: phy0-sta0: RSN: Group rekeying completed with 4a:73:8d:0f:62:c8 [GTK=CCMP]
Apr 15 19:26:49 OpenMPTCProuter daemon.notice wpa_supplicant[3203]: phy0-sta0: RSN: Group rekeying completed with 4a:73:8d:0f:62:c8 [GTK=CCMP]
Apr 15 19:26:49 OpenMPTCProuter daemon.notice wpa_supplicant[3203]: phy0-sta0: Unknown event 37
Apr 15 19:26:49 OpenMPTCProuter daemon.notice wpa_supplicant[3203]: phy0-sta0: Unknown event 37
```

## Specifications

  - OpenMPTCProuter version: [v0.60-6.1 r0+24843-acf40c022e](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter VPS version: Debian12 kernel 6.1
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI4B

Thank you for your help.

Ysurac, 2024-04-16:
For wpa_supplicant, it's related to the WIFI. You have it enabled ?
You upgraded an old config on router side or it's a new install ?

DarthVaderFR, 2024-04-16:
Hello,Yes i have 2 connections through wifi. I’ve upgraded on the router side. Thanks.

Ysurac, 2024-04-16:
In System->OpenMPTCProuter, Wizard tab, check "Advanced settings" box and check "Force retrieve settings" than save. 
As you changed the server it need to retrieve all keys informations.

DarthVaderFR, 2024-04-16:
Hello,Thanks for your answer. I’ve tried the solution but it didn’t solve my problem. Do you have any other idea ?Thanks.

Ysurac, 2024-04-16:
Disable wifi for now, and send me another log, this will help to find the issue (there is too many logs about wifi).

DarthVaderFR, 2024-04-16:
here is the logs without wifi :

`Apr 16 17:35:01 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Apr 16 17:35:01 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Apr 16 17:35:08 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Apr 16 17:35:08 OpenMPTCProuter user.notice omr-bypass: Restart dnsmasq...
Apr 16 17:35:08 OpenMPTCProuter daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Apr 16 17:35:09 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.89 cachesize 150
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5353
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 10 names
Apr 16 17:35:09 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Apr 16 17:35:09 OpenMPTCProuter daemon.warn dnsmasq-dhcp[1]: not giving name mHfUO.lan to the DHCP lease of 192.168.100.120 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.0.18
Apr 16 17:35:16 OpenMPTCProuter user.notice post-tracking-001-post-tracking: omrvpn down because gateway down, interface have no IPv6, interface have no IPv6 gateway
Apr 16 17:35:16 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Glorytun VPN down, restart it
Apr 16 17:35:17 OpenMPTCProuter daemon.err glorytun[1523]: getpeername: Socket not connected
Apr 16 17:35:17 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Apr 16 17:35:17 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Apr 16 17:35:17 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Apr 16 17:35:17 OpenMPTCProuter daemon.notice ttyd[14262]: [2024/04/16 17:35:17:0387] N: rops_handle_POLLIN_netlink: DELADDR
Apr 16 17:35:17 OpenMPTCProuter daemon.notice ttyd[14262]: [2024/04/16 17:35:17:0421] N: rops_handle_POLLIN_netlink: DELADDR
Apr 16 17:35:17 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='tun0'
Apr 16 17:35:17 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn
Apr 16 17:35:17 OpenMPTCProuter daemon.info ModemManager[21103]: hotplug: remove network interface tun0: event processed
Apr 16 17:35:17 OpenMPTCProuter daemon.info glorytun[21245]: INITIALIZED tun0
Apr 16 17:35:17 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Apr 16 17:35:17 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Apr 16 17:35:17 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Apr 16 17:35:17 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Apr 16 17:35:17 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Apr 16 17:35:17 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Apr 16 17:35:17 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='tun0'
Apr 16 17:35:17 OpenMPTCProuter daemon.info ModemManager[21272]: hotplug: add network interface tun0: event processed
Apr 16 17:35:18 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Change in routes, set ip route replace default scope global nexthop via 192.168.0.254 dev wan1 weight 100 nexthop via 172.20.10.1 dev eth1 weight 1 nexthop via 192.168.8.1 dev wan2 weight 1 (omrvpn_intf
Apr 16 17:35:33 OpenMPTCProuter user.notice MPTCP: Add route 192.168.0.0/24 via 192.168.0.254 dev wan1
Apr 16 17:35:34 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='ifb4tun0'
Apr 16 17:35:34 OpenMPTCProuter daemon.info ModemManager[3814]: hotplug: remove network interface ifb4tun0: event processed
Apr 16 17:35:34 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_dd540'
[... cut ...]
Apr 16 17:42:32 OpenMPTCProuter daemon.warn dnsmasq-dhcp[1]: not giving name mHfUO.lan to the DHCP lease of 192.168.100.120 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.0.18
Apr 16 17:42:43 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPRELEASE(eth0) 192.168.100.194 70:ee:50:16:17:ea 
Apr 16 17:42:43 OpenMPTCProuter daemon.warn dnsmasq-dhcp[1]: not giving name mHfUO.lan to the DHCP lease of 192.168.100.120 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.0.18`

Ysurac, 2024-04-16:
Seems to be the boot, there is not enough log here. I need to see some logs about proxy.
Can you also put a screenshot of your status page ?

DarthVaderFR, 2024-04-16:
[image]

DarthVaderFR, 2024-04-16:
i've restarted the router, here is the log since the reboot : 

``
Apr 16 20:30:36 OpenMPTCProuter syslog.info syslogd started: BusyBox v1.36.1
Apr 16 18:30:36 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S13openssl: Generating engines.cnf
Apr 16 18:30:36 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S13openssl: Generating providers.cnf
Apr 16 18:30:37 OpenMPTCProuter user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Apr 16 18:30:37 OpenMPTCProuter user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Apr 16 18:30:37 OpenMPTCProuter user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.89 cachesize 150
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5353
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Apr 16 20:30:37 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 8 names
Apr 16 18:30:37 OpenMPTCProuter authpriv.info dropbear[2939]: Not backgrounding
Apr 16 18:30:37 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Apr 16 18:30:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 16 18:30:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 16 18:30:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 16 18:30:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 16 18:30:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 16 18:30:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 16 18:30:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 16 18:30:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 16 18:30:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Apr 16 18:30:38 OpenMPTCProuter user.notice : Added device handler type: bonding
Apr 16 18:30:38 OpenMPTCProuter user.notice : Added device handler type: 8021ad
Apr 16 18:30:38 OpenMPTCProuter user.notice : Added device handler type: 8021q
Apr 16 18:30:38 OpenMPTCProuter user.notice : Added device handler type: macvlan
Apr 16 18:30:38 OpenMPTCProuter user.notice : Added device handler type: veth
Apr 16 18:30:38 OpenMPTCProuter user.notice : Added device handler type: bridge
Apr 16 18:30:38 OpenMPTCProuter user.notice : Added device handler type: Network device
Apr 16 18:30:38 OpenMPTCProuter user.notice : Added device handler type: tunnel
Apr 16 18:30:38 OpenMPTCProuter daemon.notice wpa_supplicant[3234]: Successfully initialized wpa_supplicant
Apr 16 18:30:38 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S25packet_steering: uci: Entry not found
Apr 16 18:30:39 OpenMPTCProuter daemon.warn netifd: You have delegated IPv6-prefixes but haven't assigned them to any interface. Did you forget to set option ip6assign on your lan-interfaces?
Apr 16 18:30:39 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is enabled
Apr 16 18:30:39 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is setting up now
Apr 16 18:30:39 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now up
Apr 16 18:30:39 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is enabled
Apr 16 18:30:39 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is setting up now
Apr 16 18:30:39 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is now up
Apr 16 18:30:39 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is enabled
Apr 16 18:30:39 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is setting up now
Apr 16 18:30:39 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is now up
[... cut ...]
Apr 16 18:32:13 OpenMPTCProuter daemon.notice ttyd[18428]: [2024/04/16 20:32:13:9381] N: [vh|1|default|eth0|eth0|7681]: lws_socket_bind: source ads 192.168.100.1
Apr 16 18:32:13 OpenMPTCProuter daemon.notice ttyd[18428]: [2024/04/16 20:32:13:9383] N: __lws_lc_tag:  ++ [wsi|1|listen|default|eth0|7681] (2)
Apr 16 18:32:13 OpenMPTCProuter daemon.notice ttyd[18428]: [2024/04/16 20:32:13:9383] N:  Listening on port: 7681
Apr 16 18:32:14 OpenMPTCProuter daemon.info procd: - init complete -
Apr 16 18:32:15 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
``

DarthVaderFR, 2024-04-17:
Hello,Did you have enough logs?I was thinking about reset everything from scratch and reimport my conf after that. But i’m fearing to return at the same points if this is some keys or values who will be reimported from the backup, and so wasting time. What do you think about it ?

Ysurac, 2024-04-17:
In fact still not enough: I can't see anything related to the VPN.
Can you try to change the VPN in System->OpenMPTCProuter, Wizard tab, "Advanced settings" checkbox ? Not OpenVPN TCP is used by default on new 0.60 installation.

Ysurac, 2024-04-17:
None mean no VPN is used.
How do you test aggregation over the VPN ?

DarthVaderFR, 2024-04-17:
with speedtest first (peak 8mbits), and then, with a download of a file on a distant server.

with all my devices connected, and in the best conditions, before the upgrade, i was able to reach ~50-80mbits/s

Ysurac, 2024-04-17:
You have disabled the proxy part ?

DarthVaderFR, 2024-04-17:
I didn’t

DarthVaderFR, 2024-04-17:
> If not, you are only testing proxy speed. If you are using Shadowsocks-libev, you should try Shadowsocks-Rust (Still in System->OpenMPTCProuter, Wizard tab, "Advanced settings" checkbox).

I was using « shadowsocks » but switching to rust (with openvpn) didn’t change anything.

DarthVaderFR, 2024-04-23:
Hello,

Hope you're doing well.

You can close that request. I reinstalled everything from scratch, no more problem with the VPN or aggregation.

Thanks for your help.

closed 2024-04-23 by DarthVaderFR