Ysurac/openmptcprouter #3287: Changing data 2x, slow speed, no stable...

MistiqueX, 2024-04-15:
Hi dear Ysurac
 about many month i try using mptcprouter - and i sad case have many problem with this.......
[image]
i have many errors with configuration - also when u changing something - it needed twice or more cycle to start work, maybe its a programm conflict.
i use raspberry 4pi + 3 em12-g and openmptcprouter v0.60-6.1 r0+24843-acf40c022e
1 - i cant have good speed, its not summ all modems and speed is to slow.
2 - cant see bands on modems
3 - always errors
4 - to much time to loose - about 3-4 month  
5 - does modem cant work more correctly?
6 - no way to aggregate frequency's and 3 modem working at once

  - OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 060.-6.1
  - OpenMPTCProuter VPS provider: timeweb
  - OpenMPTCProuter platform: RPI4/

aladdinshammr, 2024-04-15:
this happens to me sometimes when I install a new version and restore a backup of the older one.

lkyddyy, 2024-04-16:
The rpi4 platform is like this. After the system makes modifications, it needs to wait for about 30 seconds, otherwise it will not be saved

Ysurac, 2024-04-16:
Do you have a powered USB switch ? RPI are know to provide low power on USB ports.

MistiqueX, 2024-04-17:
> powered USB switch ? RPI are know to provid

I think i understand what u meen. maybe i need give more power for modem with power manager, like that way on a picture down... 
[image]

MistiqueX, 2024-04-19:
> Do you have a powered USB switch

Can u give a link for powered usb switch?

Ysurac, 2024-04-23:
I don't know this one, but any USB hub with own power supply should be ok.

MistiqueX, 2024-04-23:
> I don't know this one, but any USB hub with own power supply should be ok.

Will try it, and reply as sonn as it possibly. thank pall

MistiqueX, 2024-04-28:
Hi Ysurac, with new usb hab with power 12v 3a - manager modem - see only 1 modem. whats wrong?

closed 2024-08-03 by github-actions[bot]