Ysurac/openmptcprouter #3289: master weight is too high

fareign, 2024-04-16:
if master no internet maybe can not connect to the vps, maybe can not view any website.
in fact,we can make sure master always is ok.
so,if when master is broken down(or more fast track than other interface)? system immediately change another interface to master?
or reduce master weight?

## Specifications

  - OpenMPTCProuter version: 0.60 rc2
  - OpenMPTCProuter VPS version:0.1029 test
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:x86_64
  - Country: cn

Ysurac, 2024-04-17:
Interface status is checked with OMR-Tracker, so if master interface is down then routes are removed and another interface is used.

fareign, 2024-04-17:
> Interface status is checked with OMR-Tracker, so if master interface is down then routes are removed and another interface is used.
@Ysurac 
I know OMR-tracker do this JOB,
but
for example:
I have five interface,only the last one has internet, so it can work at all.

closed 2024-07-22 by github-actions[bot]